dresden-elektronik/deconz-rest-plugin #1801: Samjin sensors vanished from Phoscon sensors tab

srg74, 2019-08-30:
Deconz 2.05.67 - Samjin Multipurpose Sensor and Samjin Water Sensor disappears from sensor listing in Phoscon but remain visible in VNC. Is that a feature or a bug?

hellcry37, 2019-09-02:
Sensor in question is Samsung GP-U999SJVLAEA SmartThings Multipurpose and it used to work partially on deconz firmware 2.05.66
- behave 2.05.66: sensor could be paired and used for open/close and temperature (vibration did not work well, vibration sensor detected every ~ 1h)
- behave 2.05.67: when you pair the sensor apparently successfully you can't see the paired sensor in Phoscon app after pairing. sensor will appear with no name or "multi" name in VNC but not usable.

Note: can not delete sensors in vnc at all, all deleted sensors under VNC interface reappear

manup, 2019-09-02:
In my home setup I have the water sensor which is still visible and working in the Phoscon App with 2.05.67.

I'll need to test the multipurpose sensor and also if joining them freshly still works. Since there were no related changes in the Phoscon App I figure it might be an issue in the REST-API.

hellcry37, 2019-09-03:
> In my home setup I have the water sensor which is still visible and working in the Phoscon App with 2.05.67.
>
> I'll need to test the multipurpose sensor and also if joining them freshly still works. [...]

thank you

manup, 2019-09-09:
I've tested the multipurpose sensor and it works in my setup. The formerly created resources worked and were shown in the Phoscon App as well as deleting the sensor and rejoining it worked.

The test was done with deCONZ 2.05.69 but there were no changes related to the sensor since 2.05.67.

hellcry37, 2019-09-09:
for me latest update offered by deconz addon is 2.05.67, still, waiting for update, thanks for testing this

hellcry37, 2019-09-09:
I'll test after update and get back with a report

hellcry37, 2019-09-10:
Like promise I got back with a report about "Samsung GP-U999SJVLAEA SmartThings Multipurpose" sensor.
Sensor is idd working again on deCONZ 2.05.69. If the vibration trigger ~ 1h still occurs i will open a new ticket.

@manup Thank you very much for your help and tests, much appreciated.

marcelodavanzo, 2019-10-16:
> Like promised I got back with a report about "Samsung GP-U999SJVLAEA SmartThings Multipurpose" sensor.
> Sensor is idd working again on deCONZ 2.05.69. If the vibration trigger ~ 1h still occurs I will open a new ticket.
> 
> @manup Thank you very much for your help and tests, much appreciated.
> 
> SOLVED

Have you managed to fix the issue with the vibration triggering every 1h? I'm on 2.05.69 and have this same issue.

hellcry37, 2019-10-16:
opened this for it: https://github.com/dresden-elektronik/deconz-rest-plugin/issues/1848

hellcry37, 2020-02-13:
#1848 failed

Mimiix, 2020-06-05:
It seems this issue is inactive / resolved. If it is not, please re-open or create another issue.

closed 2020-06-05 by Mimiix