dresden-elektronik/deconz-rest-plugin #1802: (Not an issue) Supported Device

mrnix1001, 2019-08-30:
You can add:

Vendor: sengled
Model: E1G-G8E
Version: 0000000A

LED strip to your supported device list. I just recently moved from a CC2531(which died) to a ConBee II, and gave this LED strip a try and it works perfectly!

Smanar, 2019-08-30:
Will be nice a page that we can update ourselves to add tested devices ?

manup, 2019-08-31:
>Will be nice a page that we can update ourselves to add tested devices ? Or we can edit the wiki one, but in this case, no distinction with official suported device and beta one ?

Yes the Wiki can be edited :)

https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Supported-Devices

The phoscon.de compatibility list is currently not open for editing, here we need to find a easier way to maintain it and accept PR's.

Smanar, 2019-08-31:
And there isn't light/plug/covering section in the wiki ^^, lot of work to do.

manup, 2019-08-31:
Yep, the main reason behind this is that nearly all lights are supported out-of-the-box, since lights most of the time are supported by the generic implementation.

But anyway, a complete list would sure be nice :)

closed 2020-01-05 by stale[bot]