dresden-elektronik/deconz-rest-plugin #1803: RPI4 + USB3 HDD attached Conbee II issue

maximm07, 2019-08-30:
Since I plugged in USB3 SSD to my RPI4 deconz conbee II stick stopped working, I'm using hassio on docker..device associations are ok, becouse if i  unplug the USB drive conbee starts working. I went along and tried different install type python venv and hassio only, the issue is the same..as soon as there is something in USB3 port conbee says firmware not connected

Under the gateway settings firmware says not connected...every time I plug in the USB SSD disk..

If this is not a known issue i can provide more logs..but should be really easy to replicate..Conbee uses latest FW, and phoscon app is version 2.05.57

Smanar, 2019-08-30:
Hello, have you tried an USB extention ?
https://github.com/dresden-elektronik/deconz-rest-plugin/issues/1643

maximm07, 2019-08-30:
I tried to put the m.2 SSD enclosure aprox 1 m from RPI4..but conbee is still attached to the PI itself..no luck..will try to dig up USB extension and dislocate the Conbee II away from the PI

maximm07, 2019-08-30:
did bunch of tests..no luck, as soon as i plug in SSD to any USB port 2 or 3, or via USB hub, conbee starts acting strange..and there is no connectivity to it...BTW I did fresh install of Raspbian..maybee will try some other distro..also tried to plug Conbee to USB hub, USB2 port, USB3 port and all the combinations between. There is some devil in this RPI4 or the USB3 implementation is flawed.

Smanar, 2019-08-30:
Just by curisity can you use this commande "GCFFlasher_internal -l" with the SSD and without ?

maximm07, 2019-08-30:
Need to read about this how to use GCFFlasher_internal with docker..but I'm not sure USB3 is the issue, this happens also if I plug SSD to USB2 port 1 meeter away. Will test more.

maximm07, 2019-08-30:
Think I found a solution :), RPI4 is verry picky about USB controllers used in external ssd enclosures, although i have "the working" ASmedia one, without any hacks. But anyway I tried the workaround for non working controllers, disabling the UAS.(https://www.raspberrypi.org/forums/viewtopic.php?t=245931).and bingo..Conbee II its working :D :D. Maybe someone will find this useful.

MacJawa, 2019-08-30:
Just a guess, but did you try to plug ConBee on USB extension cable 1m from RPi?

maximm07, 2019-08-30:
No I , didn't have such long cable on hand..tried with short one..but didn't work. But this should not cause RPI not to detect the USB stick..maybe some connection issues with sensors..

Luigi-01, 2019-09-27:
I had the same issues, the combee II battles with other USB devices. (Persistent USB Devices)
You could Create rules file for all other USB devices. Than the Conbee II can have his Persistent USB.
For example see:  https://www.domoticz.com/wiki/PersistentUSBDevices

CoreDi, 2019-10-15:
Hello, same Problem here. Attached USB 3 SSD (Boot) and Conbee II is not working (No Firmware showing and can not connect Zigbee Devices). 
Can someone point me to the prefered way (UAS or Persistent)? I have an "Inateck Adapter USB 3.0 to SATA" which after the RPI4 boots from (after MMC).

CoreDi, 2019-10-16:
@Luigi-01 :
Maybe you can support? I added in /etc/udev/rules.d/99-usb-serial.rules a line for my USB SSD Boot Device (after MMC Boot) from your example site.
But it don´t work. I´m doing something wrong i´m think.

Luigi-01, 2019-10-18:
I do not know if the solution I gave to you works on a Pi.  You have to be very careful with the text file. Type mistakes are made easily. I checked some other sites as well to verify how to do this.
All other USB devices are are moved to an other name. Than the Conbee II could connect on the standard USB connection.
[image]

CoreDi, 2019-10-18:
Thank you very much.
I tested it already but no success. But it´s working now here. The Problem was the SSD on USB3. I switched it now to USB2 and it works.
But the Problem with SSD USB3 plugged in is still open.. :(

lukaszbel, 2019-11-17:
I also have the same problem.... :(

daandersson, 2019-11-20:
I have this problem too. Conbee2 connected to Usb2 port of Pi4 and SSD connected to USB3 port. Tried every possible combination of USB2/3 hubs both powered and not powered as well as extention cords for the conbee2. Nothing works. When SSD disk is removed from pi4 the conbee works. This is really frustrating and seems there is no solution to it?

czasas, 2019-12-05:
I can confirm that problem really exist, to making conbee working , USB drive needs to be put on USB 2.0

rkotulan, 2019-12-16:
I had the same issue. My config is HomeAssistant Ri4 and SSD connected to USB3 port. Conbee2 in USB2. I put ConBee2 on extension cable (2m) and then Phoscon App connected to stick. Magic :)

CC1337, 2019-12-20:
I can confirm that connecting my SSD (from where my system boots) to USB2, ConBee 2 stick starts working. No Success with active and passive USB2 and USB3 cables 0,5-5m, hubs etc. 
As this is limiting my SSD speed a lot, I hope it can be fixed with new firmware. Currently using 264A0700 on RPi 4 with Raspbian. Otherwise I have to return the stick in 25 days.

lfaggian, 2019-12-20:
I have the same issue. Pi4 with SSD connected to USB 3.0 and ConBee II (with extension cable) connected to USB 2.0.

lukaszbel, 2020-01-25:
Does anybody resolve this problem? It is very frustrating :(

00lex, 2020-01-29:
Best Solution is maybe a Raspbee Module. I use one and have no issues with my USB3 connect SSD. 

Or you can try a sata dock like this https://www.amazon.de/FIDECO-Externe-Festplatten-Dockingstation-Aluminium/dp/B075TZF3Z7/ref=pd_aw_sbs_147_6/257-3524995-1630940?_encoding=UTF8&pd_rd_i=B075TZF3Z7&pd_rd_r=63958faa-df78-4d20-899b-3d659a8b62f3&pd_rd_w=WZAqk&pd_rd_wg=jigoY&pf_rd_p=ecf0cb80-06b1-4598-85b1-6d77c4f95871&pf_rd_r=CDMMGEPQ71NPV9V7EXDZ&psc=1&refRID=CDMMGEPQ71NPV9V7EXDZ

mvlaanderen, 2020-02-01:
I just had the same issue after migrating my setup to a RPI4b with SSD (on USB3). After that I also encountered the problems mentioned by maximm07. After connecting an USB extensioncable (1,5m) the problems were gone. So it really helps ;-)

Luigi-01, 2020-02-02:
The solution I gave works fine within Ubuntu, however I noticed that a Raspberry Pi does behave a little different. The Raspberry Pi does not see all the information. for that reason you have to adjust the **99-usb-serial.rules** file.
# sudo lsusb -v | grep 'idVendor\|idProduct\|iProduct\|iSerial'
Example Z-wave stick 
idVendor           0x0658 Sigma Designs, Inc.
idProduct          0x0200 Aeotec Z-Stick Gen5 (ZW090) - UZB
iProduct              0 
iSerial                 0 

Because there is no information about the iProduct or iSerial, do not include these in the **99-usb-serial.rules** 
My files looks like this:
SUBSYSTEM=="tty", ATTRS{idVendor}=="0658", ATTRS{idProduct}=="0200", SYMLINK+="ttyUSB-Zwave"
By doing this the Z-wave stick has no conflict anymore with  the conbee II stick.
The original manual I used for this: https://www.domoticz.com/wiki/PersistentUSBDevices

kregory, 2020-02-20:
> 
> 
> Does anybody resolve this problem? It is very frustrating :(

I resolve this issue by putting the usb disk on usb2 port and adding usb-storage.quirks=174c:55aa:u
 to /boot/cmdline.txt

stijndefieuw, 2020-03-12:
you mean with 'usb disk' the SSD drive or conbee stick?

kregory, 2020-03-12:
The SSD drive

rweise, 2020-03-22:
Is there a solution for this problem? It is a bug in deconz and not in usb, because i have the same problem. If i connect a drive to usb 3.0 port Phoscon and all deconz stuff will not work anymore. I use a Raspbee (I) shield, so this problem is not hardware specific to the usb-use of a conbee stick.
It is also not a power problem because a usb-hub with power-supply changed nothing.
It can not be a solution to connect to usb 2.0 Port, because i want to use my RPI4 for NAS and Homeautomation.

mvlaanderen, 2020-03-23:
> Is there a solution for this problem? It is a bug in deconz and not in usb, because i have the same problem. If i connect a drive to usb 3.0 port Phoscon and all deconz stuff will not work anymore. [...]

I experienced the same problem you describe. For me, the solution was to use a 2m extension cable between the Pi and the Conbee II stick. Still strange, but hey.. it works :).

rweise, 2020-03-23:
The problem ist the bad shielding of usb3.0.
If you use a very long (2m) extension for the Conbee it will work or you
must place the ssd more than 2m away.
At this time i have no idea. if there is some hardware that will work.
Raspee2 seams to be a little bit better, but not good enough.

if i had the time i will work on a external Antenna for Raspbee v1 and will
report the result here to.

daandersson, 2020-03-23:
@mvlaanderen can you please confirm that you are sucessfully running RPI4 with HDD/SSD on USB3 port and simultaniously a Conbee2 stick on one of the USB2 ports? I totally cannot make this work and I have basically tried everything. My USB extension is only 1 meter but I doubt that whould make any diffference, there must be something else you have done different in your installation?

rweise, 2020-03-23:
The problem is also well known from the Fritzbox. There is where i found the solution. My test with Raspbee and a 2m USB-Cable to the HDD (a normal and a ssd) was successful. For me this is not the way to go. but it may be for Conbee user.

daandersson, 2020-03-23:
Interesting. If it is indeed shielding issue it may be better solved by more shielding on cables or a magnetic ring on the cable.

rweise, 2020-03-23:
No it is not so easy.
The problem is the high frequency and a very small leak in shielding is
enough to destroy your 2.4Ghz  wireless network.
And zigbee is not so robust like a wifi Network.
:-(

rkotulan, 2020-03-23:
I have short cable to SSD HDD and long to ConBee II stick.

mvlaanderen, 2020-03-24:
> @mvlaanderen can you please confirm that you are sucessfully running RPI4 with HDD/SSD on USB3 port and simultaniously a Conbee2 stick on one of the USB2 ports? I totally cannot make this work and I [...]

Correct. I am using an RPI4 4GB with an SSD on USB3. I connected the ConBee II stick to a USB2 port with an 2m extension cable in between.

Mimiix, 2020-06-06:
It seems this issue is resolved or otherwise inactive. If it is not, please re-open!

closed 2020-06-06 by Mimiix

burow1978, 2020-06-18:
Hello, I use an RPI4 4GB and an SSD on USB3 but with a RaspBee II and also this problem. 
The workaround to connect the SSD via a USB extension has worked.
Is there a better solution (software) for this problem?

Mimiix, 2020-06-19:
@burow1978 Unfortunately, this is out of our hands and has to do with a bug in USB 3.0. Please see this page: https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Zigbee-Error-Codes-in-the-Log  there is a link to a whitepaper from Intel.

If you still have a issue/question: Please open a new issue :)

mixu-, 2020-10-12:
May I suggest adding a note about the USB interference in the FAQ, here: https://phoscon.de/en/support#faq ?
That would have saved me several hours today.

kanintesova, 2021-02-05:
Connection is extremely picky. I have a one meter extension cable and had lots of trouble tonight. Then I move the stick like 20 cm to another location and it all works. I too have a USB 3 disk connected. It might make it worse. But extension cable is a good thing for the conbee stick.

marecabo, 2021-03-19:
I also had this issue and it seems to be gone after I ...
- used a 1.5 m  USB2 cable for the ConBee II stick (as mentioned above) and 
- wrapped some tin foil around the plastic case (where the SATA port is) of my USB3/SATA-Adapter and the attached SSD as recommended in the paper mentioned in https://github.com/dresden-elektronik/deconz-rest-plugin/issues/1598#issuecomment-508904699.

00lex, 2021-03-21:
im doing a lot of experiments too with tin foil a while ago. im using a raspbee2. 
the downside of tin foil is heat around the usb plug from the usb drive. 
and if you wrap it over the cable, data connection is lost. 

now I found out its enough to put the usb cable from the drive parallel to the plug in one line as it comes out of the pi. 
in my imagination the usb3 radiation works like a wifi Antenna

https://rootsaid.com/wp-content/uploads/2017/09/omni-pattern.gif

gloomytree, 2021-03-24:
> anyway it works. I wish the next deconz modules get the same isolations agains usb3 as the 1st gen. raspbee1 works without issues with usb3

Raspbee1 does not work without issues. I have one and it looses all connectivity when I plug in an SSD with a USB - SATA Converter.

Oderik, 2021-11-18:
@00lex and others: Can anybody confirm where the USB3 interference originates from? Is it

- USB 3 device
- USB 3 cable
- the Pi itself
- any combination of the above?

I think about shielding the Pi and conneting my Raspbee II via a self made extension. Does that make any sense?

(I have read the thoughts about cable placement - I will experiment with it, too)

Mimiix, 2021-11-18:
@Oderik https://www.intel.com/content/www/us/en/products/docs/io/universal-serial-bus/usb3-frequency-interference-paper.html

gloomytree, 2021-11-18:
> @00lex and others: Can anybody confirm where the USB3 interference originates from? Is it
>
> * USB 3 device
> * USB 3 cable
> * the Pi itself
> * any combination of the above?
>
> I think about shielding the [...]

I don't think it is worth the effort. A fast SD Card will work fine, if you are interested in upgrading, a different Formfactor than a RPi will be better suited. At least I gave up trying to shield an SSD.

rweise, 2021-11-18:
I do not know what the problem is, but i use a Deskpi case and they send some time ago an update for the internal USB 3.0 SSD/M.2 Adapter and some shielding stuff. it makes all a little bit better, but not perfekt. I use a Conbee II Stick with 1m USB-Cable without any trouble, Wifi works very well, only Bluetooth will only work in 1 or 2 m distanz. I think there is no solution where a usb 3.0 drive and a Conbee will work nearby. HF-shielding is a strange thing. For me a sd-card is also no option, because i use the RPI4 not only for home automation. i use the internal 2gb ssd as Net storage to. So i think you must accept the use of a cable for the stick.

Oderik, 2021-11-18:
> I don't think it is worth the effort. A fast SD Card will work fine, if you are interested in upgrading, a different Formfactor than a RPi will be better suited. At least I gave up trying to shield an SSD.

Thanks for the suggestion! I switched to an SSD not to gain speed performance but reliability. I run Home Assistant on it and I've heard that it's not unlikely that an SD card might fail sooner or later because of intense writing.

I just solved the issue by adding a self made shielded extension cable to move the Raspbee away from the interferences.

olemr, 2021-11-28:
Just chiming in after struggling for hours.
I'm using the Argon ONE M.2 Case for Raspberry Pi 4 and ended up needing a 1.5m passive USB extension between a black USB2 port on the pi and the Conbee II stick. Using the USB3 jumper that came with the case.

What also threw me off for a while was that after not being able to pair anything, even though the Gateway Tab showed the Conbee II Firmware version, I had to do a re-install of the Debian package.
After that, things worked fine.

Smanar, 2021-11-29:
The re-install of the Debian package have changed something ? Not an update ?

olemr, 2021-12-01:
To be fair, I had installed 2.13.02 on the Rpi4 SD card prior to the image copy to the SSD.
Struggling for hours after booting from the SSD (no SD present), I ran (in despair) the same dpkg -i command again, and things came alive.
Have added several lights, switches and sensors since. All OK.