dresden-elektronik/deconz-rest-plugin #1807: Request for Innr SP 224 support

thelogman, 2019-08-31:
Greetings, 

Please consider taking a look at the Innr SP224 Plug (US version). deCONZ will join the plug to the controller, but I believe some device capabilities are missing. The EU version of this plug, Innr SP120, reports power usage and load (watts).

[image]

[image]

Thanks,

--Barry

ebaauw, 2019-08-31:
Please provide info described in the [Wiki](https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Request-Device-Support).

thelogman, 2019-09-01:
Hi ebaauw,

Below is a screenshot of the clusters of the node. I had already included screenshots of the Basic Cluster Info Panel and the Node Info Panel above. 

Please let me know if you need anything else as perhaps I'm misunderstanding what is needed.

[image]

Thanks,

--Barry

manup, 2019-09-01:
Related PR to disable state.bri  for innr SP 224 https://github.com/dresden-elektronik/deconz-rest-plugin/pull/1804

ebaauw, 2019-09-01:
As I feared, these new plugs don't support power monitoring.  No 0x0B04 _Electrical Measurement_ nor 0x0702 _Simple Metering_ clusters.  Note that these use a Silicon Labs mac address vs Jennic for the SP120.  I think the relay hardware changed as well, the UK plug  (SP222) supports over 3000W, where the SP120 only supports 2300W (10A @ 230V).

innr never publicly mentioned the power monitoring features of the SP 120, presumably because not all their plugs would support this.  

Maybe the 0xFC82 cluster provides something useful, but we'd need a sniffer log of the plug connecting to an innr hub to find out.

kasteleman, 2019-09-01:
I know for sure, confirmed by INNR support, the SP 222 does not support
power monitoring.

ramiroasl, 2019-09-27:
> I know for sure, confirmed by INNR support, the SP 222 does not support power monitoring. Op zo 1 sep. 2019 om 13:30 schreef Erik Baauw <notifications@github.com>

Before submitting a new request, @kasteleman were you able to get an Innr SP 222 to pair to deCONZ?
I've tried pairing it as both a Light and a Switch ("Other") via the Phoscon App as well as through the REST client via Touchlink, without any luck.

ebaauw, 2019-09-27:
deCONZ doesn’t support touchlink pairing.  You need to search for devices from Phoscon or the old web app and reset the plug for regular pairing.  It should appear as a light.

ramiroasl, 2019-09-27:
> deCONZ doesn’t support touchlink pairing. You need to search for devices from Phoscon or the old web app and reset the plug for regular pairing. It should appear as a light.

Thanks for the prompt reply, @ebaauw!
I've tried the automatic Light discovery multiple times by factory-resetting the plug (holding the button for 5 secs, waiting for blinking lights, etc.) and also holding it for 10 secs until all the lights turn off, then turning it on again (power-cycling, is it?). Still doesn't pick it up.
The innr SP222 has never been paired to anything, I unpacked it a few hours ago.
Worth mentioning maybe that I've tried through both the Phoscon App and the old UI.
Any ideas? Maybe changing the ZigBee channel?

ebaauw, 2019-09-27:
I only have the SP120 EU plug.  According to the [SP222 installation manual](http://www.innrlighting.com/clients/innr/themes/uploads/1560876928_SP%20222%20Installation%20Manual_V8.pdf), the green LED should blink once a second when it's ready to pair.  The blinking should stop when it has been paired.  The plug should be visible in the deCONZ GUI at this time.

I think to power-cycle the plug, simply unplug and re-insert it.

closed 2020-02-01 by stale[bot]