dresden-elektronik/deconz-rest-plugin #1809: Report back non-ikea bulbs when using ikea dimmer.

vandalon, 2019-09-01:
Hi,

Would it be possible to have non-ikea bulbs repot back their state when changing state (brightness and on/off) via the ikea dimmer? Now when using the dimmer ikea bulbs immediately report back their state after a change but hue and innr bulbs take up to 10 seconds to report back even do the change in brightness or on/off state is instant visiable.

ebaauw, 2019-09-01:
That requires support by the bulb firmware.  Hue and innr bulbs don't support attribute reporting.  IKEA and OSRAM bulbs do.

vandalon, 2019-09-01:
And polling would be to intensive?

ebaauw, 2019-09-01:
Bulbs that don't report attributes are polled periodically already (that's causing the update).  Unfortunately, this doesn't scale well; the more bulbs in your network, the longer it takes before a bulb is polled again (up to 10 seconds in your case, but I've seen way worse).

In theory, doing a fast-poll should be possible, if the dimmer uses group commands (unicast commands might not be visible to the coordinator).  It doesn't scale as well as reporting, but not as badly as regular polling.  It would depend on the REST API having an up-to-date administration of the lights in the group (to know which lights to poll) and their capabilities (to know what clusters to poll).

Updating the cache of the light states on seeing a command would scale better, but this becomes very challenging for relative commands (like _Move_ for setting `bri_inc`, instead of absolute _Move to Level_ for setting `bri`.

vandalon, 2019-09-01:
How fast would fast-poll be more or less?

ebaauw, 2019-09-01:
That would of course depend on the number of lights in the group, and the other traffic happening on the network, but I'd guess 1-2 sec.

vandalon, 2019-09-02:
That sounds acceptable :) This would make automations in home assistant and/or node-red  work faster at least :)

vandalon, 2019-09-02:
For reference: I'm using node-red to set a specific light next to tv to the same brightness as another light. but only if the tv is on, if the tv is off, the lights should always be off :)

Another way to do this would be using a dummy light connected to the ikea dimmer and sync all other lights with that light, this would still give some delay though.

vandalon, 2019-09-02:
Do you know of OSRAM lighty & smart+ both have the report feature?

ebaauw, 2019-09-02:
The lightify plug and E14 colour temperature lights and the smart+ plug that I’ve seen do.  I don’t know for sure about any other devices.

vandalon, 2019-09-02:
Ok, just ordered a Smart+ A60 10 W/ E27 RGBW, i'll let you know :)

vandalon, 2019-09-03:
I can confirm the above lamp does _not_ support reporting :)

vandalon, 2019-09-03:
It seems to work ok. But it is slower then with the ikea bulbs and when dimming to bri: 1 it takes a while before it reports that. (up to 5 seconds?)

ebaauw, 2019-09-03:
You might want to check the reporting configuration in the deCONZ GUI.  Open the _Cluster Info_ panel, select the bulb’s _Level Control_ cluster and double click on the _Current Level_ attribute.

vandalon, 2019-09-03:
Awesome! that helps a lot! Thanks Erik!

ebaauw, 2019-09-03:
That will send a report every second, even when there's no changes.  You want to set it to: 1/300/1, send a report per second when brightness changes by 1 or more and send a periodic report per 5 minutes.

vandalon, 2019-09-03:
Ah ok, i thought so, but when checking the websocket data i did not see a report every second

vandalon, 2019-09-03:
actualy, when setting it to 1,300,1 it again takes a long time before the report is send out when going to 1 or 255 in bri

closed 2020-01-08 by stale[bot]