dresden-elektronik/deconz-rest-plugin #1811: Is there a way to make ikea rgb bulbs work with colortemp settings?

vandalon, 2019-09-01:
I have an ikea tradfri rgb bulb. it seems that it does not react to ct (colortemp) requests where others (innr en philips) do. Is there a way to work around this and get this working within deconz?

Smanar, 2019-09-01:
I don't understand, if it's a RGB bulb it's support color mode, not temperature mode ?
There is some bulb with both (in philips like you said), but from my memory, ikea don't. The problem is from the bulb, hardware.

vandalon, 2019-09-01:
I know, but it should be possible to translate ct to color, so maybe that can be build into deconz?

lbschenkel, 2019-09-07:
deCONZ has a workaround for that, and it'll internally convert `ct` into `xy` when talking to IKEA color lamps. There's a catch, though: it'll only work for unicast commands (sent directly to the light). The workaround doesn't kick in for group commands.

vandalon, 2019-09-07:
I've tested this and indeed this is true, but home assistant and phoscon don't show a ct slider :)
Also it would be very nice to have for groups as well since that is what most people use in bigger rooms I guess

vandalon, 2019-09-07:
I understand this is an issue since that would cause deconz to send color specs to all lights and not ct, and that would again fail light which are ct only I guess

closed 2020-01-12 by stale[bot]