dresden-elektronik/deconz-rest-plugin #1812: Gledopto GL-B-008Z bulb transitions delayed in Homebridge

alialrahma, 2019-09-01:
I have a Conbee II running the latest version (2.05.67) and have a single GL-B-008Z (Version 2.0.0) bulb in my setup. The bulb shows a delayed response when performing any action (i.e. power on/off, brightness, color) in HomeKit via Homebridge (homebridge-hue plugin). As examples, find below the logs from Homebridge Docker. Similar logs occur for any other action.

The logs below are from controlling the bulb through HomeKit on an iPhone X (iOS 12.4.1). Only one action was made: set brightness from 100% to 44%.
```
[9/1/2019, 8:21:27 PM] [Phoscon-GW] Lamp: homekit brightness changed from 100% to 44%
[9/1/2019, 8:21:28 PM] [Phoscon-GW] Lamp: set homekit brightness from 44% to 51%
[9/1/2019, 8:21:29 PM] [Phoscon-GW] Lamp: set homekit brightness from 51% to 44%
```

The following logs are from controlling the bulb through the Phoscon App website, which are amplified in comparison to HomeKit. Only one action was made: set brightness from 31% to 100%.
```
[9/1/2019, 8:19:35 PM] [Phoscon-GW] Lamp: set homekit brightness from 31% to 100%
[9/1/2019, 8:19:36 PM] [Phoscon-GW] Lamp: set homekit brightness from 100% to 43%
[9/1/2019, 8:19:36 PM] [Phoscon-GW] Lamp: set homekit brightness from 43% to 100%
[9/1/2019, 8:19:37 PM] [Phoscon-GW] Lamp: set homekit brightness from 100% to 70%
[9/1/2019, 8:19:38 PM] [Phoscon-GW] Lamp: set homekit brightness from 70% to 90%
[9/1/2019, 8:19:39 PM] [Phoscon-GW] Lamp: set homekit brightness from 90% to 100%
```

ebaauw, 2019-09-01:
See #1799.

alialrahma, 2019-09-04:
This issue is still persisting. A REST API access shows the code below, with clearly no signs of transitiontime. I tried to add it via the PUT command `IP_address:80/api/id/lights/1/state` with JSON request of `{ "transitiontime": 0 }` and it returned a 200 OK response, but no signs of changes to the lights. If I am able to add transitiontime via VNC or REST, it would fix my issue. 

```
{
  "ctmax": 65535,
  "ctmin": 0,
  "etag": "b37ba5f49564051183ef7c7cd05a105d",
  "hascolor": true,
  "manufacturername": "GLEDOPTO",
  "modelid": "GL-B-008Z",
  "name": "Lamp",
  "state": {
    "alert": "none",
    "bri": 170,
    "colormode": "ct",
    "ct": 366,
    "effect": "none",
    "hue": 43776,
    "on": true,
    "reachable": true,
    "sat": 251,
    "xy": [
      0.1507,
      0.0426
    ]
  },
  "swversion": "2.0.0",
  "type": "Extended color light",
  "uniqueid": "00:12:4b:00:1c:43:27:e1-0b"
}
```

ebaauw, 2019-09-04:
`transitiontime` is write-only and doesn't show in the REST resource.  Technically it's a ZigBee command parameter, rather than a ZigBee attribute.  You need to specify it in conjunction with an actual command, e.g. PUT with a body of `{"bri": 254, "transitiontime": 0}`.

alialrahma, 2019-09-04:
Apologies, I may have not been clear and might have not understood how REST works. I would like to make a permanent change to transitiontime by setting it to 0. It is currently still showing the transition delay on Homebridge. I have attempted to push a change in Transition Time in Level Control of my bulb in the VNC but it did not set it permanently, and perhaps did set it temporarily as it was responsive for a few minutes.

ebaauw, 2019-09-04:
>  I would like to make a permanent change to transitiontime by setting it to 0.

That is technically not possible.  Transitiontime is not a bulb setting.

alialrahma, 2019-09-04:
Thanks for the clarification. I believe the issue will never be resolved then and I am glad I only purchased 1 Gledopto bulb to test instead of the intended 14.

closed 2019-09-04 by alialrahma