dresden-elektronik/deconz-rest-plugin #1813: Lost control over all lamps! :/

tnowak, 2019-09-02:
Not sure if this is right place but I have a major and urgent problem with Conbee II and/or deCONZ. I've bought Conbee II stick, paired some Philips, Osram and Aqara bulbs and first FLS-PP lp. All worked fine for about 2 weeks. Yesterday I had to reboot RPi with docker running deconz and I’ve totally lost control over all my lights! I can't even turn them off.......

API still works fine, my integration with FHEM and Loxone seem to change bulb icon state in deCONZ, but actual lights don't change their state. After couple of minutes deconz in started they all become greyed out!

[image]

In Phoscon App I see such Conbee II details:
Version is 2.05.67 / 26.08.2019
Firmware 264A0700

In that Raspberry PI I have a z-stick also that occupies ACM0, so conbee is ACM1:
It is visible inside raspberry system direct:

root@loxberry:~# udevadm info /dev/ttyACM1
P: /devices/platform/soc/3f980000.usb/usb1/1-1/1-1.2/1-1.2:1.0/tty/ttyACM1
N: ttyACM1
S: serial/by-id/usb-dresden_elektronik_ingenieurtechnik_GmbH_ConBee_II_DE1996953-if00
S: serial/by-path/platform-3f980000.usb-usb-0:1.2:1.0
E: DEVLINKS=/dev/serial/by-id/usb-dresden_elektronik_ingenieurtechnik_GmbH_ConBee_II_DE1996953-if00 /dev/serial/by-path/platform-3f980000.usb-usb-0:1.2:1.0
E: DEVNAME=/dev/ttyACM1
E: DEVPATH=/devices/platform/soc/3f980000.usb/usb1/1-1/1-1.2/1-1.2:1.0/tty/ttyACM1
E: ID_BUS=usb
E: ID_MODEL=ConBee_II
E: ID_MODEL_ENC=ConBee\x20II
E: ID_MODEL_ID=0030
E: ID_PATH=platform-3f980000.usb-usb-0:1.2:1.0
E: ID_PATH_TAG=platform-3f980000_usb-usb-0_1_2_1_0
E: ID_REVISION=0100
E: ID_SERIAL=dresden_elektronik_ingenieurtechnik_GmbH_ConBee_II_DE1996953
E: ID_SERIAL_SHORT=DE1996953
E: ID_TYPE=generic
E: ID_USB_CLASS_FROM_DATABASE=Communications
E: ID_USB_DRIVER=cdc_acm
E: ID_USB_INTERFACES=:020201:0a0000:
E: ID_USB_INTERFACE_NUM=00
E: ID_VENDOR=dresden_elektronik_ingenieurtechnik_GmbH
E: ID_VENDOR_ENC=dresden\x20elektronik\x20ingenieurtechnik\x20GmbH
E: ID_VENDOR_FROM_DATABASE=Dresden Elektronik
E: ID_VENDOR_ID=1cf1
E: MAJOR=166
E: MINOR=1
E: SUBSYSTEM=tty
E: TAGS=:systemd:
E: USEC_INITIALIZED=9876427694

I run deCONZ in docker with such variables:
DECONZ_WEB_PORT=8080
DECONZ_WS_PORT=8443
DECONZ_DEVICE=/dev/ttyACM1
mapped devices:
/dev/ttyACM1 --> /dev/ttyACM1
volume mapping:
/root/.local/share/dresden-elektronik/deCONZ --> /opt/deconz/
/etc/localtime --> /etc/localtime
network: host

Inside deconz docker ACM1 is also visible:
root@loxberry:/# udevadm info /dev/ttyACM1
P: /devices/platform/soc/3f980000.usb/usb1/1-1/1-1.2/1-1.2:1.0/tty/ttyACM1
N: ttyACM1
E: DEVNAME=/dev/ttyACM1
E: DEVPATH=/devices/platform/soc/3f980000.usb/usb1/1-1/1-1.2/1-1.2:1.0/tty/ttyACM1
E: MAJOR=166
E: MINOR=1
E: SUBSYSTEM=tty
and no other devices are available:
root@loxberry:/# udevadm info /dev/ttyACM0
Unknown device, --name=, --path=, or absolute path in /dev/ or /sys expected.
root@loxberry:/# udevadm info /dev/ttyAMA0
Unknown device, --name=, --path=, or absolute path in /dev/ or /sys expected.
[... 152 lines cut ...]
12:23:05:846 0x00212EFFFF032B30 error APSDE-DATA.confirm: 0xD0 on task,
12:23:16:644 0x0017880106BAE90B error APSDE-DATA.confirm: 0xD0 on task,
12:23:27:448 0x0017880106BAE729 error APSDE-DATA.confirm: 0xD0 on task,
12:23:38:355 0xF0D1B800000FFAF3 error APSDE-DATA.confirm: 0xD0 on task,
12:23:49:147 0xF0D1B800000FF301 error APSDE-DATA.confirm: 0xD0 on task,
12:23:51:230 Current channel 11,
12:23:51:239 Device TTL 2018 s flags: 0x7,
12:24:00:939 0x00178801021D40B2 error APSDE-DATA.confirm: 0xD0 on task,
12:24:12:745 0x84182600000E266D error APSDE-DATA.confirm: 0xD0 on task,
12:24:23:545 0x84182600000E279F error APSDE-DATA.confirm: 0xD0 on task,
12:24:34:444 0xF0D1B8000010741A error APSDE-DATA.confirm: 0xD0 on task,
12:24:45:349 0xF0D1B80000106F3D error APSDE-DATA.confirm: 0xD0 on task,
12:24:51:229 Current channel 11,
12:24:51:239 Device TTL 1957 s flags: 0x7,
12:24:56:139 0xF0D1B800000FF2FC error APSDE-DATA.confirm: 0xD0 on task,
12:24:56:252 dev /dev/ttyAMA0,
12:24:56:253 GW firmware version: 0x264a0700,
12:24:56:253 GW firmware version is up to date: 0x264a0700,
12:25:06:947 0x00212EFFFF032B30 error APSDE-DATA.confirm: 0xD0 on task,
12:25:17:842 0x00212EFFFF032B30 error APSDE-DATA.confirm: 0xD0 on task,
12:25:28:645 0x0017880106BAE90B error APSDE-DATA.confirm: 0xD0 on task,

Important information: before restart Conbee II worked fine with just a short 10cm USB cord extension (so ~10 cm from z-stick). When I saw those APSDE-DATA erros I've event replaced it with a 2m USB cord extension. But it doesn't change anything.

Please help, before my girlfriend will kill me for not able to switch off/on lights.

manup, 2019-09-06:
Hey @tnowak since communication was through mail. Do you have any updates on the new setup, is it still up?

tnowak, 2019-09-06:
Finally I had to reset all lights. It was somehow easy to reset Aqara and Osram bulbs, but to reset Philips bulbs I had to buy Philips Dimmer Switch. Dimmer has already come, I've reset last three bulbs and now all is online again and no problems since. I'm still worried because I don't know what has happened. I suspect Conbee and Aeontec z-stick might have changed their system names during reboot, but I'm not sure if this might brake things so hard. I've changed naming scheme for calling Conbee II to /dev/seria/by-id anyway. The other issue that worries me is that Phoscon/deCONZ does not report setup changes correctly (hidden page list). I would like to thank manup for helping me deal with this and despite we didn't find solution finally, we've exchange many emails during 2 days and such support from manufacturer is exceptional.

closed 2020-01-11 by stale[bot]