dresden-elektronik/deconz-rest-plugin #1814: IKEA TRÅDFRI Motion sensor: what is meaning of state.dark and state.presence?

maxstein242, 2019-09-02:
Dear all,

The following question relates to IKEA TRÅDFRI Motion sensor, 2.0.022.

I receive the following event using web socket of Deconz REST API and struggle to unerstand the meaning / logic of the attributes:
- state.dark
- state.presence (I assume true will indicate motion, but what does event with state.presence == false mean?)

Here is a sample entry:

{
  "e": "changed",
  "id": "5",
  "r": "sensors",
  "state": {
    "dark": true,
    "lastupdated": "2019-09-02T16:25:40",
    "presence": false
  },
  "t": "event",
  "uniqueid": "14:b4:57:ff:fe:53:56:f3-01-0006",
  "timestamp": "2019-09-02T16:25:40.714Z"
}

thanks for your help and best regards,
Max

ebaauw, 2019-09-02:
When the sensor is on Night mode, `state.dark` reflects whether it’s dark (after motion has been detected).  When it’s on Day mode, `state.dark` will be true.

The sensor doesn’t report anything when motion is no longer detected.  DeCONZ resets `state.presence` to false after `config.duration` seconds.

maxstein242, 2019-09-03:
Thank you @ebaauw ! 

If there is a movement recognized before configuration.duration seconds pass, will the countdown start once again for the duration of configuration.duration seconds?

ebaauw, 2019-09-03:
If the sensor sends another motion detected message, the timer is reset.  The sensor seems to take some time (60-90 seconds ?) to “recharge” before sending another message, though.  Hence the REST API plugin resets `config.duration` to 60 when it’s less than that.

closed 2020-01-08 by stale[bot]