dresden-elektronik/deconz-rest-plugin #1815: Change Hue Outdoor Sensitivity

skeptizzle, 2019-09-03:
Hi.
Is it possible to change the sensitivity of the Hue Outdoor Motion Sensor? 
I can't seem to find the "sensitivity" parameter in the REST config on the sensor.

Thank you!

ebaauw, 2019-09-03:
Odd, it should be there.

Could you please list the REST resource (e.g. through `ph get /sensors/xx`)?

Could you please double-check in the deCONZ GUI that the _Sensitivity_ attribute is shown in the _Cluster Info_ panel for the _Occpancy Sensing_ cluster?  If you double-click the attribute, can you read its value?

skeptizzle, 2019-09-03:
Hi @ebaauw 

Thanks for coming back to me. 
I'm new to this hole backend of deCONZ and i have overlooked that the sensors have different ID's depending on presence, temperature and lightlevel.

So i was simply looking at the wrong ID.

While i have your attention, can you explain the difference between sensitivity and sensitivitymax? 
And what does the delay do?

Thank you very much for pointing me towards digging deeper 😄

ebaauw, 2019-09-03:
Sensitivity max is fixed at 2.  It's the maximum value for sensitivity, which runs from 0 (low) to 2 (high).

`config.delay` sets the time before the Motion sensor reports "no motion" to the gateway (which it translates to `state.present` becoming false) after it no longer detects motion.  The Hue bridge doesn't expose this, even though it's supported by the sensor (it's the _PIR occupied to unoccupied delay_ attribute).

skeptizzle, 2019-09-03:
> Sensitivity max is fixed at 2. It's the maximum value for sensitivity, which runs from 0 (low) to 2 (high).
>
> `config.delay` sets the time before the Motion sensor reports "no motion" to the gateway [...]

Well, mine look like this:
`{
    "config": {
        "alert": "none",
        "battery": 100,
        "delay": 0,
        "ledindication": false,
        "on": true,
        "pending": [],
        "reachable": true,
        "sensitivity": 3,
        "sensitivitymax": 4,
        "usertest": false
    },
    "ep": 2,
    "etag": "29fbff3d3e17e6245b70fe6e02fa6ba4",
    "manufacturername": "Philips",
    "modelid": "SML002",
    "name": "PIR Indkørsel",
    "state": {
        "lastupdated": "2019-09-03T08:59:35",
        "presence": false
    },
    "swversion": "6.1.0.25261",
    "type": "ZHAPresence",
    "uniqueid": "00:17:88:01:06:44:10:b8-02-0406"
}`

I tried both 1 and 3 for sensitivity, but i can't see a change.. 😞

ebaauw, 2019-09-03:
Ah, the outdoor sensor has more sensitivity levels.  Hadn't seen that before, but now `sensitvity_max` makes a lot more sense.  I suppose the Hue app also shows more levels?

I my experience with the indoor sensor, sensitivity mostly influences the distance within which the sensor detects motion, not so much the amount of motion.  You might want to set `config.usertest` while testing sensitivity to make the sensor blink when detecting motion.  I think that setting resets itself after a couple of minutes.

Also note that these are PIR (passive infrared) sensors.  They detect motion by changes in the infrared spectrum.  They detect motion passing by the sensor more easily than motion towards or away from the sensor.  I find they had more difficulty detecting motion during the recent heat waves (when my room temperature rose above 30°C).

skeptizzle, 2019-09-03:
So I will try 0, it's getting triggered by cars on the road running parallel to the sensor at around 12-13 meters away even with the value 1..
I will set the delay as well, because I get triggers every few seconds.. 
Is the sensitivitymax only so that one know what the max is?

skeptizzle, 2019-09-03:
If that doesn't do the trick I will mask some of the sensor..

skeptizzle, 2019-09-03:
So i ended up not seeing any noticeable difference when changing sensitivity so i masked some of the sensor and that fixed the problem.
It's quite crazy the distance the sensor covers, but again, i mounted it around 2.5-2.7 meters above ground for aesthetically reasons.

closed 2019-09-03 by skeptizzle