dresden-elektronik/deconz-rest-plugin #1816: Aqara Wall switch QBKG04LM - How to enable decoupled mode (disconnect button from relay)

am1ter, 2019-09-03:
Hello everyone!

Could anybody please help me with disconnecting button from relay for this switch (Aqara QBKG04LM, 1 line, no neutral)? 

In other thread realted to 2 line switch Aqara QBKG12LM https://github.com/dresden-elektronik/deconz-rest-plugin/issues/1420#issuecomment-483057308 @ebaauw clarified that Aqara switches can work in decoupled mode and all you need to do is write new values (`0xFE`) in attributes `0xFF22` and `0xFF23`. Unfortunatelly, deconz don't show this parameters for my switch QBKG04LM with default general.xml, despite in this files there is relevant lines.

```
<attribute id="0xFF22" name="Xiaomi Disconnect 1" type="u8" showas="hex" access="rw" required="o" mfcode="0x115F">
<description>Set to 0x12 (0xFE) to connect (disconnect) the left button to (from) the relay.</description>
</attribute>
```

From this topic https://github.com/dresden-elektronik/deconz-rest-plugin/issues/1487#issuecomment-488347147 I discovered that my switch QBKG04LM uses another manufacturing code - `0x1037` instead of `0x115F`. When I changed `mfcode` to `0x137` in the file general.xml to correct value I discovered that disconnecting parameters appeared in deconz interface with a `0x00` value. But when I try to read or write a new value, for example `0xFE`, I get an error:

> failed UNSUP_MANUF_GENERAL_COMMAND

[image]

I read as many topics realted to my case as I could, but didn't find anything what could help me. Moreover I use Conbee II so I can't use ZSHARK for sniffing my xiaomi zigbee gateway and can't look out a command that disconnect switch button from relay.

I would really appreciate if anyone can give me advice how I can try to fix my problem.

ebaauw, 2019-09-03:
> @ebaauw clarified that Aqara switches can work in decoupled mode 

All I clarified is that one particular model Aqara switch can work in decoupled mode.  There’s no such thing as “Aqara switches”: every model works differently, and they seem to be using the same commercial model name for different technical models.

am1ter, 2019-09-03:
> All I clarified is that one particular model Aqara switch can work in decoupled mode. There’s no such thing as “Aqara switches”: every model works differently, and they seem to be using the same [...]

Thank you for your reply. In fact this Aqara switch can work in decoupled mode when it is connected to Xiaomi Aqara Gateway. This option works fine. But I can't find out a way to do the same thing using Deconz and Conbee II gateway.

ebaauw, 2019-09-03:
Just a long shot: Did you try the attribute definition without a manufacturer code?

am1ter, 2019-09-03:
> Just a long shot: Did you try the attribute definition without a manufacturer code?

Yes, I did. In a such case there is another error:

`writing failed`

[image]

ArkadiuszNiemiec, 2019-09-24:
I have the same problem. My switches ID is 0x1037 and I cannot find a way to enable decoupled mode...

ArkadiuszNiemiec, 2019-10-02:
I have bought a CC2531 dongle to snif a network. Can I connect a switch do Xiaomi HUB and sniff what it sends when I enable a decoupled mode?

am1ter, 2019-10-04:
> I have bought a CC2531 dongle to snif a network. Can I connect a switch do Xiaomi HUB and sniff what it sends when I enable a decoupled mode?

I think you could. I would really appreciate if you would upload your sniffed data.

ArkadiuszNiemiec, 2019-10-07:
> > I have bought a CC2531 dongle to snif a network. Can I connect a switch do Xiaomi HUB and sniff what it sends when I enable a decoupled mode?
> 
> I think you could. I would really appreciate if you would upload your sniffed data.

I will post it as soon as I have the hardware.

KapJI, 2019-10-20:
I have the same problem. I can confirm that this switch support decoupled mode by sending 0xFE to 0xFF22 and 0xFF23. I added support of this feature to zigbee2mqtt few months ago and it works fine: https://github.com/Koenkk/zigbee-herdsman-converters/pull/287
But deCONZ with Conbee 2 can't write this attribute for some reason.

KapJI, 2019-10-22:
After sniffing traffic and comparing with what zigbee2mqtt does I found that manufacturer code 0x115F should be used. Deconz detects it incorrectly and then uses it to match with extra attributes from general.xml
@ebaauw can you clarify how deconz detects mfcode?

KapJI, 2019-10-23:
zigbee2mqtt for example just has it hardcoded: https://github.com/KapJI/zigbee-shepherd-converters/blob/be0c869339da485842b2b56bf5cd370f22ba2b89/converters/toZigbee.js#L15

KapJI, 2019-10-23:
@am1ter @ArkadiuszNiemiec if you're still interested I created custom endpoint for rest plugin: https://github.com/KapJI/deconz-rest-plugin/tree/set-mfcode
It can write attribute with correct manufacturer code. Compile plugin, replace existing one. 
Then you can use it like:
```
curl -X POST http://127.0.0.1:8090/api/<api_key>/attributes/<id> -d '{"key": 65314, "value": 254}'
```
Use `<id>` from `/api/<api_key>/sensors`.
65314 for left button, 65315 for right button.
254 for decoupled mode, 18 or 34 for enabling relay control back.

ArkadiuszNiemiec, 2019-10-24:
@KapJI great job, thank you! I will be back in my flat in two weeks. I will update you here after I test it :+1:

am1ter, 2019-10-24:
@KapJI Thank you very much for your solution! I would really appreciate if you could give me some additional details. I compiled plugin and replaced existing, but I am not sure that I did it correctly.
A couple of questions:

1. I got an api key. When I send GET request to <api_key>/sensors I get a correct answer. But I am not sure what part of answer I should use as <id>. Below an example of answer:

```
 "2": {
        "config": {
            "on": true,
            "reachable": false,
            "temperature": 2400
        },
        "ep": 4,
        "etag": "26ed2d6fa261328a18f2deb36c289794",
        "manufacturername": "LUMI",
        "mode": 1,
        "modelid": "lumi.ctrl_neutral1",
        "name": "lumi.ctrl_neutral1",
        "state": {
            "buttonevent": 1002,
            "lastupdated": "2019-08-26T19:45:33"
        },
        "swversion": "11-25-2017",
        "type": "ZHASwitch",
        "uniqueid": "00:15:8d:00:02:4d:4f:cb-04-0006"
    }
```
Is my id for POST request is "2" or "00:15:8d:00:02:4d:4f:cb-04-0006"?

2. When I try to send a POST command 

> curl -X POST http://192.168.0.115:40850/api/XXXXXXXXXXX/attributes/2 -d '{"key": 65314, "value": 254}'

I get an error:

```
[
{"error":
{"address":"/attributes/2","description":"method, POST, not available for resource, attributes/2","type":4}
}
]
```

So I suppose that I made a mistake while compiling plugin. 
I would be grateful for any advice!

am1ter, 2019-10-29:
@KapJI, Thank you very much for your solution. Everything works perfect. My mistake was moving copiled plugin to incorrect docker container. When I copied it in the correct I got an opportunity to write attributes.

I updated first post with link to this solution.

KapJI, 2019-10-29:
@am1ter Great to hear!

ArkadiuszNiemiec, 2019-11-06:
@ebaauw you are the only contributor here, what do we need to do to merge @KapJI work to master?

ebaauw, 2019-11-06:
Technically, a pull request.

Not sure if it should be merged, though.  I think it should be a PUT instead of a POST.  Also, I don't like the hardcoded endpoint and mfcode.

You should be able to write the attribute using the CLI plugin (https://github.com/ma-ca/deconz-cli-plugin).

KapJI, 2019-11-13:
@ebaauw I think the proper fix will be using the correct mfcode in Phoscon itself. But as sources are closed, I can't submit pull request there :(

ebaauw, 2019-11-13:
The deCONZ core programme is closed source, but the _Cluster Info_ panel is configurable through `general.xml`, which is part of this repository.  That's how I added support for decoupled mode for the other Xiaomi switch.

> I believe there's hardcoded mapping somewhere between model id and mfcode.

No.  Per the ZigBee standard, the deCONZ core programme uses the _Manufacturer Code_ from the _Device Descriptor_ to determine which manufacturer-specific attributes (from `general.xml`)  to show in the GUI.

The issue with this Xiaomi switch is that it uses a different codes in the descriptor vs the _Basic_ cluster.  Working around this mutilation of the standard seems way beyond the current design and function of the core programme.  The values of the _Basic_ cluster attributes are used only by the REST API plugin, not by the core programme.

The "proper" fix would be for Xiaomi to release firmware that adheres to the standard.  I don't think that's likely to happen, so, unfortunately, we're stuck with "improper" workarounds.  Which, imho, is still preferable over not supporting devices that don't adhere to the standard.

ArkadiuszNiemiec, 2019-11-15:
> Which, imho, is still preferable over not supporting devices that don't adhere to the standard.

Yes, please! I have equipped my whole house with these switches with the intention to use some of them as wireless switches for smart lamps or bulbs. That's the reason why I have bought the ConBeeII and set up Deconz :smile: 

Ccan you suggest or help to develop the improper but acceptable workaround for this? I have found your PR with decoupled mode for another switch (#1444) and tried to follow it but it's still too hard for me to grasp.

ebaauw, 2019-11-15:
I don't see a workaround using the deCONZ GUI.  For the more technical people: send the _Write Attributes_ command using the deconz-cli-plugin (there you can specify mfcode yourself).

For the non-technical people, proper support by the REST API (and by its clients, like Phoscon) would be needed.  As you only set this once when commissioning the switch, I see no urge to do this.  Also, without access to a switch, development and testing would be very hard.

I suppose adding a `config.detached` REST attribute would be the proper way.  Then we run into the same issue as with `config.group`: different switch endpoints have different settings, but are combined in a single REST resource.  Best left to API v2.

ArkadiuszNiemiec, 2019-11-15:
> For the more technical people: send the _Write Attributes_ command using the deconz-cli-plugin (there you can specify mfcode yourself).

I have prepared a docker with `deconz-cli-plugin` and I am currently trying to write the attribute. I am not quite sure what is wrong so maybe you can help.
I am typing:
`zclattrmanu 0x6341 1 0x0000 0x115F 02FF2320FE`
where: `0x6341` is NWK (address), `1` is the ep, `0x0000` is the cluster`, `0x115F` is manufacturer code and then there is a command: `02` to write, `FF23` attribute identifier (16 bits according to docs), `20` is data type (u8) and `FE` is data to write.

Is that right? What is wrong? Thank you in advance!

ebaauw, 2019-11-16:
I think you need to specify the payload in Big Endian?  So `0223FF20FE`.

ArkadiuszNiemiec, 2019-11-16:
> I think you need to specify the payload in Big Endian? So `0223FF20FE`.

You are right, it worked! I have a decoupled switch now. Thank you and thanks @KapJI

exdee85, 2019-11-21:
Hello. I bought QBKG03LM switches and ran into the problem of adding. More precisely, a switch is added. At first it is added as 4 devices, but after restarting the gui, only two remain. But the sleepy problem is not that. How to implement a logic switch. One of the buttons should control the smart lamp. I tried to compile the plugin and set the parameter through it, succession reported, but at the same time, as the relay clicked, it continued to click.

SAOPP, 2020-03-06:
Guys, when Deconz start work fine with all the features of these switches? Why is it impossible to include the decoupled mode in the phoscon functionality? Why is it possible in s**t Z2M and not in Deconz?

SilverFire, 2020-04-11:
It works fine, check out the following post: https://github.com/dresden-elektronik/deconz-rest-plugin/issues/1420#issuecomment-483057308

Mimiix, 2020-06-06:
It seems this issue is resolved or otherwise inactive. If it is not, please re-open!

closed 2020-06-06 by Mimiix

50l3r, 2020-06-11:
> @am1ter @ArkadiuszNiemiec if you're still interested I created custom endpoint for rest plugin: https://github.com/KapJI/deconz-rest-plugin/tree/set-mfcode
> It can write attribute with correct [...]

How can i install  your version from windows 10.

This link is broken: https://www.dresden-elektronik.de/deconz/win/

chbndrhnns, 2020-06-12:
I’d like to understand decoupled mode better. Where do I find more information?

SAOPP, 2020-06-12:
What specifically interests you?

chbndrhnns, 2020-06-12:
- What are the implications and advantages of enabling the decoupled mode? I understand it would allow to connect switches to lights without phoscon? Is that right?
- Is there a way to find out which devices support it?
- given the hacks listed above that are required to enable it, are there risks I should be aware of?

ebaauw, 2020-06-12:
> I understand it would allow to connect switches to lights without phoscon? Is that right?

No.  It detaches the button(s) from the output line(s), so you can use them as wireless switch(es) without switching the wired light(s).

You can use any device without Phoscon - I never even use Phoscon.  It's just a browser-based front-end app, connecting to deCONZ over the REST API, see [deCONZ for Dummies](https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/deCONZ-for-Dummies).

To use wireless switches without deCONZ, they need to act as a controller, i.e. send commands directly to the lights (typically through a group).  Whether a switch can act as controller is determined in its firmware.  Xiaomi switches cannot; nothing we can do to change that.  You can recognise controllers in the API by the ZHASwitch `/sensors` resource exposing `config.group`.   And in the GUI by having (a) client (grey) _On/Off_ cluster(s).

50l3r, 2020-06-12:
@ebaauw could you throw some light to my problem? Im stuck at this point. Thanks in advance

SAOPP, 2020-06-12:
@chbndrhnns Just like Eric said. This is the decoupling of the button from the physical relay of the switch. Those your relay is constantly on (convenient for smart lighting), and the keys turn into logical ones with their own events, 1002, 2002 and 3002 - exactly the same as the Xiaomi wireless switch of the first revision, 3 actions. In this case, u can also control two relays via home automation system like Home Assistant and etc.

ebaauw, 2020-06-12:
@50l3r You should be able to set decoupled mode using the cli plugin or the rest plugin changes by @KapJI, see above.  In both cases you need to compile and install a plugin yourself.  I don't think compiling on Windows is supported.

50l3r, 2020-06-19:
> @50l3r You should be able to set decoupled mode using the cli plugin or the rest plugin changes by @KapJI, see above. In both cases you need to compile and install a plugin yourself. I don't think compiling on Windows is supported.

Thats very frustrating. I cant advance in this area. I have "no neutral" version and the lights blink constantly. Deconz cant detect my decoupled option and if i try to setup the switches with xiaomi gateway it cant integrate with other home devices.

do you know any alternative to the aqara wall switch? Maybe wireless version?

50l3r, 2020-06-19:
i can compile rest plugin inside home assistant? i host my home assistant in a proxmox vps.

If i access via vnc i view the gui:

[image]

Smanar, 2020-06-20:
>Thats very frustrating. I cant advance in this area. I have "no neutral" version and the lights blink constantly. Deconz cant detect my decoupled option and if i try to setup the switches with xiaomi gateway it cant integrate with other home devices.

But, this option can nothing for you, it s an electricial problem. Or you want to use only the switch and let the relay always "on" ?

BTW this option is on Cluster 0x0000 , attribute 0xff22 and 0xff23.

50l3r, 2020-06-21:
> > Thats very frustrating. I cant advance in this area. I have "no neutral" version and the lights blink constantly. Deconz cant detect my decoupled option and if i try to setup the switches with xiaomi gateway it cant integrate with other home devices.
> 
> But, this option can nothing for you, it s an electricial problem. Or you want to use only the switch and let the relay always "on" ?
> 
> BTW this option is on Cluster 0x0000 , attribute 0xff22 and 0xff23, not visible on yours, wich one deconz version have you ? The manufacter is 0x115f on your device too ?

Yes, i want to let relay always on and only use switch events with home assistant to execute actions like start vacuum or someting.

Thats my deconz version:
[image]

How can i get the manufacturer version? Maybe here?
[image]

rskhaira, 2020-08-15:
Is there any chance that somebody who knows how to do this could make a video while doing it.. Might be easier for us noobs to be a able to ape the steps.. Thanx

SAOPP, 2020-08-16:
[image]

Easy, open the basic cluster, read data, and open the channel 1 or 2, or maybe u got 1 way rocker, so then only 1 relay. Next double click on the channel:

[image]

...and type needed code for connect and disconnect releays. Read comments in the window. It's all.

intositeme, 2020-11-28:
Also interes

> Yes, i want to let relay always on and only use switch events with home assistant to execute actions like start vacuum or someting.

Yes also interested to find out if this is possible. Have a smart light connected to a aqara with neutral switch that has decoupled mode.

SAOPP, 2020-11-28:
@

> Also interes
> 
> > Yes, i want to let relay always on and only use switch events with home assistant to execute actions like start vacuum or someting.
> 
> Yes also interested to find out if this is possible. Have a smart light connected to a aqara with neutral switch that has decoupled mode.

?

Shurik85, 2021-05-07:
Good afternoon. I read a lot, but I never found an answer. There is Deconz and Homebridge Hue. In the Aqara QBKG12LM switch, the relay is decoupled from the button, while it is still possible to control the relay via the Phoscon interface. In HomeKit, the relay can also be controlled. But how can I now use the button as a condition for a logical action?

ReX1983, 2023-02-12:
Hello, I have a QBKG04LM but there is no Xiaomi Disconnect entry under the basic cluster (deconz v.2.20.1).
Any idea what I should do?

This is what I see:
[image]

Smanar, 2023-02-13:
This part only appear if the manufacture code is 0x1037, Can you check wich one you have ? your device can just don't support it.

ReX1983, 2023-02-13:
> This part only appear if the manufacture code is 0x1037, Can you check wich one you have ? your device can just don't support it.

I have just paired few more switches (until now connected to to the Xiaomi Gateway). These are all no neutral switches, some 1 rocket and some 2 rockets, they are all 0x1037, but none of them has the disconnect options.

Ideas?

Smanar, 2023-02-14:
There is something bad, the manufacture code is checked and valided because we can see the title "Man,ufacture specific" and "Xiaomi Specific" but all is empty ?

```
        <attribute-set id="0x4000" description="Manufacturer specific" mfcode="0x1037">
          <attribute id="0x4001" name="128-Bit security key" type="seckey" access="r" required="m" mfcode="0x1037"></attribute>
          <attribute id="0x4002" name="IEEE address" type="uid" access="rw" required="m" mfcode="0x1037"></attribute>
        </attribute-set>
        <attribute-set id="0xFF00" description="Xiaomi specific" mfcode="0x1037">
          <attribute id="0xFF0D" name="Xiaomi Sensitivity" type="u8" access="rw" required="o" mfcode="0x115f"></attribute>
          <attribute id="0xFF22" name="Xiaomi Disconnect 1" type="u8" showas="hex" access="rw" required="o" mfcode="0x115f">
            <description>Set to 0x12 (0xFE) to connect (disconnect) the left button to (from) the relay.</description>
          </attribute>
```

What is your deconz version ?

ReX1983, 2023-02-14:
I am on Deconz v.2.20.1.

Is it possible that my local deconz database / configuration is compromised?

These are devices that I paired years ago, when I was on a much older deconz version. Deconz at the time didn't work so well with these devices, so I had to remove them from deconz and repair to the Xiaomi hub. 

Few days ago I tried to pair them back to deconz and to my surprise the friendly name I gave them is still there. So it seems that the deconz config/database hasn't been properly cleaned when I removed them years ago.

Smanar, 2023-02-14:
>Is it possible that my local deconz database / configuration is compromised?

No I don't think this feature just work with deconz itself and the xml file (updated at every deconz update), nothing to see with the database.
I will ask in dev channel, realy no idea from my side.

ReX1983, 2023-02-14:
> > Is it possible that my local deconz database / configuration is compromised?
> 
> No I don't think this feature just work with deconz itself and the xml file (updated at every deconz update), nothing to see with the database. I will ask in dev channel, realy no idea from my side.

Is this logic hardcoded or in an xml file? If it is a file, could you give me the full path and filename so I check mine?

Smanar, 2023-02-15:
it's an XML file, on classic OS it's /usr/share/deCONZ/zcl/general.xml

ReX1983, 2023-02-15:
> it's an XML file, on classic OS it's /usr/share/deCONZ/zcl/general.xml

I can see the same path under preferences, but unfortunately, I don't have access to it, as the installation is managed by Home Assistant in a docker container.

Is there a way to re-pair the node and ensure that deCONZ doesn't reuse settings previously used?

As I mentioned earlier, I think I might have brought older settings when I restored an old installation using the backup/restore feature in Phoscon.

ReX1983, 2023-02-16:
Further update...

I did a "Reset Node" (F5) on each one of these switches and now the Xiaomi Disconnect attributes are displayed under the Basic Cluster.

I was able to read and set the values and I can confirm that the buttons are now in "disconnected" mode.

In Home Assistant I can still control the relay but there is no event when the buttons are pressed.

Following Home Assistant [instructions](https://www.home-assistant.io/integrations/deconz/), I tried to listen to "deconz_event" but nothing is captured there.

When I press the button I see the following the deCONZ container log:
> ZCL attribute report 0x00158D0001F4864F for cluster: 0x0006, ep: 0x05, frame control: 0x18, mfcode: 0x0000

This is the right button of a double rocker no neutral switch.

Any idea on how to capture the event?

Should I see these disconnected buttons reported as sensors in Phoscon (I don't)?

Smanar, 2023-02-16:
In disconnect mode, you need to have 2 devices, the "light" one and a "sensor" one the ZHASwitch.
You haven't a value in the state/buttonevent field ?

(If you can see it in HA, you have access to the API in phoscon/help/API Information/sensor/.. )

ReX1983, 2023-02-16:
I don't see any sensor in Phoscon, under API information. The only sensors are the one from other devices, not these switches. 

For each two rockets Aqara switch, I still see two lights and one switch, under the API.

Is there any sort of re-pair/refresh I should do in deCONZ so Phoscon updates the way the devices are exposed?

Smanar, 2023-02-16:
There is a DDF now for thoses device https://github.com/dresden-elektronik/deconz-rest-plugin/blob/master/devices/xiaomi/xiaomi_qbkg04lm_switch.json

One can have be included before the DDF was here, so with legacy code.

```
            "name": "Switch 84",
            "state": {
                "buttonevent": null,
                "lastupdated": "none"
            },
            "type": "ZHASwitch",
            "uniqueid": "00:15:8d:00:01:f4:86:4f-04-0006"
```

But they are not working buttonevent=none.
This value need to be updated whe using it, else you need to see in logs with "info_l2" some information about a button map.

ReX1983, 2023-02-16:
I am not sure I follow. What should I do?

[16 more replies]