dresden-elektronik/deconz-rest-plugin #1817: Eurotronic Spirit ZigBee

rkotulan, 2019-09-03:
My setup:
Home Assistant with ConBee II (2.05.67)
I connect to VNC and was able to pair with Eurotronic Spirit ZigBee.

[image]

But I do not know hot show in Phoscon App or directly in HA?

Any suggestion please?

5mauggy, 2019-09-04:
Hi rkotulan,

For now it is not possible to see the thermostat in Phoscon web app.
In HA, with DeCONZ integration, it should be available right away like any other sensor. The thermostat card in Lovelace wasn't working as expected with this thermostat but I just fixed this bug. The fix is committed to dev branch, but not yet on the master branch.
Regards,

rkotulan, 2019-09-04:
Hi 5mauggy,

thank you for replay. I can not see thermostat in HA.
In VNC I can see
[image]
Is it possible to rename entity in VNC?

In HA I have integration with DeCONZ
[image]
But I can not see the thermostat.

What can I miss?

Thank you,
Rudolf

rkotulan, 2019-09-04:
@5mauggy when do you expect that your fix will be in master branch? Winter is coming :)

5mauggy, 2019-09-04:
Hi @rkotulan,
I couldn't rename the thermostat in deCONZ app. But you can rename it with a simple API call. To do so, you can either use a REST Client (like Tabbed Postman Chrome extension) or a command line tool like cURL.
Just have a look at REST API documentation http://dresden-elektronik.github.io/deconz-rest-doc/getting_started/ everything is well explained there.
Once you have your API key, get a list of all sensors by running a GET request to /`api/<apikey>/sensors`. From the response read your thermostat id. Then run a PUT request to `/api/<apikey>/sensors/<id>` with the the following data `{ "name" : "Custom Name" }`.
The cURL command would be something like this:
`curl -X PUT -H "Content-Type: application/json" -d '{"name":"Custom name"}' http://localhost:8080/api/01234abc56/sensors/4`

I am sorry I can't help you much with seeing the thermostat in HA. I am new to home automation, I only started to play with deCONZ and HA two weeks ago and all my sensors appeared in HA without doing anything special. I am using [this](https://hub.docker.com/r/homeassistant/home-assistant/) docker container for HA.

I have no idea when my fix will be merged into master branch. Anyway, there is only one file (
[homeassistant/components/deconz/climate.py](https://github.com/home-assistant/home-assistant/commit/62338dd28ec2f328892353eed234f45411e99ffc#diff-c486cf6d4c7ca60be0d4d286e9e07307)) to edit to apply the fix on your HA installation.

rkotulan, 2019-09-04:
Hi @5mauggy 
thank you so much. 
First, in the beginning, my entity does not show in API. But I tried to remove and add again and after some testing (tried to read and write Device enabled attribute) thermostat appeared in API and in HA.

5mauggy, 2019-09-04:
I am glad I could help you sort this.
Regards,

closed 2019-09-05 by rkotulan