dresden-elektronik/deconz-rest-plugin #1818: How to update?

Michaelnorge, 2019-09-03:
Sorry for this maybe stupid question, but how I can update my Phoscon App?
I´m using Raspberry Headless together with ioBroker.

I´m using that tutorial: [https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Update-deCONZ-manually](url) 
But with the link in the description [https://www.dresden-elektronik.de/rpi/deconz-firmware/?C=M;O=D](url) I just become v. 2.05.64, but I see 2.05.67 on that site: [https://github.com/dresden-elektronik/deconz-rest-plugin/releases](url)

How I can get an update on my version?
I bought a couple of Trådfri motion sensors and there is a fix at this version.

Greetings and thanks for help
Michael

Smanar, 2019-09-03:
I m not sure, but theses link are only to update firwmare ?
Take a look here > https://phoscon.de/en/conbee2/install

Smanar, 2019-09-04:
I don't know ioBroker, so I don't know the better way to update your installation but
- To update Phosocn, you just need to update deconz, it's not in the firwmare
- There isn't a new firmware for all deconz version
- Yes the 2.05.67 is the last version, but It seem the Ikea Tradfri motion bug is not include in it > https://github.com/dresden-elektronik/deconz-rest-plugin/issues/1676 lasts posts.

Michaelnorge, 2019-09-04:
I asked Dresden Elektronik for some help and I got fixed an update to 2.05.67.
And you are right, the bug is still not fixed in this version.
So - still need to wait a bit!

Here is a copy of the answer from Dresden Electronik, but in german:

vielen Dank für Ihre Nachricht an den dresden elektronik support.

Der Link zeigt zur Firmware des ConBee, deren Update sollte immer in der Phoscon App angeboten werden. Hier liegen die Files von deCONZ. Damit kann deCONZ manuell geupdatet werden:
wget https://www.dresden-elektronik.de/rpi/deconz/beta/deconz-latest-beta.deb
sudo dpkg -i deconz-latest-beta.deb
Der empfohlene und auf der Phoscon Seite beschriebene Weg ist über das Beta-APT Repository:
https://www.phoscon.de/de/conbee2/install#raspbian

Mit freundlichen Grüßen

Andreas Preuß
Support

closed 2019-09-05 by Michaelnorge