dresden-elektronik/deconz-rest-plugin #1821: Deleted 3rd party apps re-appear after reboot

mplogas, 2019-09-04:
**Describe the bug**
When deleting 3rd-party apps through the REST API (using Hue Essentials or Postman) by  doing a DELETE on ```<ip>/api/<myid>/config/whitelist/<userid to delete>``` I receive a ```200 OK [ { "success": { "username": "<userid to delete>" } } ]```

To confirm, I re-issue the DELETE command and receive a ```200 OK []``` and double-check with a GET on ```<ip>/api/<myid>/config```, with the entries being purged from the whitelist

However, upon rebooting through Phoscon, all deleted 3rd-party-apps are back in the whitelist and removed 3rd-party apps can access deconz.

**To Reproduce**
1. Add two 3rd-party apps, one for Postman, the other one for testing
2. check the whitelist for both apps
3. delete the testapp using Postman
4. re-issue the delete command, an empty array should be returned
5. check the whitelist for the testapp-uid (should not be listet anymore)
6. reboot deconz through Phoscon
7. check the whitelist again, all deleted 3rd-party apps are back

**Expected behavior**
I would have expected a deleted 3rd-party app to be deleted and stay deleted. Re-appearing after reboot is weird and unexpected

**Additional context**
I stumbled upon this [previously](https://github.com/diyhue/diyHue/issues/167) but thought, that's an issue of DIYHue. However today, when trying to integrate Deconz into Alexa I figured out that this seems to be a Deconz/Phoscon bug instead.

weltmeyer, 2019-10-08:
i can confirm: i deleted an app whitelist key with

DELETE http://10.0.XXX.XXX/api/XXX/config/ and get a success message.

then i restart the gateway and it is back..

mplogas, 2019-12-05:
Hello @manup @ChrisHae @ebaauw. Are there any updates on this? Can you reproduce the issue? How do I "force delete" the deleted API keys?

mplogas, 2020-01-02:
I moved to Debian Buster on a MacMini (with a ConBee2) and the issue is gone. It is still reproducable with the provided Raspi-Image. So if you're running deConz on a RasPi, please be aware of this potential security issue.

closed 2020-05-09 by stale[bot]