dresden-elektronik/deconz-rest-plugin #1822: Support not answering for 11 days | Conbee II + IKEA LED Power Supply with wireless receiver

reinisb, 2019-09-04:
Hi,

I sent an email to support@dresden-elektronik.de 11 days ago but I still have not received an answer. Is this normal? I have posted the contents of the email below, I hope someone can answer. Thanks!

--- 

Hi,

I have had a raspberry pi running homebridge for a year or so - I use multiple homebridge plugins to drive different smart home devices. Now, I recently moved into a new rental apartment, which has IKEA kitchen lights powered by "IKEA LED Power Supply with wireless receiver" (please see attached image). I can control those lights via the IKEA TRÅDFRI 5-button wireless remote control.

Now, as far as I understand ikea uses zigbee hence it would be great to add zigbee support to my raspberry pi & homebridge so that I could control also those IKEA kitchen lights via homebridge. I am about to buy the "Conbee II", but before I do that I would appreciate if you could please verify that it would work with my IKEA kitchen lights setup since that is not fully clear from the description.

[image]

bastshoes, 2019-09-04:
Yes it works. See https://github.com/dresden-elektronik/deconz-rest-plugin/issues/466

hellcry37, 2019-09-05:
I think that mail is a black hole, i also sent 2 mails there more then 30 days ago and no answer

reinisb, 2019-09-05:
@bastshoes - thanks for the info. What would it take to add it to the supported devices page? https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Supported-Devices

@hellcry37 - yeah, that's crazy. I mean, they are definitely losing sales because of this. If @bastshoes would not have answered that it works, I would not have bought the Conbee II.

bastshoes, 2019-09-05:
@reinisb you can creat PR to add device to wiki

Smanar, 2019-09-06:
No need to create a PR, everyone can edit the wiki.

hellcry37, 2019-09-07:
@Smanar I know maybe is hard for them, I am sympathetic to the issues they have but:
1. I bought a product from them and now the things that worked did not work, I have no answer from them
2. Sent a mail more then 30 days ago to buy some sensors from them, according to their site where is clearly stated "contact us if you want to buy this product", I did that nothing happen til now

Hope they will get out from this crisis sooner rather then later.

P.S. you want my 2 cents?
Dedicate all resources you have to fix all bugs on all supported devices, then add more support for more devices, and get a manager so the left hand know what right one is doing.
example: If a guy integrates something and brakes an old supported device like "samjin multi sensor" then I am not seeing this as a good thing, because surprise I was using that.

closed 2020-01-12 by stale[bot]