dresden-elektronik/deconz-rest-plugin #1826: Please support for Philips Hue bulb firmwareoption “Last Known State”  (in latest Hue Bulb firmware)

carver7, 2019-09-06:


carver7, 2019-09-06:
The best feature of Hue bulbs compared to others is the Last Known State option in the latest Hue firmware. 
Please support this nice native feature!

- When the power goes out and comes back on in the middle of the night, lights do not come back on and wake everyone up.
- Lights accidentally physically switched off retain their off/on state and their color and brightness settings.
- Lights switched on with a Hue switch can be set to come back on at their previous settings or one can choose to override Last Known State with a programmed setting
- There is no safety issue whatsoever because the Hue bulbs will be forced to turn on if one turns the lamp on and off twice within ~30 seconds.

Thank you

ebaauw, 2019-09-06:
See #1014

carver7, 2019-09-13:
> See #1014

Thanks!

jokerigno, 2020-01-01:
@ùcan you enlight me? I Can't find in the post what to do!

carver7, 2020-01-30:
@jokerigno 
As described in #1014 , it’s not formally supported and it will require a ton of work to fully do so.

carver7, 2020-01-30:
@ebaauw @manup 
Potential workaround for configuring power on behavior of Hue bulbs that might be practical enough for users to implement who have a Hue bridge, and if the DeCONZ touchlink feature can be modified to mimic the ‘steal a bulb’/force pair feature that IKEA Tradfri remote can do.

When doing some experimenting last fall, I noticed that after custom configuring the power-on behavior of a Hue bulb using the Hue phone app and bridge, I could then pair this bulb to the Tradfri bridge while preserving my non-default power-on configuration by specifically doing the following:

- Leave this configured Hue bulb paired to the Hue bridge
- Then force-pair it to the Tradfri bridge using a Tradfri remote (by following the “steal” a bulb method which In part involved holding the remote with 1cm or so of the bulb, which I noticed physically resembles the deCONZ touchlink procedure)

Sure enough, after turning the lamp with the Hue bulb screwed into it off and then on again one time (I.e simulate a power outage and restoration) this force-paired-to-Tradfri Hue bulb remained off as desired.

I was not able to do the same with deCONZ touchlink feature though, so I suppose deCONZ’s touchlink only works on an unpaired bulb (which obviously will have reverted modified power-on behavior to the factory defaults)

Would this workaround (modify the deCONZ touchlink feature to support force-pairing a bulb already paired to another Zigbee radio) be practically feasible to implement in deCONZ as an alternative to the large amount work required to formally do so until presumably it might be in deCONZ api V2 is developed?

(Referencing #1014 )

ebaauw, 2020-01-30:
deCONZ doesn’t support touch-link pairing, only searching and resetting.  I never looked into touch linking, but I would imagine adding support would be more elaborate than supporting the power-on settings through the API.

If you want to change the power-on settings for a larger number of lights, best use the deconz-cli-plugin.  See #1055.

Mimiix, 2020-06-05:
It seems this issue is inactive / resolved. If it is not, please re-open or create another issue!

closed 2020-06-05 by Mimiix