dresden-elektronik/deconz-rest-plugin #1827: Battery levels as own entites

embeddeddev, 2019-09-06:
Unfortunalety the battery levels of devices like Philips motion sensor and Xiaomi/Aqara ambience, door/window and vibration sensors are modelled as sub-attributes of their sensor values like temperature and humidity so they aren't directly accessible as entities in Home Assistant.

Contrary ZHA component of Home Assistent directly exports these battery levels as own entities.

In addition current implementation provides the values redundantly but in fact its only one unique device attribute which should be exported as such one.

Looking forward to receive a reply.

erikced, 2019-09-06:
How to expose battery levels in HA should probably be discussed [pydeconz](https://github.com/Kane610/deconz) instead of here.

embeddeddev, 2019-09-06:
Thx for the hint!

ebaauw, 2019-09-06:
I’m all in favour of exposing the battery (or rather: the _Power Configuration_ cluster 0x0001) as a separate resource, but I’m not sure whether we should do that before API v2.

Technically we could introduce a `ZHABattery` sensor type, exposing the battery as `state.battery` with a corresponding `state.lastupdated` (a long-time wish of mine).  It would also enable us to expose the battery for devices that are currently exposed a `/lights` resources (like the IKEA FYRTUR).  On the downside, this would almost double the amount of `/sensors` resources.

@manup, what domyou think?

embeddeddev, 2019-09-06:
Hi @ebaauw, thx for your reply and your support!

Indeed it double's the number of sensors for single sensor devices, but in reality it IS a sensor w/ important information. ZHA already grabbed this issue and provides battery levels for all battery based sensors (even if no _Power Configuration_ cluster visible in deCONZ GUI!?)

Related to @erikced's reply I opened an [mirror issue on pydeconz](https://github.com/Kane610/deconz/issues/42). Now I am really unsure to which this topic has to be associated to due to my lack of technical internals.

ebaauw, 2019-09-06:
Exposing it thru the REST API is one thing, actually using it in apps or plugins to home automation systems is another.  That needs to be discussed in their respective repositories.  We’d keep the current `config.battery` around, at least for a while for backwards compatibility, and probably forever in Hue compatibility mode.

closed 2020-01-11 by stale[bot]