dresden-elektronik/deconz-rest-plugin #1831: Looking for a reliable router for my xiaomi aqara motion sensors

pergolafabio, 2019-09-07:
Hi,

my mesh is verry easy, i only have end devices to turn on my lights
My end devices are xiaomi pir sensors and ikea remotes
the conbee stick is on my server on the 2th floor, so to cover my house i have placed on every floor an Osram smart socket plug
but after reading on internet and blogs, seems the osrams are not verry reliable, so i want to replace them with other ones to try out
can someone give me some advice for some smart plugs? they need to be EU based

if you look my mesh, its quite easy : 
https://www.dropbox.com/s/n6knob7rr35q65j/mesh-xiaomi.JPG?dl=0
but one sensor, the one on groundfloot needs to go to 2 x osrams plus, and i think thats why, it isnt working all the time...
i tried already changing wifi signals, also on my accesspoints, doesnt help
xiaomi_sensors_1 in the picture is the one thats bugging me in 20% of the time

i now, have those ones : https://encrypted-tbn0.gstatic.com/images?q=tbn:ANd9GcTG5Dh3oickP23aBZKNsfVMoacdpnEo4WLxsfp5-u6hJXJ1pGorJA

as you can see they are EU based

i was thinking to buy Xiaomi ones, because they should work with xiamo sensors :) , but cant find EU editions, and dont wanna work with adapters between it

thnx in advance

Smanar, 2019-09-07:
Hu, I realy don't know good plugs, the more used are ikea and osram, and I see bad return alls days with thems, perhaps the innr one. Better router are bulbs.

There is a special device DIY, that work fine as router > https://www.zigbee2mqtt.io/how_tos/how_to_create_a_cc2530_router.html
This router works fine, but you can have some problem with it, for exemple deconz don't display connection for him, so impossible to see the mesh, you can find some return on this github with this device.

Smanar, 2019-09-07:
I m interested too by this question.

Seriously, no-one have good return ?

pergolafabio, 2019-09-07:
Thnx but don't wanna go the road of a 2530... Still want to use the socket were I plug it in.
That's why Osram is/was a good choice, also because of the hardware button on it to turn it of/on... That button is missing on the Ikea version...

Smanar, 2019-09-07:
Oups, sorry, I have edited my post, just before you, to avoid 2 time the same issue ^^.

bastshoes, 2019-09-07:
I have this one https://lifecontrol.ru/devices/smart-plug/

After I've added 3 of them my network improved.

pergolafabio, 2019-09-07:
Thnx , appreciate feedback... Are they also on supported wiki list? Do they also report power usage?

pergolafabio, 2019-09-07:
@basthoes, is it this one?  

https://github.com/dresden-elektronik/deconz-rest-plugin/issues/1541

pergolafabio, 2019-09-07:
Hmmz, seems not so easy to buy, bit able to find a commercial webshop like Amazon or something ;)

Oh, and btw, that button on it, is that a power switch? Can you turn it on/off localy? Or is it the pair button?

bastshoes, 2019-09-08:
@pergalofabio This plug is not in wiki yet. It's reports voltage, amperage, Active power, power consumption. Button on case act's as switch for local operation and for pairing. Lifecontrol is local brand. But device it's self produced in China. I think there should be white label devices same as this one.

pergolafabio, 2019-09-08:
Aha , that looks the perfect solution...
Now I need to find a shop... Seems to be difficult ;)

Where did you buy them from?

bastshoes, 2019-09-08:
Locally in Russia on their site.

pergolafabio, 2019-09-09:
seems difficult to get
i am going to try the ikea smart plugs first, gonnan replace the osrams by them

anyone with experience? ikea smart plugs as router for the xiaomi motion pir sensors?

pergolafabio, 2019-09-21:
Ok, got myself some Ikea outlets, much better then osrams, all green lines instead of orange, seems distance is better... After few days testing, my xiaomi pir detected motion every time

closed 2019-09-21 by pergolafabio

Smanar, 2019-09-26:
Nice, thx.
So you can confirm to me ikea outlet are working fine ?

pergolafabio, 2019-09-26:
yes, 2 weeks i use, now my xiaomi respond 100% 
in my first reply , you see orange/yellow lines, now placed ikea plugs on exact same place, all GREEN ;)