dresden-elektronik/deconz-rest-plugin #1832: RICI01 Log flooding

bastshoes, 2019-09-08:
Log flooded with following records

```
11:04:44:086 ZCL attribute report 0x00124B0009FE31E2 for cluster 0x0B04, ep 0x01

11:04:44:280 no button map for: RICI01 ep: 0x01 cl: 0x0B04 cmd: 0x0A pl[0]: 005

11:04:44:280 ZCL attribute report 0x00124B0009FE31BB for cluster 0x0B04, ep 0x01

11:04:44:544 no button map for: RICI01 ep: 0x01 cl: 0x0B04 cmd: 0x0A pl[0]: 00B

11:04:44:544 ZCL attribute report 0x00124B0009FDF62F for cluster 0x0B04, ep 0x01

11:04:44:548 no button map for: RICI01 ep: 0x01 cl: 0x0B04 cmd: 0x0A pl[0]: 005

11:04:44:548 ZCL attribute report 0x00124B0009FDF62F for cluster 0x0B04, ep 0x01

11:04:45:085 no button map for: RICI01 ep: 0x01 cl: 0x0B04 cmd: 0x0A pl[0]: 005

11:04:45:085 ZCL attribute report 0x00124B0009FE31E2 for cluster 0x0B04, ep 0x01

11:04:45:279 no button map for: RICI01 ep: 0x01 cl: 0x0B04 cmd: 0x0A pl[0]: 005

11:04:45:279 ZCL attribute report 0x00124B0009FE31BB for cluster 0x0B04, ep 0x01

11:04:45:538 no button map for: RICI01 ep: 0x01 cl: 0x0B04 cmd: 0x0A pl[0]: 005
```

exdee85, 2019-09-09:
I have the same problem.

bastshoes, 2019-09-09:
@manup @ebaauw why do smart plug Electrical Measurements reports treated as button events?

ebaauw, 2019-09-09:
They're not - that's basically what the message is saying.

bastshoes, 2019-09-09:
Ok, is it because of rerouting interval?

bastshoes, 2019-09-09:
@ebaauw sorry I didn't get.
First line state that there is no button map. Is it mean that system expects some button event? 
Second line is actual measurement values report.

ebaauw, 2019-09-09:
No, it means that it checks whether a map has been configured for the device to create buttonevents.  It hasn't, so no buttonevents are created.

bastshoes, 2019-09-09:
Thank you very much for clarification.
Just last question. When configuring report interval via GUI, is it in seconds?

ebaauw, 2019-09-09:
Yes

closed 2019-09-09 by bastshoes