dresden-elektronik/deconz-rest-plugin #1833: Is there a was to Update Osram Ledvance Downlights TW?

lons81, 2019-09-08:
Hello

How can i Update my Ledvance Downlights 25W to a newer Firmware?
In every Switch on the go to cold white with 100% and dont remember the last state.

Thanks, Marcel

ebaauw, 2019-09-08:
> How can i Update my Ledvance Downlights 25W to a newer Firmware?

Afaik OSRAM/Ledvance haven't published their firmware, so I'm afraid the only answer is: through their gateway.

> In every Switch on the go to cold white with 100% and dont remember the last state.

Unlike Hue, OSRAM use a manufacturer-specific cluster for the power-on settings, see #102.  They have been for as long as I remember (over several years), so I doubt you'd actually need a firmware update for this functionality.

Unlikely that REST API will support this anytime soon, but the `general.xml` from the above commit adds support to the deCONZ GUI:
[image]

Copy `general.xml` to `/usr/share/deCONZ/zcl` and restart deCONZ.  You should then see the updates in the GUI.  Set the bulbs brightness and colour temperature to the desired values and press the button to store them as new power-on defaults.  I don't think they support powering-on the lights as off.  Their app used to demand a minimal brightness, but I don't think their lights actually enforce this.

I have only tested this on an OSRAM Classic B40 TW - LIGHTIFY with firmware V1.05.10.

lrybak, 2019-12-04:
Hi Eric (@ebaauw),
I've copy the general.xml file to the /usr/share/deCONZ/zcl but see no change in GUI. Still no power-on Defaults option. Any hint?

[image]

ebaauw, 2019-12-04:
You need to select the 0xfc0f cluster.

lrybak, 2019-12-04:
Where it is?

lrybak, 2019-12-04:
Forgive me, I was blind. Now I found it, thx.

closed 2020-04-09 by stale[bot]

7wells, 2020-10-06:
Please consider also this post over there:
https://github.com/dresden-elektronik/phoscon-app-beta/issues/213#issuecomment-701205883