dresden-elektronik/deconz-rest-plugin #1834: Device support: ROBB Smarrt insert dimmer

Klumper, 2019-09-08:
# ROBB Smarrt Dimmer

I've tried to connect my new zigbee dimmer to deCONZ and integrate it within HomeAssistant, which looks alot like the [iCasa ICZB-IW11D](https://github.com/dresden-elektronik/deconz-rest-plugin/issues/1124). The connected dimmer does not show up within the Phoscon App but is visible with vnc.

It seems like my other sensors are connected to this dimmer and not directly to the Conbee II stick I use, which is why they don't work properly. The '/lights' api endpoint also returns an empty object.

According to the [manual](https://en.robbshop.nl/amfilerating/file/download/file_id/824/), the device supports the following:
**Input Clusters**
* 0x0000: Basic
* 0x0003: Identify
* 0x0004: Groups
* 0x0005: Scenes
* 0x0006: On/off
* 0x0008: Level Control
* 0x0b05: Diagnostics

**Output Clusters**
*  0x0019: OTA

## Device information
Name: HUE insert dimmer Zigbee ROBB Smarrt
SKU: ROB_200-004-0
EAN: 7439647744702
[Bought here](https://en.robbshop.nl/build-in-dimmer-hue-zigbee-robb)
[English manual](https://en.robbshop.nl/amfilerating/file/download/file_id/824/)
[Dutch manual](https://en.robbshop.nl/amfilerating/file/download/file_id/894/)

## Cluster Info
[image]
## Node Info
[image]

ebaauw, 2019-09-08:
deCONZ hasn't read the descriptors.  Try to select _Read Node Descriptor_ and _Read Simple Descriptor(s)_ from the dropdown menu.  The second dropdown menu should appear, with the endpoints and clysters.

Next, open the network from the old web app, or search for devices from Phoscon.  Then in the _Node Info_ panel, read the _Basic_ cluster attributes.  That should trigger the creation of the `/lights` resource.

Smanar, 2019-09-08:
"Next, open the network from the old web app, or search for devices from Phoscon. Then in the Node Info panel, read the Basic cluster attributes. That should trigger the creation of the /lights resource."

BTW, was an old question I have.
If I want to test the inclusion in the API, without de-paring/re-pairing a device. I can just delete it from the API, and use this trick to re-create them ?

Klumper, 2019-09-08:
@ebaauw Thanks for the quick reply! Looks like the _Read Node Descriptor_ remains checked but the _Read Simple Descriptor(s)_ option unchecks instantly. I've tried searching for new devices after selecting the _Read Node Descriptor_ option within the old web app without result.

ebaauw, 2019-09-08:
@Klumper 
You might need to toggle the squares a couple of times, a few seconds apart.  Did the second dropdown appear?  You need to read the _Basic_ cluster attributes to trigger the creation of the resource.

@Smanar 
Yes.  Sometimes not all resources are deleted, though.  In that case, I typically remove them from the database using sqlitebrowser.  Make sure to close deCONZ and make a backup before hacking the database.

closed 2020-01-13 by stale[bot]