dresden-elektronik/deconz-rest-plugin #1835: Ledvance Lamps Work in Phoscon but dont get find in Hue Binding from Openhab

lons81, 2019-09-08:
Hello

Every Lampe & Plug works with Phoscon.
(Hue lamps, Osram Plug, Gledopto driver & Osram Ledvance)
But in Openhab the Ledvance where mit discovered...
All others lamps & plugs will be discovered by Hue Binding.
Is there a other protocol or something else that Openhab cant find the Ledvance Downlights?
I have 10 pieces of them and all of them doesnt come in Openhab.

Thanks, Marcel

lons81, 2019-09-10:
Hello

I posted this problem also in OpenHab board and here is a quote from "**cweitkamp**"

> The “type”: “Color temperature light” is important for discovery of devices by the Hue binding. As you can see the LEDVANCE bulbs return “type”: “Color dimmable light” which will not recognized as valid by the Hue binding. I am not sure if that type is correct. Maybe you should post your findings into your related deCONZ issue and ask for verification over there.

Therefore the output from /lights/:

**Osram Ledvance:**
"etag": "9883a8119c818ac46c095a292868329d",
"hascolor": true,
"manufacturername": "LEDVANCE",
"modelid": "Down Light TW",
"name": "Wohnzimmer Couch",
"state":{"alert": "none", "bri": 254, "colormode": "ct", "ct": 370,…},
"swversion": "00102002",
"type": "Color dimmable light",
"uniqueid": "d5:3a:00:b8:d1:f0:00:00-01"

And this is a **Philips lamp** which is find trough OpenHab:
“ctmax”: 454,
“ctmin”: 153,
“etag”: “5bf6eff2f879f06e73692eec543a07b9”,
“hascolor”: true,
“manufacturername”: “Philips”,
“modelid”: “LTC015”,
“name”: “Esszimmer Licht”,
“state”:{“alert”: “none”, “bri”: 254, “colormode”: “ct”, “ct”: 366,…},
“swversion”: “1.46.13_r26312”,
“type”: “Color temperature light”,
“uniqueid”: “00:17:88:01:03:37:1b:c2-0b”

Is the `Color dimmable light` correct?
The Osram lamp is the same like Philips also dimmable & color temperature light.

Thanks alot, Marcel

manup, 2019-09-10:
The type is directly derived from the light descriptor. But the type `Color temperature light` would be more appropriate for the TW version. We can correct this statically.

closed 2020-01-17 by stale[bot]