dresden-elektronik/deconz-rest-plugin #1836: Xioami motion sensor go to unreachable after 5 mn.

Smanar, 2019-09-08:
Hello, an user have a curious bug with Xiaomi motion detector, log will be explicit enought.

```
2019-09-08 13:19:15.058 (Zigbee) ### WebSocket Data : {'e': 'changed', 'id': '9', 'r': 'sensors', 'state': {'lastupdated': '2019-09-08T11:19:15', 'presence': False}, 't': 'event', 'uniqueid': '00:15:8d:00:02:73:26:9d-01-0406'}
2019-09-08 13:22:20.080 (Zigbee) ### WebSocket Data : {'config': {'battery': 100, 'duration': 90, 'on': True, 'reachable': False, 'temperature': 2600}, 'e': 'changed', 'id': '9', 'r': 'sensors', 't': 'event', 'uniqueid': '00:15:8d:00:02:73:26:9d-01-0406'}
```

He have the same bug, on all this kind of device, and only them, other Xiaomi sensor are not impacted.

The sensor works fine, detect something, have is state to "true", then "false" and only 3mn after "unreachable".

He have deconz V 2.05.67 / Firmware 26490700 / Conbee II / nuc with ubuntu server 18.04.3 / Headless

Smanar, 2019-09-09:
For more explanations, he was previously on a raspberry, and have migrated to a NUC, I think ha have just make a backup then a restore on the new machine.

BTW, how deconz decide a batterie device is unreacheable ? it's just no request during a specific period (and I think this period is superior at 5 mn) ?

Smanar, 2019-09-11:
No one have tips/advice/explanation ?
I realy don't have idea.
- Only the same type of device, and all of them.
- What can cause an "unreachable" state on Xiaomi device (after 5mn) ?

Smanar, 2019-09-13:
In my situation, the device is working, he just go in "unreachable state" after 5 mn.

BobPuts, 2019-09-13:
Got the same. After a few minutes it doesn't notice any motion anymore.

merdok, 2019-09-26:
Got the same problem with Aqara Motion sensors. I need to delete them and re-add them every time this happens to make them work for a short time again. I find this weird since they are marked as supported in the compatibility list but clearly there are still some issues...
Did you maybe found a solution for this problem?

Smanar, 2019-09-26:
Nope, after 13 dayx, no responses.
In this situation, the device is still working, he haven't problem with it, just the sensor is marked as "unreachable".

On my side , I don't need a solution, it's not critic, but I will be happy just with an response, for me the problem is from deconz itself, because I don't see what other thing can decide this kind of state after a fixed delay.

closed 2020-01-31 by stale[bot]

ebaauw, 2020-03-15:
`reachable` is utterly useless and best ignored, see #2590.