dresden-elektronik/deconz-rest-plugin #1839: Aqara MagicCube

kugelkopf123, 2019-09-08:
Nice work with the Update .69 and integration of the MagicCube. 
But there is one more action that you can do. Throw up in the air and catch it. Gives Buttonevent "7008"

Fl0ca, 2019-09-08:
The cube is working with the switcheditor? The state is changing on the config page (where you can give it a devicename), but with a function applied in the Editor nothing happens ?! 

Don't know wheres the problem :(

manup, 2019-09-09:
>Nice work with the Update .69 and integration of the MagicCube.
But there is one more action that you can do. Throw up in the air and catch it. Gives Buttonevent "7008"

Thanks, we've just seen the drop/throw gesture in this  https://www.youtube.com/watch?v=LeuHpBdwmag video from DrZzs :) Looking forward to add the gesture to the Phoscon App switch editor soon.

>The cube is working with the switcheditor?

Yes it's in 2.05.69, a short overview is given in the release notes:
https://github.com/dresden-elektronik/deconz-rest-plugin/releases/tag/V2_05_69

>he state is changing on the config page (where you can give it a devicename), but with a function applied in the Editor nothing happens ?!

>Don't know wheres the problem :(

In previous releases the Xiaomi Cube was sometimes not paired completely (with two sensor resources). In such cases you would need to repair the cube. You can see that the cube is fully paired when the switch details show both, the button event as well as the rotation angle.

closed 2019-09-09 by kugelkopf123