dresden-elektronik/deconz-rest-plugin #1841: REST-API hue value not accepted

Knapoc, 2019-09-09:
I'm not sure if this is a duplicate of #973 / 7. Sometimes when the hue value is set over the REST-API, deconz is reporting a success, but different values are set for hue.
I first encountered this issues in HomeAssistant ([a short video of the behaviour](https://giphy.com/gifs/Yo8pz0Hbi2bAnxiGwc/html5)). The logs report the following: https://pastebin.com/raw/x2DRJpyS

Issuing a similar command directly, results in the same behaviour.

PUT request:
```
{
  "on": true,
  "bri": 146,
  "hue": 38922,
  "ct": 153,
  "sat": 254,
  "transitiontime": 10
}
```

Response:
```
[
    {
        "success": {
            "/lights/3/state/on": true
        }
    },
    {
        "success": {
            "/lights/3/state/bri": 146
        }
    },
    {
        "success": {
            "/lights/3/state/hue": 38922
        }
    },
    {
        "success": {
            "/lights/3/state/sat": 254
        }
    },
    {
        "success": {
            "/lights/3/state/ct": 153
        }
    }
]
```

GET request:
```
{
    "ctmax": 500,
    "ctmin": 153,
    "etag": "48fcb8b5e494a28badcf7a360899856a",
    "hascolor": true,
    "manufacturername": "Philips",
    "modelid": "LCT003",
    "name": "Hue Spot 1",
    "state": {
        "alert": "none",
        "bri": 146,
        "colormode": "ct",
        "ct": 153,
        "effect": "none",
        "hue": 34180,
        "on": true,
        "reachable": true,
        "sat": 254,
        "xy": [
            0.3124,
            0.3273
        ]
    },
    "swversion": "5.127.1.26420",
    "type": "Extended color light",
    "uniqueid": "a"
}
```

Within deconz, the bulb reports the following color capabilities:

- Data Type: 16-bit bitmap
- Access: read only
- Hue saturation :white_check_mark:
- Enhanced Hue saturation :white_check_mark:
- Color loop :white_check_mark:
- CIE 1931 XY :white_check_mark:
- Color temperature :white_check_mark:

ebaauw, 2019-09-09:
You can only specify one colour per request and `ct` trumps `hue`/`sat`.

Knapoc, 2019-09-09:
Ah that explains it. Thank you.

closed 2019-09-09 by Knapoc