dresden-elektronik/deconz-rest-plugin #1844: Xiaomi switches running all lights

pidiman, 2019-09-09:
Hello coders,

I have 3 xiaomi switches and every switch control one light. 
Everything works fine but from one day when I push one of them then every lights are on(toggle).. 

Any helps ? Thanks a lot.

Smanar, 2019-09-10:
Suddenly ?
Are you using phoscon automation, or an other application for that ?

On my side, if I had a small installation, I would take a look in groups in the API, if I had a too big one,  I would delete 1 inter, and re-do it, to see if the bug persist.

pidiman, 2019-09-10:
I use Home Assistant automations.

Smanar, 2019-09-10:
So make a tries with HASS turned off, if the issue is still here it's a deconz problem (take a look in group) else take a look in your HASS scripts.

pidiman, 2019-09-10:
I have combination : light is sonoff tasmota relay and switch is xiaomi switch deconz. If I turn off Hass my automation doesnt work :)

Smanar, 2019-09-10:
Ha you are using Batterie switch ? Not the one that can control light ?
So it will be easy to check where the problem is from, I m sure somewhere you can find websocket log from deconz in HASS.
Check if you have 1 notification when you press a switch or 3, or a grouped notification.

For switch it's something like that
`WebSocket Data : {'r': 'sensors', 'uniqueid': '00:15:8d:00:02:f0:8c:12-01-0012', 'state': {'buttonevent': 1002, 'lastupdated': '2019-09-10T19:04:55'}, 'id': '12', 't': 'event', 'e': 'changed'}`

pidiman, 2019-09-10:
This is my log

```
2019-09-10 21:07:46 DEBUG (MainThread) [pydeconz.websocket] Websocket data: {"e":"changed","id":"5","r":"sensors","state":{"buttonevent":1000,"lastupdated":"2019-09-10T19:07:46"},"t":"event","uniqueid":"00:15:8d:00:02:3d:0b:fc-01-0006"}
2019-09-10 21:07:46 DEBUG (MainThread) [pydeconz.websocket] Websocket data: {"e":"changed","id":"5","r":"sensors","state":{"buttonevent":1002,"lastupdated":"2019-09-10T19:07:46"},"t":"event","uniqueid":"00:15:8d:00:02:3d:0b:fc-01-0006"}
2019-09-10 21:07:46 INFO (MainThread) [homeassistant.components.automation] Executing Posteľ led - wall switch - Zapni/Vypni
2019-09-10 21:07:46 INFO (MainThread) [homeassistant.helpers.script] Script Posteľ led - wall switch - Zapni/Vypni: Running script
2019-09-10 21:07:46 INFO (MainThread) [homeassistant.helpers.script] Script Posteľ led - wall switch - Zapni/Vypni: Executing step call service
2019-09-10 21:07:46 INFO (MainThread) [homeassistant.components.automation] Executing Kupelna led - on/off left switch
2019-09-10 21:07:46 INFO (MainThread) [homeassistant.helpers.script] Script Kupelna led - on/off left switch: Running script
2019-09-10 21:07:46 INFO (MainThread) [homeassistant.helpers.script] Script Kupelna led - on/off left switch: Executing step call service
2019-09-10 21:07:46 INFO (MainThread) [homeassistant.components.automation] Executing Zapni/Vypni kuchynské led prepínačom
2019-09-10 21:07:46 INFO (MainThread) [homeassistant.helpers.script] Script Zapni/Vypni kuchynské led prepínačom: Executing step call service

```
I dont understand why but this switch number 5 control only one automation. The last one " Zapni/Vypni kuchynske led prepinacom" ...

Smanar, 2019-09-11:
Yep, deconz only send notice for the device ID = 5, so can't be the guilty.
Your sonoff use MQTT ? so you can too take a look in mqtt log.

closed 2020-01-16 by stale[bot]