dresden-elektronik/deconz-rest-plugin #1846: Xiaomi Magic Cube not showing up in Switch Editor

andjandre, 2019-09-10:
I updated to the latest snapshot but my Magic Cube isn't showing up in the Switch Editor. I restarted DeconZ and Phoscon and repared my Magic Cube a couple times but it still isn't appearing in Switch Editor. I can still use it within Homekit with Homebridge. Odd thing is that I cannot pair it within the "Add Switch" dialogue but with the "Add Sensor" functionality. It then is pairing successful and showing up within switches.

Looking forward for your held because i was really looking forward for this feature.

wilfredsmit, 2019-09-14:
I second that, although I have a Xiaomi Mi Magic Cube. And I also misses the 'analog'/rotation sensor in the REST Api.

wilfredsmit, 2019-09-14:
I've managed to get my Mi Magic Cube in the Switch Editor. I tried to search for the sensor a second time: Add new switch and push the reset button on the cube for two to five seconds (blue led on). In the mean time move and rotate the cube. In the REST api I know see two sensors: sensors/42 (flip, etc.) and sensors/46 (rotate).

KennethLavrsen, 2019-11-05:
I have added the Magic Cube 5 times now and hammered rotated moved. It shows up. But how am I supposed to get a Switch Editor to show? What is the user interface?

Smanar, 2019-11-05:
Hello,
Home page > select a group > edit > switch editor.

https://www.phoscon.de/en/app/doc

KennethLavrsen, 2019-11-05:
Thanks @Smanar - I had never found that without your help

closed 2020-03-11 by stale[bot]