dresden-elektronik/deconz-rest-plugin #1847: Ikea fyrtur Hassio

Thoit, 2019-09-10:
Hello,

I've bought some Ikea Fyrtur blinds and helped in the beginning sending some codes to got it working (see: https://github.com/dresden-elektronik/deconz-rest-plugin/issues/1121). I'm still trying to connect the blinds to the Conbee stick to use it in Hassio (Home Assistant). I've updated the Gateway to the newest version and updated Home Assistant to the newest version. I'm trying to connect the blinds by pressing the two buttons on the blinds for 5 seconds and click the Add Lights button in Deconz. After spending several retries it's not possible to connect.

When will there be support in Hassio for the blinds are can someone help me to connect them?

SigNoget, 2019-09-12:
Hello,

I've been able to connect my blinds to the Conbee II stick by first pressing Add Lights (so the search started) and afterwards pressing the 2 buttons on the blinds for at least 5 seconds. They appeared almost immediately. 

Even though the reversed order shouldn't be the issue - it is worth a try though :) 

The blinds/covers appeared in hassio afterwards but for now I am unable to control them, unfortunately.

Revorge, 2019-09-13:
Weird, as far as I can tell you are doing things correct. I've got everything working perfectly...

Try deleting the blinds from the GUI, pressing "Add lights" in Phoscon and then resetting the blind by holding both buttons. They should appear as a light in Phoscon within a minute and after that in hassio under developer tools -> states as cover.xxxx

Thoit, 2019-09-16:
The problem is solved. My add on wasnt upgraded anymore so I was using an old version of Deconz. After some struggles with the move to the original add on I was able to connect the blinds to the gui after removing them in VNC. The blinds were visible after a reboot of home assistant. Then the up and down buttons weren't working. I've connected through VNC again and read the data a couple of times. Then rebooted again and is all working now. Now I'm a happy man! Thanks for supporting!

closed 2019-09-16 by Thoit