dresden-elektronik/deconz-rest-plugin #1849: Lost control of all lamps and plugs but sensor work

kses, 2019-09-10:
Hi!

Thank you for a great product! It has been rock solid until now.

Running Conbee (version 1) with Deconz in a docker container on a NUC running Ubuntu.

It appears I have lost control over all lamps (ikea) and all pluggs (ikea). Sensor and buttons sometimes work and sometimes not (ikea and xiaomi). Running deconz in a docker container. sw:2.05.69 fw:26320500.

I have recently done an update drive which might be the source of the problem. Have tried re-downloading the image. Restarted it. Rebooted host. Plugged/unpluggged the conbee drive. Reset to previous network setting according to issue #1226. Tried to check with touch old web ui for a few lamps which show correct id and channel.

The gui sometimes shows the mesh lines and sometimes not. I can not control any lights from phoscon or HA. Some lights show as available some does not.

I am out of ideas of what to try next except a complete reset and restart from the beginning. All ideas appreciated!

[image]
[image]
[image]
[image]

Xieldan, 2019-09-11:
Same here. Since I updated to 2.05.69, I lost control of nearly all my devices (Tradfri bulbs, Osram gardenspots, gardenpole and lightstrip, Osram Smart Plugs). In deconz, only a few nodes are connected.  Will probably have to reset the whole system. :-( Running deConz on Buster on a Rpi4, WiringPi library is installed.

manup, 2019-09-11:
>Will probably have to reset the whole system

NO! Please don't do that this almost never needed.

Can you provide a few details on your system:
* RaspBee, ConBee or ConBee II
* Which firmware version is installed
* If ConBee (I/II) do you use an USB extension cable?

kses, 2019-09-11:
I have conbee I on a USB extension. About 2m
I tried downgrading to 2.05.64, does not help.

manup, 2019-09-11:
Ok so the next check should be if there are any suspicious changes in the network parameters.
For that please check the Zigbee configuration page as described here:

https://github.com/dresden-elektronik/deconz-rest-plugin/releases/tag/V2_05_59

Which firmware version is shown on the Settings > Gateway page?

kses, 2019-09-11:
fw:26320500
I tried "downgrading" the network parameters yesterday. Did not help. No indication of change in the table either. Attaching screenshot

[image]

Xieldan, 2019-09-11:
I am using Raspbee, firmware is 26330500. This is what I see in the configuration screen:
[image]

Thanks for looking into it!

kses, 2019-09-13:
Do you need any logs or something? I am out of ideas but will try to get whatever you need to figure this out.

Xieldan, 2019-09-13:
Don’t know what happened, but my system is back now and everything meshed. I did absolutely nothing, not even a reboot of the Raspi. 🤔

kses, 2019-09-13:
The only strange thing I noticed is the new version message. Could this indicate a issue?

[image]

Smanar, 2019-09-14:
Can be an indication, for my problem an user have installed the V 69 but have displayed in phoscon V67

![https://easydomoticz.com/forum/download/file.php?id=6263)
![https://easydomoticz.com/forum/download/file.php?id=6262)

manup, 2019-09-15:
>The only strange thing I noticed is the new version message. Could this indicate a issue?

The version message is displayed since the installed firmware 0x26320500 is not the latest version 0x26330500.

A small test could you please power-cycle one of the greyed out IKEA lights and check if it comes back?

kses, 2019-09-15:
Hi!

Since I can not shut off zigbee lights at the moment via home assistant I have had to do it manually. Some show up in phoscon even if they are unpowered, those that are on usually show the latest thing I tried to set them to, but sometimes revert to actual state or grey. If I power cycle or turn an unpowered device on something usually change. I can not control any of them though. Plugs or light even if they show up.

The firmware, I have tried to update it in the old web ui since a update button does not show in phoscon. But it does not seem to work

Attaching current GUI and phoscon view.

[image]

The 2 floor lights are unpowered at the moment...
[image]

hellcry37, 2019-09-16:
It happen to me once also on ConBee II v2.05.69 (HA installed in docker / ubuntu) To solve it I had to restart HA deconz adddon.

kses, 2019-09-17:
> It happen to me once also on ConBee II v2.05.69 (HA installed in docker / ubuntu) To solve it I had to restart HA deCONZ add-on.

Thank you, but it does not work. I have tested restarting host, container, downgrading, recreating, it has no effect.

@manup are there any cached or temporary files that could cause this that do not gets cleared? Or hardware error? I can receive some messages since I sometimes get a status and sometimes get messages from sensor but can not send any commands to lights.

FezVrasta, 2019-09-17:
I think I'm having this very same issue...
https://github.com/dresden-elektronik/deconz-rest-plugin/issues/1876

kses, 2019-09-17:
I realized I have not attched any logs. Please see below.
The GW refuse to update firmware(through old web ui). Is it possible to do i manually?

`11:13:06:704 no button map for: lumi.weather ep: 0x01 cl: 0x0403 cmd: 0x0A pl[0]: 000,
11:13:06:704 ZCL attribute report 0x00158D00036BD85C for cluster 0x0403, ep 0x01,
11:13:11:773 discovery for zombie 0xd0cf5efffec7e088 dropped, last try was 109 s ago,
11:13:12:252 discovery for zombie 0x000d6ffffe9f8659 dropped, last try was 341 s ago,
11:13:12:732 discovery for zombie 0x000d6ffffe9c3047 dropped, last try was 231 s ago,
11:13:13:212 discovery for zombie 0xd0cf5efffeb4251b dropped, last try was 584 s ago,
11:13:18:054 GW firmware version: 0x26320500,
11:13:18:054 GW firmware version shall be updated to: 0x26330500,
11:13:27:006 no button map for: lumi.weather ep: 0x01 cl: 0x0402 cmd: 0x0A pl[0]: 000,
11:13:27:006 ZCL attribute report 0x00158D00036BD85C for cluster 0x0402, ep 0x01,
11:13:27:038 no button map for: lumi.weather ep: 0x01 cl: 0x0405 cmd: 0x0A pl[0]: 000,
11:13:27:038 ZCL attribute report 0x00158D00036BD85C for cluster 0x0405, ep 0x01,
11:13:27:070 no button map for: lumi.weather ep: 0x01 cl: 0x0403 cmd: 0x0A pl[0]: 000,
11:13:27:070 ZCL attribute report 0x00158D00036BD85C for cluster 0x0403, ep 0x01,
11:13:30:011 discovery for zombie 0xd0cf5efffec7e088 dropped, last try was 127 s ago,
11:13:30:492 discovery for zombie 0x000d6ffffe9f8659 dropped, last try was 359 s ago,
11:13:30:972 discovery for zombie 0x000d6ffffe9c3047 dropped, last try was 250 s ago,
11:13:31:453 discovery for zombie 0xd0cf5efffeb4251b dropped, last try was 602 s ago,
11:13:41:110 Current channel 15,
11:13:41:142 Device TTL 5940 s flags: 0x7,
11:13:48:253 discovery for zombie 0xd0cf5efffec7e088 dropped, last try was 145 s ago,
11:13:48:733 discovery for zombie 0x000d6ffffe9f8659 dropped, last try was 377 s ago,
11:13:49:213 discovery for zombie 0x000d6ffffe9c3047 dropped, last try was 268 s ago,
11:13:49:693 discovery for zombie 0xd0cf5efffeb4251b dropped, last try was 620 s ago,
11:14:06:493 discovery for zombie 0xd0cf5efffec7e088 dropped, last try was 164 s ago,
11:14:06:973 discovery for zombie 0x000d6ffffe9f8659 dropped, last try was 396 s ago,
11:14:07:453 discovery for zombie 0x000d6ffffe9c3047 dropped, last try was 286 s ago,
11:14:07:932 discovery for zombie 0xd0cf5efffeb4251b dropped, last try was 638 s ago,
11:14:17:536 no button map for: lumi.weather ep: 0x01 cl: 0x0402 cmd: 0x0A pl[0]: 000,
11:14:17:536 ZCL attribute report 0x00158D00036BD85C for cluster 0x0402, ep 0x01,
11:14:17:567 no button map for: lumi.weather ep: 0x01 cl: 0x0405 cmd: 0x0A pl[0]: 000,
11:14:17:568 ZCL attribute report 0x00158D00036BD85C for cluster 0x0405, ep 0x01,
11:14:17:599 no button map for: lumi.weather ep: 0x01 cl: 0x0403 cmd: 0x0A pl[0]: 000,
11:14:17:599 ZCL attribute report 0x00158D00036BD85C for cluster 0x0403, ep 0x01,
11:14:24:733 discovery for zombie 0xd0cf5efffec7e088 dropped, last try was 182 s ago,
11:14:25:213 discovery for zombie 0x000d6ffffe9f8659 dropped, last try was 414 s ago,
11:14:25:692 discovery for zombie 0x000d6ffffe9c3047 dropped, last try was 304 s ago,
11:14:26:173 discovery for zombie 0xd0cf5efffeb4251b dropped, last try was 657 s ago,
11:14:37:917 no button map for: lumi.weather ep: 0x01 cl: 0x0402 cmd: 0x0A pl[0]: 000,
11:14:37:917 ZCL attribute report 0x00158D00036BD85C for cluster 0x0402, ep 0x01,
11:14:37:949 no button map for: lumi.weather ep: 0x01 cl: 0x0405 cmd: 0x0A pl[0]: 000,
11:14:37:949 ZCL attribute report 0x00158D00036BD85C for cluster 0x0405, ep 0x01,
11:14:37:980 no button map for: lumi.weather ep: 0x01 cl: 0x0403 cmd: 0x0A pl[0]: 000,
11:14:37:981 ZCL attribute report 0x00158D00036BD85C for cluster 0x0403, ep 0x01,
11:14:41:101 Current channel 15,
11:14:41:133 Device TTL 5880 s flags: 0x7,
11:14:42:972 discovery for zombie 0xd0cf5efffec7e088 dropped, last try was 200 s ago,
11:14:43:453 discovery for zombie 0x000d6ffffe9f8659 dropped, last try was 432 s ago,
11:14:43:933 discovery for zombie 0x000d6ffffe9c3047 dropped, last try was 323 s ago,
11:14:44:412 discovery for zombie 0xd0cf5efffeb4251b dropped, last try was 675 s ago,
11:15:01:213 discovery for zombie 0xd0cf5efffec7e088 dropped, last try was 218 s ago,
11:15:01:693 discovery for zombie 0x000d6ffffe9f8659 dropped, last try was 450 s ago,
11:15:02:173 discovery for zombie 0x000d6ffffe9c3047 dropped, last try was 341 s ago,
11:15:02:653 discovery for zombie 0xd0cf5efffeb4251b dropped, last try was 693 s ago,
11:15:19:453 discovery for zombie 0xd0cf5efffec7e088 dropped, last try was 237 s ago,
11:15:19:932 discovery for zombie 0x000d6ffffe9f8659 dropped, last try was 468 s ago,
11:15:20:412 discovery for zombie 0x000d6ffffe9c3047 dropped, last try was 359 s ago,
11:15:20:892 discovery for zombie 0xd0cf5efffeb4251b dropped, last try was 711 s ago,
11:15:28:052 GW firmware version: 0x26320500,
11:15:28:053 GW firmware version shall be updated to: 0x26330500,
11:15:37:692 discovery for zombie 0xd0cf5efffec7e088 dropped, last try was 255 s ago,
11:15:38:173 discovery for zombie 0x000d6ffffe9f8659 dropped, last try was 487 s ago,
11:15:38:653 discovery for zombie 0x000d6ffffe9c3047 dropped, last try was 377 s ago,
11:15:41:110 Current channel 15,
11:15:41:142 Device TTL 5820 s flags: 0x7,
11:15:41:287 Announced to internet,
11:15:55:933 discovery for zombie 0xd0cf5efffec7e088 dropped, last try was 273 s ago,
[... 8 lines cut ...]
[... cut ...]
11:16:51:612 discovery for zombie 0x000d6ffffe9c3047 dropped, last try was 450 s ago,
11:16:52:093 discovery for zombie 0xd0cf5efffeb4251b dropped, last try was 72 s ago,
11:17:08:892 discovery for zombie 0xd0cf5efffec7e088 dropped, last try was 346 s ago,
11:17:09:372 discovery for zombie 0x000d6ffffe9f8659 dropped, last try was 578 s ago,
11:17:09:853 discovery for zombie 0x000d6ffffe9c3047 dropped, last try was 468 s ago,
11:17:10:332 discovery for zombie 0xd0cf5efffeb4251b dropped, last try was 91 s ago,`

kses, 2019-09-17:
Is there some issue with the firmware updater? I get `GW firmware update exit code 255` when I try to update from old web ui.

`19:04:32:540 wait reconnect 11 seconds,
19:04:33:540 wait reconnect 10 seconds,
19:04:34:540 wait reconnect 9 seconds,
19:04:35:540 wait reconnect 8 seconds,
19:04:36:540 wait reconnect 7 seconds,
19:04:37:540 wait reconnect 6 seconds,
19:04:38:540 wait reconnect 5 seconds,
19:04:39:540 wait reconnect 4 seconds,
19:04:40:540 wait reconnect 3 seconds,
19:04:41:540 wait reconnect 2 seconds,
19:04:42:540 wait reconnect 1 seconds,
19:04:42:615 Device firmware version 0x26320500,
19:04:42:630 unlocked max nodes: 200,
19:04:42:915 Device protocol version: 0x010A,
19:04:43:107 Current channel 15,
19:04:43:171 CTRL ANT_CTRL 0x02,
19:04:43:283 Device protocol version: 0x010A,
19:04:43:475 Current channel 15,
19:04:43:539 CTRL ANT_CTRL 0x02,
19:04:43:542 binding for cluster 0x0000 of 0xD0CF5EFFFE7B5784 exists (verified by reporting),
19:04:43:543 skip configure report for cluster: 0x0000 attr: 0x4000 of node 0xD0CF5EFFFE7B5784 (seems to be active),
19:04:43:813 ZCL attribute report 0xD0CF5EFFFEB4251B for cluster 0x0000, ep 0x01,
19:04:43:844 ZCL attribute report 0xD0CF5EFFFEB4251B for cluster 0x0000, ep 0x01,
19:04:44:900 0x000D6FFFFE3BB925 error APSDE-DATA.confirm: 0xE1 on task,
19:04:46:540 binding for cluster 0x0000 of 0xD0CF5EFFFEB4251B exists (verified by reporting),
19:04:46:541 skip configure report for cluster: 0x0000 attr: 0x4000 of node 0xD0CF5EFFFEB4251B (seems to be active),
19:04:46:662 0xD0CF5EFFFEB4251B error APSDE-DATA.confirm: 0xE1 on task,
19:04:46:694 0xD0CF5EFFFE7B5784 error APSDE-DATA.confirm: 0xE1 on task,
19:04:48:503 0xD0CF5EFFFEC7E088 error APSDE-DATA.confirm: 0xE1 on task,
19:04:49:054 don't close database yet, keep open for 900 seconds,
19:04:49:688 0x000D6FFFFE2207C1 error APSDE-DATA.confirm: 0xE1 on task,
19:04:51:077 GW firmware version: 0x26320500,
19:04:51:077 GW firmware version shall be updated to: 0x26330500,
19:04:51:529 0x000D6FFFFE9C3047 error APSDE-DATA.confirm: 0xE1 on task,
19:04:53:338 0x000D6FFFFE9F8659 error APSDE-DATA.confirm: 0xE1 on task,
19:04:55:131 0x000D6FFFFE408337 error APSDE-DATA.confirm: 0xE1 on task,
19:04:56:940 0x000D6FFFFE3BB925 error APSDE-DATA.confirm: 0xE1 on task,
19:04:59:101 Current channel 15,
19:04:59:133 Device TTL 4308 s flags: 0x7,
19:05:03:729 0xD0CF5EFFFEB4251B error APSDE-DATA.confirm: 0xA7 on task,
19:05:10:277 0xD0CF5EFFFE7B5784 error APSDE-DATA.confirm: 0xA7 on task,
19:05:13:833 max transmit errors for node 0xD0CF5EFFFE7B5784, last seen by neighbors 309 s,
19:05:16:490 0xD0CF5EFFFEB4251B error APSDE-DATA.confirm: 0xA7 on task,
19:05:22:653 0x000D6FFFFE9C3047 error APSDE-DATA.confirm: 0xE1 on task,
19:05:22:750 0xD0CF5EFFFE7B5784 error APSDE-DATA.confirm: 0xA7 on task,
19:05:22:750 max transmit errors for node 0xD0CF5EFFFE7B5784, last seen by neighbors 318 s,
19:05:24:511 0xD0CF5EFFFEC7E088 error APSDE-DATA.confirm: 0xE1 on task,
19:05:26:336 0x000D6FFFFE2207C1 error APSDE-DATA.confirm: 0xE1 on task,
19:05:28:113 0x000D6FFFFE9C3047 error APSDE-DATA.confirm: 0xE1 on task,
19:05:29:938 0x000D6FFFFE9F8659 error APSDE-DATA.confirm: 0xE1 on task,
19:05:31:732 0x000D6FFFFE408337 error APSDE-DATA.confirm: 0xE1 on task,
19:05:33:493 0x000D6FFFFE3BB925 error APSDE-DATA.confirm: 0xE1 on task,
19:05:38:136 max transmit errors for node 0xD0CF5EFFFEB4251B, last seen by neighbors 14 s,
19:05:39:961 0xD0CF5EFFFEB4251B error APSDE-DATA.confirm: 0xA7 on task,
19:05:39:961 max transmit errors for node 0xD0CF5EFFFEB4251B, last seen by neighbors 16 s,
19:05:45:997 0xD0CF5EFFFE7B5784 error APSDE-DATA.confirm: 0xA7 on task,
19:05:45:997 max transmit errors for node 0xD0CF5EFFFE7B5784, last seen by neighbors 342 s,
19:05:52:257 0xD0CF5EFFFEB4251B error APSDE-DATA.confirm: 0xA7 on task,
19:05:52:258 max transmit errors for node 0xD0CF5EFFFEB4251B, last seen by neighbors 10 s,
19:05:58:694 0xD0CF5EFFFE7B5784 error APSDE-DATA.confirm: 0xA7 on task,
19:05:58:694 max transmit errors for node 0xD0CF5EFFFE7B5784, last seen by neighbors 354 s,
19:05:59:110 Current channel 15,
19:05:59:142 Device TTL 4248 s flags: 0x7,
19:06:00:535 0xD0CF5EFFFEC7E088 error APSDE-DATA.confirm: 0xE1 on task,
19:06:02:345 0x000D6FFFFE2207C1 error APSDE-DATA.confirm: 0xE1 on task,
19:06:08:845 0x000D6FFFFE9C3047 error APSDE-DATA.confirm: 0xA7 on task,
19:06:10:638 0x000D6FFFFE9F8659 error APSDE-DATA.confirm: 0xE1 on task,
19:06:12:383 0x000D6FFFFE408337 error APSDE-DATA.confirm: 0xE1 on task,
[... 107 lines cut ...]
19:07:49:540 wait reconnect 8 seconds,
19:07:50:540 wait reconnect 7 seconds,
19:07:51:540 wait reconnect 6 seconds,
19:07:52:540 wait reconnect 5 seconds,
19:07:53:539 wait reconnect 4 seconds,
19:07:54:540 wait reconnect 3 seconds,
19:07:55:540 wait reconnect 2 seconds,
19:07:56:540 wait reconnect 1 seconds,
19:07:56:607 Device firmware version 0x26320500,
19:07:56:629 unlocked max nodes: 200,
19:07:56:918 Device protocol version: 0x010A,
19:07:57:110 Current channel 15,
19:07:57:174 CTRL ANT_CTRL 0x02,
19:07:57:286 Device protocol version: 0x010A,
19:07:57:478 Current channel 15,
19:07:57:542 CTRL ANT_CTRL 0x02,
19:07:58:823 0x000D6FFFFE3BB925 error APSDE-DATA.confirm: 0xE1 on task,
19:07:59:111 Current channel 15,
19:07:59:143 Device TTL 4128 s flags: 0x7,
19:08:05:053 GW firmware version: 0x26320500,
19:08:05:053 GW firmware version shall be updated to: 0x26330500,
19:08:05:388 0xD0CF5EFFFEB4251B error APSDE-DATA.confirm: 0xA7 on task,
19:08:05:388 max transmit errors for node 0xD0CF5EFFFEB4251B, last seen by neighbors 106 s,
19:08:07:213 0xD0CF5EFFFE7B5784 error APSDE-DATA.confirm: 0xE1 on task,
19:08:09:055 0xD0CF5EFFFEC7E088 error APSDE-DATA.confirm: 0xE1 on task,`

kses, 2019-09-17:
I updated the old Docker manual way, but it thought I also had a Conbee 2 connected...

[image]

kses, 2019-09-17:
With the new FW I get a grayer view in phoscon (all lights are grey). If I restart a bulb it appears for a while but I can not control it. (It does not respond)

This is the log for when the tv light gets restarted.

`19:39:15:564 discovery for zombie 0x000d6ffffe9c3047 dropped, last try was 323 s ago,
19:39:16:045 discovery for zombie 0xd0cf5efffeb4251b dropped, last try was 436 s ago,
19:39:23:147 LightNode 4: TV light updated,
19:39:23:148 DeviceAnnce of LightNode: 0xd0cf5efffec7e088 Permit Join: 0,
19:39:23:244 ZDP discovery done in 96 ms19:39:24:596 0xD0CF5EFFFEC7E088 error APSDE-DATA.confirm: 0xE1 on task,
19:39:26:325 0xD0CF5EFFFEC7E088 error APSDE-DATA.confirm: 0xE1 on task,
19:39:28:053 don't close database yet, keep open for 900 seconds,
19:39:28:126 0xD0CF5EFFFEC7E088 error APSDE-DATA.confirm: 0xE1 on task,
19:39:29:903 0xD0CF5EFFFEC7E088 error APSDE-DATA.confirm: 0xE1 on task,
19:39:31:744 0xD0CF5EFFFEC7E088 error APSDE-DATA.confirm: 0xE1 on task,
19:39:33:521 0xD0CF5EFFFEC7E088 error APSDE-DATA.confirm: 0xE1 on task,
19:39:35:331 0xD0CF5EFFFEC7E088 error APSDE-DATA.confirm: 0xE1 on task,
19:39:36:684 discovery for zombie 0x000d6ffffe9f8659 dropped, last try was 115 s ago,
19:39:36:764 reuse dead link (dead link container size now 4),
19:39:37:132 0xD0CF5EFFFEC7E088 error APSDE-DATA.confirm: 0xE1 on task,
19:39:37:165 discovery for zombie 0x000d6ffffe9c3047 dropped, last try was 345 s ago,
19:39:37:644 discovery for zombie 0xd0cf5efffeb4251b dropped, last try was 457 s ago,
19:39:38:941 0xD0CF5EFFFEC7E088 error APSDE-DATA.confirm: 0xE1 on task,
19:39:40:718 0xD0CF5EFFFEC7E088 error APSDE-DATA.confirm: 0xE1 on task,
19:39:42:536 0xD0CF5EFFFEC7E088 error APSDE-DATA.confirm: 0xE1 on task,
19:39:44:329 0xD0CF5EFFFEC7E088 error APSDE-DATA.confirm: 0xE1 on task,
19:39:46:098 0xD0CF5EFFFEC7E088 error APSDE-DATA.confirm: 0xE1 on task,
19:39:46:741 delay sending request 121 dt 0 ms to 0xD0CF5EFFFEC7E088, cluster 0x0006,
19:39:46:840 delay sending request 121 dt 0 ms to 0xD0CF5EFFFEC7E088, cluster 0x0006,
19:39:46:940 delay sending request 121 dt 0 ms to 0xD0CF5EFFFEC7E088, cluster 0x0006,
19:39:47:041 delay sending request 121 dt 1 ms to 0xD0CF5EFFFEC7E088, cluster 0x0006,
19:39:47:140 delay sending request 121 dt 1 ms to 0xD0CF5EFFFEC7E088, cluster 0x0006,
19:39:47:240 delay sending request 121 dt 1 ms to 0xD0CF5EFFFEC7E088, cluster 0x0006,
19:39:47:340 delay sending request 121 dt 1 ms to 0xD0CF5EFFFEC7E088, cluster 0x0006,
19:39:47:355 0xD0CF5EFFFEC7E088 error APSDE-DATA.confirm: 0xE1 on task,
19:39:48:428 0xD0CF5EFFFEC7E088 error APSDE-DATA.confirm: 0xE1 on task,
19:39:50:189 0xD0CF5EFFFEC7E088 error APSDE-DATA.confirm: 0xE1 on task,
19:39:52:038 0xD0CF5EFFFEC7E088 error APSDE-DATA.confirm: 0xE1 on task,
19:39:53:005 discovery for zombie 0x000d6ffffe9f8659 dropped, last try was 131 s ago,
19:39:53:484 discovery for zombie 0x000d6ffffe9c3047 dropped, last try was 361 s ago,
19:39:58:555 0xD0CF5EFFFEC7E088 error APSDE-DATA.confirm: 0xA7 on task,
19:40:06:104 Current channel 15,
19:40:06:136 Device TTL 2591 s flags: 0x7,
19:40:09:324 discovery for zombie 0x000d6ffffe9f8659 dropped, last try was 147 s ago,
19:40:09:805 discovery for zombie 0x000d6ffffe9c3047 dropped, last try was 378 s ago,
19:40:14:270 0xD0CF5EFFFEC7E088 error APSDE-DATA.confirm: 0xA7 on task,
19:40:14:270 max transmit errors for node 0xD0CF5EFFFEC7E088, last seen by neighbors 5 s,
19:40:14:726 max transmit errors for node 0xD0CF5EFFFEC7E088, last seen by neighbors 5 s,
19:40:16:783 max transmit errors for node 0xD0CF5EFFFEC7E088, last seen by neighbors 7 s,
19:40:18:512 max transmit errors for node 0xD0CF5EFFFEC7E088, last seen by neighbors 9 s,
19:40:20:538 0xD0CF5EFFFEC7E088 error APSDE-DATA.confirm: 0xA7 on task,
19:40:20:538 max transmit errors for node 0xD0CF5EFFFEC7E088, last seen by neighbors 11 s,`

kses, 2019-09-19:
@manup any ideas of how to continue? Complete reset of everything and hope it does not happen again?

riston, 2019-09-21:
I have a similar problem all the sensors, remotes are connected but lights are all disconnected and can't be paired again. Has anyone found a solution to this problem?

Gateway: 2.05.69 / 06/09/2019
Firmware version: 264A0700

FezVrasta, 2019-09-21:
I hope the maintainers understand this is causing lot of issues to the customers...

FezVrasta, 2019-09-23:
@manup do you have any estimate for when a hotfix will be released?

Raniz85, 2019-09-23:
I've got this same issue after restarting my Raspberry Pi today. None of the lights, senors or switches are reachable. Powercycling them changes nothing in either UI.

I initially thought it was just my motion sensor acting up (since I had issues with it when I first set it up) but later saw that nothing had reconnected with the hub.

MacJawa, 2019-09-24:
Guys, I'm in similar situation. Luckily I only have few bulbs and I was able to make them alive again. Check issue #1857

kses, 2019-09-24:
@manup my setup works now and I do not know why. All of a sudden it started reporting 0xE1 so I did a RF scan with my AP. Showed nothing strange. Moved the usb device 1m and it started working. But I do not think it was the original problem since it recently reported a new error code and the setup have not been moved at all before that date

Raniz85, 2019-09-26:
I've reset all my bulbs and had to redo all my scenes to get everything working.

After that I've restarted my Pi 2 times and I haven't had the same issue again - everything is working fine now - I'm very hesitant to restart it again though :/

luci84tm, 2019-11-08:
Hi there, 
I face also a problem after switching from pi3 to pi4 docker deconz, many of the devices do not join the network:
[image]

What can I do without resetting everything ?

daSteB, 2019-12-05:
same issue for me since today in the morning :/ all lights and most of my switches are not reachable any more. they are still shown in phoscon app, but cannot be used.
what is 
hardware/sofware information:
[image]

network information seems strange... there seems to be a mismatch. I never changed that consciously.. and always thought I was using channel 25. In the zigbee configuration page it shows up 20:

[image]

[image]

Following types of lights are used (all not reachable)
Philips	LWG001
Philips	LWB010
innr	RB 245
innr	RB 165
OSRAM	Plug
OSRAM	Classic A60 W clear - LIGHTIFY
OSRAM	Outdoor Lantern B50 RGBW OSRAM

Screenshot of switches with some reachable and some not:
[image]

p.s. when trying to create a backup, I also get a failure - without any additional information. just a red cross in the middle of the popup

daSteB, 2019-12-05:
short update: restoring the latest zigbee configuration that was using channel 20 was fixing the issue for me. Thank god, all lights and switches are reachable again - without a lot of manual work for a full new setup. Nevertheless this still is a weird thing - what does change zigbee channels suddenly?!

hellcry37, 2019-12-07:
It happen to me again on ConBee II v2.05.71, all lights stopped responding but there were seen ok in home assistant and deconz web interface. Just did not take any commands. To solve it I had to power off all lights. All sensors worked ok in this time.

IIIdefconIII, 2019-12-10:
Same to mate, all lights stopped working somewere overnight

[image]

I had to reset the lights 6 times power off (ikea bulbs) and then they were added and working again.

Raniz85, 2020-04-17:
When I had this issue I had 3 hue bulbs, 1 IKEA bulb and 1 Innr bulb. Haven't had it since though despite a few restarts and upgrades.

Mimiix, 2020-06-05:
It seems this issue is inactive / resolved. If it is not, please re-open or create another issue!

closed 2020-06-05 by Mimiix