dresden-elektronik/deconz-rest-plugin #1855: There doesn't seem to be a ping/heartbeat feature implemented in WebSocket feature

Kroellie, 2019-09-13:
I'm subscribing to the websocket for monitoring purposes, and have noticed connectivity drops. Upon trying to implement a fix on client side, it seems there is no ping/heartbeat feature built into deconz websockets. Could this be implemented as per RFC?

ebaauw, 2019-09-13:
Are you connecting to the websocket through `localhost` or over the network?  What platform do you run deCONZ on?  And the websocket client?

I've been running `dc_eventlog` as a service next to each of my deCONZ instances for years, and I've never seen a websocket disconnect.  Even when you have no ZigBee devices connected, notifications for the built-in Daylight sensor are issued at least every couple of hours.

Kroellie, 2019-09-13:
deCONZ is running in a docker on my Synology. No issues there. 

I'm connecting to websockets service over local LAN from within a container subscribing to websocket with NodeJS which pushes data to MQTT. InfluxDB is pulling the data from MQTT, and Grafana is subsequently showing it. 

I've noticed this morning that grafana wasn't showing metrics anymore since yesterday morning. I've started another websocket subscription on my workstation, and it was showing data coming through the websocket. Looking at the NodeJS container logs: that one wasn't receiving anymore.

So I decided to smarten-up the NodeJS script with connectivity checks. Now I have errorhandling on the connectivity. However, there still is the possibility that the connection is there, but there is no data coming through. For that, the websocket RFC describes a ping / pong mechanism, which currently is lacking in deCONZ.

closed 2020-01-19 by stale[bot]