dresden-elektronik/deconz-rest-plugin #1857: Tradfri 5 button remotr

repiuk, 2019-09-13:
The < and > buttons do not work. All others can be setup with the phoscon app. I would prefer a standard binding to a light like ikea does. A simple cycling off the colours.
Upgraded the remote firmware without effect. Removing it and repairing with conbee has no effect. 
Tried the standard ikea setup by pairing the light to conbee and pairing the remote to the light results in a light not responding to conbee. This workaround worked with Hue. 
I'm using home assistant.

fluffymadness, 2019-09-13:
I can confirm this. I have the exact same behavior. Conbee II with 2.05.69 and firmware 26490700

repiuk, 2019-09-13:
I have selected color temp for the  < > buttons tried the same color temp on a philips dimmer. Nothing happens.. . Maybe color temp action is the problem.

repiuk, 2019-09-13:
There is something very wrong in the current version. The remote does work if I control a white spectrum ikea bulb. On a color light it does do nothing with color temp It's very unpredictive the color temp keeps fluctating, sometimes the bulb flashes. The Phoscon GUI sometime responds with the color temp slider sometimes it doensn't. The GUI also jumps from English to German, refushes to save the group or the delete switch key simply is missing. Very very strange

salopette, 2019-09-14:
I also have the problem.

I used to be able to change colors, now nothing happens anymore

hteggi, 2019-09-16:
I also have the same problem with the color temperature.

And I also have problem with assigning two Ikea remotes to different lights. No matter how I configure them they control the same lights when connected to conbee II. This works fine with the ikea hub. 

I have only had the conbee device for 2 days using raspberry pi 3 with hassoss so I have no idea if this worked in previouses versions.

MacJawa, 2019-09-23:
I can confirm that I was able to control color of Ikea bulb using Tradfri remote <> keys, but it stopped working. I upgraded from 2.05.65 to 2.05.67 recently. At first, one of my tradfri bulbs gone offline (showed offline status when light was on) and I wasn't able to control it. I didn't know why, because another one worked fine and I had no time to investigate. 
My remote is paired to this working one and it worked fine just yesterday. Today when I uplugged / plugged this last working bulb, it went offline as well. I only found one way to make it work - reset with 6x on/off sequence. It came back online after that. At the same time discovering was running, not sure if needed (can test tomorrow with another bulb, which is still offline). After it back to life, I had to pair remote again to the light. After pairing I am able to change brightness and turn the on/off, but it does not react on <> keys to change color.

MacJawa, 2019-09-24:
So I had to re-pair the bulb after it went offline, reset only didn't help. Fortunately I only have 3 bulbs. I also re-paired the remote, but still the <> keys doesn't change color. If I will found how to install older version of deconz via apt on raspberry, I will probably give it a try, as my system is pretty small yet, so it wouldn't hurt too much to do some tests.

RoPe311, 2019-10-09:
Same problem for me with rpi 3B+, conbee II (2.05.69 and firmware 26490700, headless version).
The < > buttons do nothing although I see the events 4002 and 5002 in FHEM.

m50434, 2019-10-25:
I can confirm the issue as well.
rpi 3B+, conbee II (2.05.69 and firmware 26490700, headless version).
I see the events of pressing the button in IoBroker.... but the color of the lights does not change.

Dinth, 2019-11-24:
i always had this issue, but recently - i believe the latest update - this has been fixed. 
Still not perfect though, as the editor shows some glitchy behaviour while setting up up, down, left and right keys, but its certainly possible to assign brightness and colour temperature to those keys.

StuartUSA, 2020-03-24:
Can anyone else confirm this issue is resolved?

Dinth, 2020-03-24:
Partially, please see my last comment

mvieleers, 2020-04-10:
With firmware 2.05.75 this is absolutely not working.
I don't think it has to do with the buttons, though.
I can program changing brightness on the brightness buttons, but also on the <> buttons, this will work.
Whenever I program next/previous scene or change color temp on the brighness buttons or <> buttons, it will not work.