dresden-elektronik/deconz-rest-plugin #1859: deCONZ GUI "not connected"

Allras82, 2019-09-13:
Hi.

I have a Conbee II and a RPI3
It works fine in the web app and i can control and add light and switches.

But when i start the deCONZ GUI it stays as "not connected"

How do i get my GUI to work?

BR
Allan

Allras82, 2019-09-13:
I get your point.
Today I just reinstalled my entire system so it now is based at the desktop version or rasbian. Iv also followed the instruction from Dresden in how to install deconz...

https://phoscon.de/en/conbee/install#raspbian

Iv installed latest beta.

Does that lead to at headless setup of deconz?

Smanar, 2019-09-13:
Nope, desktop version is the version with GUI but you can swap the "working mode".

Headless Version :
sudo systemctl disable deconz-gui
sudo systemctl stop deconz-gui
sudo systemctl enable deconz

GUI version :
sudo systemctl disable deconz
sudo systemctl stop deconz
sudo systemctl enable deconz-gui

Allras82, 2019-09-13:
Got it... it works now.
Thanks 😃

closed 2020-01-19 by stale[bot]