dresden-elektronik/deconz-rest-plugin #1860: Support for new IKEA GU10

JanLp, 2019-09-14:
Is it possible to add support for the new LED bulb from IKEA (IKEA LED1837R5) [https://ikeacy.azureedge.net/files/Buying_guides/HFB10/en/Smart_lighting_buying_guide_AUG19CY_en.pdf](url)
Thanks in advance.

Kane610, 2019-09-14:
Go to the wiki and follow the guide on how to request support for new devices

JanLp, 2019-09-15:
> Go to the wiki and follow the guide on how to request support for new devices

I don't know how to do it. I use a Conbee 2 with Hass.io and add my devices via Phoscon

manup, 2019-09-15:
We haven't yet tested this particular downlight but usually all IKEA lights should work out of the box. Have you tried to reset the light with the IKEA reset sequence before joining?

https://phoscon.de/en/support#faq-ikea1

closed 2019-09-18 by JanLp