dresden-elektronik/deconz-rest-plugin #1861: TRADFRI bulb E26 WS opal 980lm reports color mode as "xy"

glward, 2019-09-14:
The REST api seems to be reporting this bulb as color mode "xy" even though it's only a color temperature bulb:

```
{
    "ctmax": 454,
    "ctmin": 250,
    "etag": "da6d097ee7330bd24d91488002420aef",
    "hascolor": true,
    "manufacturername": "IKEA of Sweden",
    "modelid": "TRADFRI bulb E26 WS opal 980lm",
    "name": "Writing Desk Lamp",
    "state": {
        "alert": "none",
        "bri": 254,
        "colormode": "xy",
        "ct": 250,
        "on": true,
        "reachable": true
    },
    "swversion": "1.2.217",
    "type": "Color temperature light",
    "uniqueid": "00:0b:57:ff:fe:dd:47:63-01"
}
```

Setting the color temperature still works correctly, both via the REST api and the webapp, but it seems odd. Attempting to change the color mode over the REST api does not return _anything_, no errors or success:

```
glward@ares:docker/ha$ curl -H "Content-Type: application/json" -X PUT "http://IP/api/KEY/lights/8/state" -d '{"colormode": "ct"}'
glward@ares:docker/ha$
```

sub0ne, 2019-09-15:
I can confirm this. When I query my bulbs state I got this in return:

{ 
   "ctmax":65535,
   "ctmin":0,
   "etag":"1fccf86c78b24a8414433d4f4468960e",
   "hascolor":true,
   "manufacturername":"IKEA of Sweden",
   "modelid":"TRADFRI bulb E27 WS opal 1000lm",
   "name":"Kellerleuchte 1",
   "state":{ 
      "alert":"none",
      "bri":254,
      "colormode":"xy",
      "ct":100,
      "on":true,
      "reachable":true
   },
   "swversion":"2.0.022",
   "type":"Color temperature light",
   "uniqueid":"00:0d:6f:ff:fe:03:b1:1d-01"
}

Kane610, 2019-09-23:
@manup can we get a comment on this?

manup, 2019-09-23:
I think since these are are plain color temperature lights we can fix the color mode to `ct`.
It's similar to the LEDVANCE bulbs issue https://github.com/dresden-elektronik/deconz-rest-plugin/commit/b30f4f8a493cf6d65bc3581fb96bb2a7978f38d7

Looking forward to get this fixed for 2.05.70.

Kane610, 2019-09-23:
Great!

It would be nice if we could separate what "hascolor": true, means to actually color and add something like a haswhitespectrum  or similar while we're at it :)

Seraphis411, 2019-09-25:
Same issue with the IKEA/TRADFRI "LEPTITER Recessed spot light" (which isn't officially listed as supported but working well besides):

    "etag": "1be33fbb2d1b027fe7c7959c28e16942",
    "hascolor": true,
    "id": "2",
    "manufacturername": "IKEA of Sweden",
    "modelid": "LEPTITER Recessed spot light",
    "swversion": "2.1.022",
    "type": "Color temperature light",
    "uniqueid": "00:0d:6f:ff:fe:19:48:17-01"

yamlCase, 2019-10-11:
Does anyone have a workaround until 2.05.70 comes out?  My home automation's principal investor is getting upset that the closest I can come with the color slider has a green tint XD

Flurkmark, 2019-10-12:
@manup I also would very much appreciate at least a commit to fix this. I am stuck in a project and can't write anything more until I can reliably pull ct from the light. I understand that you guys are busy, but for many people this is a major issue.  Danke.

yamlCase, 2019-10-12:
@Flurkmark I think the fix is in for 2.05.70, we just have to wait for the build.  I think this weekend I will try a manual build from the source and report back on if it works or not. 

Is there a nightly available @manup ?

Flurkmark, 2019-11-03:
This is not fixed as of now, version 2.05.70, tried a build. Pretty bad bug for me, since I can't change color temp, kind of the reason for buying color temp lights..

bedaes, 2019-11-08:
The same thing happens for all sizes of IKEA Floalt panels, I'm already on deCONZ 2.05.70:

```
{
      "ctmax":454,
      "ctmin":250,
      "etag":"0916428f1bd5fd7c6ded911b7f6047e9",
      "hascolor":true,
      "manufacturername":"IKEA of Sweden",
      "modelid":"FLOALT panel WS 30x30",
      "name":"Warehouse Panel",
      "state":{
         "alert":"none",
         "bri":10,
         "colormode":"xy",
         "ct":454,
         "on":true,
         "reachable":false
      },
      "swversion":"1.2.217",
      "type":"Color temperature light",
      "uniqueid":"00:0d:6f:ff:fe:1b:c3:61-01"
   },
```

Flurkmark, 2019-11-17:
@manup 
zcl_tasks.cpp line 384
If I comment 
``
useXy = useXy || task.lightNode->manufacturerCode() == VENDOR_IKEA;
``
they work as expected, at least so far as I have tested.

sverleysen, 2019-11-21:
I'm having the same issue

Nicxe, 2019-11-23:
I have the same issue

Jpsy, 2019-12-30:
Same issue here.
In Home Assistant this results in CT lamps not reporting back any color temperature at all.
I just migrated a group of lamps from a Hue bridge to a Conbee stick, just to find that all my automations are broken (because they rely on the bulbs reporting back their current color temp). Very annoying.
I am using latest HA (0.103.5) and latest HA deCONZ add-on (5.0).

guennikasse, 2020-01-06:
Same issue here.
If I change the color_temp using a TRADFRI Switch the color is displayed as attributes in Home Assistant. If I use the slider in HA color_temp attributes disappears.
I am using latest HA (0.103.5) and latest HA deCONZ add-on (5.0).

jensflorian, 2020-01-07:
Same issue here with IKEA Floalt 30x30 panels using deCONZ 2.05.72. In phoscon app, color temperature slider works as expected, but after connecting TRADFRI or Hue Dimming switches,  setting color temperature to cooler on switches is unresponsive and results in mismatch of  bulbs reporting back their current color temp and actual color state.

jdeluyck, 2020-01-16:
Same issue with TRÅDFRI bulb E14 opal 400lm lamps...

alexlopezcifuentes, 2020-01-16:
Same issue here with an Ikea Bulb. When Home Assistant is restarted the attribute "color_temp" is enabled. However, every time I use it in automation it disappears...

sverleysen, 2020-02-07:
> @manup
> zcl_tasks.cpp line 384
> If I comment
> `useXy = useXy || task.lightNode->manufacturerCode() == VENDOR_IKEA;`
> they work as expected, at least so far as I have tested.

Does this still fixes this issue? Or did you encounter some other issues with this change?
How can I implement this in my own setup?

Flurkmark, 2020-02-07:
It fixes it, I'm still on .69 since it's stable for me. 
Follow the docs on how to compile the plugin, first installing the .deb
Just delete the mentioned line. 

It was supposed to be a fix for deconz build in scenes afaik. I don't use them, but I guess they will missbehave with IKEA light after the fix.

sverleysen, 2020-02-27:
@manup, any update? 
This would be fixed in 2.05.70, I installed 2.05.74 today but it's still not fixed.

llonca13, 2020-02-28:
any plans on fixing this? Just some feedback would be great to know is this being worked on, or we should start looking for alternatives (new bulbs etc)...
thanks

sverleysen, 2020-02-28:
I have made a PR for this fix. Hopefully they will merge it.
https://github.com/dresden-elektronik/deconz-rest-plugin/pull/2507

Flurkmark, 2020-02-28:
> I have made a PR for this fix. Hopefully they will merge it.
> #2507

I didn’t do it initially, since I thought it would be fixed quickly after I found it. Hopefully it will get attention now. They should just back out the commit that introduced it and redo the scene fix in my opinion. Fingers crossed...

miki9987, 2020-03-02:
Even with this fix I still don't get color_temp reported back..
https://github.com/home-assistant/home-assistant/issues/26424

straaat, 2020-04-04:
I have  this issue as well. Very annoying. If I VNC into deCONZ I can set the color temperature using the color control cluster of the light. It then also shows the correct value in Home Assistant/OpenHAB.

But it seems the Phoscon interface as well as the connected Home Assistant do not set a color temperature value but a XY value instead...

ebaauw, 2020-04-04:
See #2507

DerOetzi, 2020-05-18:
Same issue with all my GU10 WS TRADFRI lamps. Investigated on this and found following behavior:

1. Deconz started: all lamps send correct => colormode: ct, ct value reported correctly via REST API
2. Phoscon app change color temperature of group => colormode:ct, ct value reported correctly
3. Phoscon app or Postman request to single lamp => colormode: xy, ct not changing in report
Can repeat 3 as often as I want colormode stays for this lamp at xy and not reporting ct value changes
4. Phoscon app change color temperature of group => all lamps in group switch back to colormode: ct and reporting correct ct value

Hope this helps for finding a solution

closed 2020-06-15 by stale[bot]