dresden-elektronik/deconz-rest-plugin #1866: Ikea Color Bulbs don't react to Alexa color changes

jniclas, 2019-09-15:
I have several Ikea Bulbs of:

Hersteller IKEA of Sweden
Modell TRÅDFRI Lampe E27 CWS opal 600lm
Version 1.3.009

with Gateway:

Version 2.05.64 / 22.4.2019
Firmware 26490700

and for a few weeks now, they don't change the color triggered by alexa. The Alexa app shows the error "malfunction of the device". With Hue Essentials and Phoscon App colorchanges work.

abmatus, 2019-12-16:
I have the same issue with TRÅDFRI Lampe E27 CWS opal 960lm.
Is there a workaround?

jniclas, 2019-12-17:
After a backup, I have reinstalled phoscon with version 2.05.69, loaded the backup and reconnected all bulbs in the alexa app. Now it works again, but this solution is quite time consuming.

acid115, 2020-01-08:
Hi, 
I have the same issue.
But only since i have banned my echo 1.gen  and use the deConz Devices through my echo 3.gen.
So what happened since version 2.05.70 ?
Is a downgrade possible from 2.05.71 to 2.05.69 or may i lost all my devices? 
Thx

acid115, 2020-01-09:
Hi again.
With tcpdump i can see, that the echo 3.gen sends only "hue" and "sat"-value in http request to change color. But the bulb in deConz needs "x,y" value instead to change the color (i have tested)
So does anyone has an idea?
Could it help to delete the bulb device and reregister it to deConz and Amazon. Or is it time wasting?
Or is the solution to buy an philips rgb instead ? :/

closed 2020-05-15 by stale[bot]