dresden-elektronik/deconz-rest-plugin #1867: phoscon breaks schedules

jose-d, 2019-09-15:
When there is "on" schedule, Phoscon app will break it.
Reproducer:

**1) create schedule in Wireles Light Control (WLC) web:**

[image]

detail of the Group **ON** schedule:

[image]

**2) go to the Phoscon web GUI, schedule is still there:

[image]

**3) click on the "ON" action - it will disappear and schedule is automatically saved**

(and there is no way how to add the action again here)

[image]

**4) go back to WLC - the "Group **ON**" schedule changed to "Group **OFF**"

[image]

expected behavir:

Phoscon is not breaking schedules from WLC or schedules planning is implemented in PWA.

manup, 2019-09-15:
Interesting, I've put it on our check list. I guess this might be related to a prior decision to favour scenes over "on" or "brightness" settings for a schedule in the Phoscon App, since scenes are more fine grained and allow setting the state for each light individually.

So we have a few choices to improve the issue:

1. Treat WLC schedules or any schedules which are not created with the Phoscon App as readonly, perhaps with a small hint why they are readonly
2. Add the functionally provided by WLC to the Phoscon App
3. Don't show non Phoscon App schedules at all

jose-d, 2019-09-15:
>  to favour scenes over "on" or "brightness" settings for a schedule in the Phoscon App

ah, thanks, ok, this was quite unobvious for me - as I had no scenes for WC fan, I thought the schedule support is simply not yet implemented there in PWA - that was the reason why I used WLC.

So for my usecase it's enough to create scene fan_on and fan_off.

Perhaps some hint that one needs to add scene at the "Schedules of group"  (http://192.168.1.208/pwa/group-time.html) could be added.

manup, 2019-09-15:
>Perhaps some hint that one needs to add scene at the "Schedules of group" (http://192.168.1.208/pwa/group-time.html) could be added.

Indeed we should make this more clear, the page looks way too empty when no scenes are defined for a group.

https://phoscon.de/en/app/doc#group-schedules-en

closed 2020-01-20 by stale[bot]