dresden-elektronik/deconz-rest-plugin #1871: IKEA blinds moves down first, before moving up, on up command

DarkFox, 2019-09-15:
> Closing the blind (moving down) is okey. But when I open the a blind (moving up) by a few percentage (or completely) initially it starts to move down (closing direction) then stops and a second later it moves up to the target position. It does like this all the time.

_Originally posted by @A320Peter in https://github.com/dresden-elektronik/deconz-rest-plugin/issues/1121#issuecomment-531517786_

A320Peter, 2019-09-18:
It is on firmware 26330500 version 2.05.69 / 9/6/2019. On Home Assistant 0.95.4 it was working correctly. I experience the mentioned issue since upgrade to HA 0.98.5, on hassio.

BeyondRico, 2019-09-20:
Kinda the same problem here. When it goes from 0% to 100% there is no problem, but when i try to get it up it moves by around 2 cm. So it looks like it's forgetting it's min and max height. I first need to lower it (it goes over 100% and then wraps up on the wrong side) and then need to press up again to get it to 0%.

DarkFox, 2019-09-20:
> Kinda the same problem here. When it goes from 0% to 100% there is no problem, but when i try to get it up it moves by around 2 cm. So it looks like it's forgetting it's min and max height. I first [...]

This sounds like a different issue.

BeyondRico, 2019-09-20:
Can confirm, when connected with the Ikea hub it has the same problem. my bad.

BeyondRico, 2019-09-20:
> 
> 
> > Kinda the same problem here. When it goes from 0% to 100% there is no problem, but when i try to get it up it moves by around 2 cm. So it looks like it's forgetting it's min and max height. I first need to lower it (it goes over 100% and then wraps up on the wrong side) and then need to press up again to get it to 0%.
> 
> This sounds like a different issue, perhaps unrelated to deconz.

Can confirm, when connected with the Ikea hub it has the same problem. my bad.

DarkFox, 2019-09-20:
> Can confirm, when connected with the Ikea hub it has the same problem. my bad.

No probs. It sounds like your unit is faulty, so I'd suggest contacting IKEA support, or just go and get it replaced.

olemr, 2019-11-18:
No fault. I bought one last week and it behaves the same.
Hope Ikea fix it in an upcoming firmware release.

ioangogo, 2019-11-28:
Looking at some of the setup videos from ikea, this behaviour seems to be the blinds checking where it is before completing the action

closed 2020-04-03 by stale[bot]