dresden-elektronik/deconz-rest-plugin #1873: IKEA TRADFRI remote FW update ZB3.0 breaks groups

Marlor, 2019-09-16:
In Phoscon you cant click or use the auto generated group from a TRADFRI remote if it got the ZB3.0 FW.
[image]

manup, 2019-09-18:
Interesting, we need to test this. Can you please provide the firmware version for the Zigbee 3.0 remote, it should be visible in the Phoscon App switch details page.

Marlor, 2019-09-18:
Version 2.3.014
fw.ota.homesmart.ikea.net/Tradfri_OTA_release_signed_2019_08_27_162854/bin/159699-5.1-TRADFRI-remote-control-2.3.014.ota.ota.signed

Changelog:
Remote Control V-2.3.014.
 * Upgrade to ZB3.0.

FelixUster, 2019-11-07:
same issue here

ebaauw, 2019-11-08:
Just updated one of my Trådfri remotes to 2.3.014.  Indeed it's now ZigBee 3.0.  It still sends the same commands after the update, but it sends them to group 0x0000.  Note the missing `config.group`.  Looks like it no longer picks a random group on reset, very much like the other ZigBee 3.0 Trådfri devices.

@manup, I had a very hard time pairing the remote to my test network after the update and reset.  Not sure what's going on here.  deCONZ 2.05.70, Raspberry Pi 4 (buster) with ConBee II, REST API plugin from distribution.  deCONZ would not read the simple descriptors while the network is open.  Confirm by sniffing that it doesn't send the command (tried other nodes as well, tried restarting deCONZ).  After closing the network, the descriptor would be read.  So I had to:
- Open the network from the old web app;
- Reset the remote (4x press reset button);
- Wait for the node to appear;
- Close the network from the old web app;
- Press the _Read Simple Descriptor(s)_ from the left dropdown; press a button on the remote (to wake up) -> right drop down appears;
- Open the network from the old web app;
- Read the _Basic_ cluster attributes; press a button -> REST API resource gets created.

WladyX, 2019-11-10:
Is there any way to change the group of the remote?
I don't want to controll all the lights with it. Or is it possible to remove all lights from group 0?

Thanks!

WladyX, 2019-11-17:
As a workaround I was able to downgrade the firmware on the remote and it seems ok so far.

hlmy1, 2019-12-26:
> As a workaround I was able to downgrade the firmware on the remote and it seems ok so far.

How to downgrade Ikea dimmer-remote from 2.3.014 to 1.2.214 ?

ebaauw, 2019-12-26:
> Is there any way to change the group of the remote?

In the deCONZ GUI, create a bindings from the client (grey) _On/Off_, _Level Control_, and _Scenes_ clusters to the intended group.  Make sure to wake the remote by pressing a button.  When successful, `config.group` should be populated after pressing a button.

On pairing, the REST API plugin should create the group and setup the bindings for you, but as I've indicated above, there seem to be some issues sometimes with reading the _Simple Descriptor_ while the network is open.

WladyX, 2019-12-26:
downgrading is the same procedure as for upgrading the fw on it.
after some trial and error i did:

reboot the whole raspberry
re-pair
select it in the table
select the related update file
click update button
if the process does not start after a period:
pull out the battery, press a button a few times and put battery in again
put it in again

this helped: https://github.com/dresden-elektronik/deconz-rest-plugin/issues/947

muratacar73, 2020-01-25:
> 
> 
> As a workaround I was able to downgrade the firmware on the remote and it seems ok so far.

Hello,
can you share the old firmware for the IKEA TRADFRI remote.
I can't find it.
Thanks

Marlor, 2020-05-25:
Now that all the groups (not just the ZB3.0 ones) in Phoscon are gone, should i close this issue?
Or is this a new bug?

Mimiix, 2020-06-06:
It seems this issue is resolved or otherwise inactive. If it is not, please re-open!

closed 2020-06-06 by Mimiix