dresden-elektronik/deconz-rest-plugin #1875: Orvibo smoke sensor SF21

mn26826, 2019-09-16:
It would be great if someone could add support for the SF21 smoke sensor by Orvibo. The SF20 is already supported, so maybe it's an easy job? I don't know ;)

Here are two screenshots:

[image]

[image]

Thanks in advance. Really appriciate it.

udoewich, 2019-09-18:
@mn26826 can you please copy (i.e. select the value field and use Ctrl+C) the complete Model Identifier and post it here?  I believe it's missing the last four digits in your screenshot.

mn26826, 2019-09-19:
@udoewich the complete Model ID is 98293058552c49f38ad0748541ee96ba.

closed 2019-09-22 by manup

BeamMeUpTo, 2019-11-03:
Tried to connect an Orvibo SF21 with Phoscon App but it seems not to recognize him...
get sensors/lights shows nothing, too.

But in deCONZ (Raspbee Version 2.05.69 26330500) it is visible - with the same Model-ID!

Is there some kind of trick? First Resetted the sensor --> press 5 seconds Connect-Button.
For connecting it I started phoscon --> connect new sensor --> Press 2 sec Connect-Button! 

But no new sensor :(

Any hints for me? Please...
Thanks

[image]

[image]

[image]

udoewich, 2019-11-03:
@BeamMeUpTo support has been added, but it's not in .69 - it should be part of the next release (.70 presumably?).

BeamMeUpTo, 2019-11-04:
Deleted the node in deconz. Resetted the sensor.
Upgraded to .../raspbian/beta/deconz-2.05.70-qt5.deb
Phoscon --> New Sensor.... Nothing :(

But visible in deconz again - no phoscon/REST-API (get sensors/lights) recognition...
Still wrong version? No yet comitted?

Thanks for the help

udoewich, 2019-11-05:
I'm not sure, as .70 hasn't released officially yet (not listed under releases).  Someone else would have to chime in as to how a smoke sensor would show up in Phoscon, as I don't have any sensors of that kind in my setup.

I assume @mn26826 hasn't been able to test the integration yet either?

BeamMeUpTo, 2019-11-21:
I'm now on V2_05_71

Deleted the old node in deconz. Resetted the sensor.
Phoscon --> New Sensor.... Nothing :(

But visible in deconz again - no phoscon/REST-API (get sensors/lights) recognition...

Same question? Still wrong version? Not yet comitted?
Can I do something to help accelerate the development?

Thanks Beam

udoewich, 2019-11-22:
I checked just now: the support for both SF20 and SF21 is in the source code since .70.

Again, since I don't have one of these, I can't be 100% sure it's properly supported.

I suggest the follow procedure:

2) reset the device, to get rid of any old pairing info
3) once the device shows up in deCONZ again, look at the Basic Cluster and make sure it shows the Model Identifier (might take a while to show up), since that is what Phoscon needs for proper identification
4) put your device in pairing mode and have it physically close to the ConBee/RaspBee
5) go the the Devices > Sensors page in Phoscon and use the "Add new sensor" button

That should normally do it.

BeamMeUpTo, 2019-11-22:
Did it exactly your way... sensor shows up in deconz - then read the basic Cluster, while phoscon searches for new sensors - but did not find any. REST-API as well: get sensors - no new device is listed :(
Thats what deCONZ shows, after reading BASIC-Cluster while phoscon searches for new sensors
:
[image]

udoewich, 2019-11-23:
Your last screen shot does not show any active links to the device.  AFAIK, it will only communicate properly if it has at least one link.

Can you capture some of the deCONZ log messages?  Look for messages with the address of your device (0x000d6f0010dd0c6c).

@mn26826 is your SF21 working with .70 or .71?

BeamMeUpTo, 2019-11-23:
Its my first deCONZ log! I have no clue if it is the right way - please help, if its the wrong way.

I tried to generate the log with: 

`deCONZ --dbg-info=1 --dbg-zdp=1 --dbg-zcl=1 --db-aps=1 --dbg-http=1 --dbg-error=1 > deCONZ.log`

but the connected Orvibo with id 0x000d6f0010dd0c6c never showed up in the log - but is connected, see new screenshot.

[image]

udoewich, 2019-11-24:
Sorry, looks like I missed that the SF21 uses a different vendor and MAC prefix compared to the SF20.  I just issued a pull request with the fix.

BeamMeUpTo, 2019-11-24:
Hopefully fixed then ;) - in .72??
Feel free to ask me for some more tests... if the fix is committed in the next version. 

Thanks, Beam

udoewich, 2019-11-24:
Hopefully, yes. :)  If all goes well, it should be part of the next release.

BeamMeUpTo, 2019-12-14:
So, finally I got the SF21 working with deconz-2.05.72-qt5.deb.

Sensor shows up in deconz - then read the basic Cluster, while phoscon searches for new sensors! Phoscon detects the new sensor - ready ;) REST-API as well: get sensors output

40: SF21 smoke sensor OG.FL.Rauchmelder ZHAFire    

But phoscon itself shows no new device - is that right?
Thanks so far ;)

udoewich, 2019-12-17:
Great, and thanks for the letting us know!

Sorry, but I have no experience with Zigbee alarm sensors in Phoscon, so don't know where they would show up (if at all).

BeamMeUpTo, 2019-12-27:
Just for letting you know.... I ordered 2 more devices and all (now 3) got recognized ;)

mkuutti, 2020-01-16:
Hi, is the SF21 supported or not Phoscon yet? BeamMeUpTo’s comment leaves me uncertain. As well as the fact that I'm getting the same behaviour as he did on 14 Dec, device is found by Phoscon but does not show up in the sensors list. I'm on 2.5.72.

BeamMeUpTo, 2020-01-16:
They do not show up in Phoscon!!!
But you need the Phoscon "New Sensor Connect"-Button to pair the sensor. 
SF21 is accessed through REST-API ONLY!!! - which in my case is managed by "FHEM".

mkuutti, 2020-01-16:
Right. I’ll use Home on my Apple devices. I hope Phoscon gets to support them at some point.

Thanks.

Mikko

BeamMeUpTo, 2020-10-11:
Today, I received ten new "Schwaiger ZHS20" smoke sensors, which are totally the same as Orvibio SF20 and SF21 (perhaps only differences in model-id). 
Disadvantage of the Orvibos is the complicated pairing process - with paring while read-out the basic cluster. While the Schwaiger ones integrate fluently in Phoscon AND deCONZ with correct naming.

But PHOSCON is only recognizing the Schwaiger ones and none of the Orvibios! 
-> What makes the difference? 
-> Any additional information needed? 
-> Which? - I would try to provide! Talk to me ;)

RaspBee with 
deCONZ 
2.05.83 und 
Firm 
26350500

Mimiix, 2020-10-12:
@BeamMeUpTo Request device support. Just open a new issue. They need to be whitelisted.

Starsurfer78, 2020-12-21:
I have also the SF21 smoke sensor. After a fault Fire Alarm, the Senor show Fire = 1. (in FHEM)
Does anyone know how I can reset it?
Delet and repair did't change the status.

SwoopX, 2020-12-21:
Can you please open a seperate issue for your experience? This is a device support request.
Especially the environment details requested there are important.

PR #1889 (2019-09-22): ORVIBO SF21 Smoke Sensor support
Adds support for the ORVIBO SF21 smoke sensor. Closes #1875.