dresden-elektronik/deconz-rest-plugin #1876: Most of the lights unresponsive

FezVrasta, 2019-09-16:
Hi, I started getting these logs, and most of my lights don't respond anymore, ideas?

```
deconz           | 21:08:18:294 0x90FD9FFFFE01C6A2 error APSDE-DATA.confirm: 0xA7 on task
deconz           | 21:08:24:944 0x90FD9FFFFE820117 error APSDE-DATA.confirm: 0xA7 on task
deconz           | 21:08:26:041 delay sending request 195 dt 2 ms to 0x90FD9FFFFE8294D2, cluster 0x0300
deconz           | 21:08:26:141 delay sending request 195 dt 2 ms to 0x90FD9FFFFE8294D2, cluster 0x0300
deconz           | 21:08:26:240 delay sending request 195 dt 2 ms to 0x90FD9FFFFE8294D2, cluster 0x0300
deconz           | 21:08:26:341 delay sending request 195 dt 3 ms to 0x90FD9FFFFE8294D2, cluster 0x0300
deconz           | 21:08:26:441 delay sending request 195 dt 3 ms to 0x90FD9FFFFE8294D2, cluster 0x0300
deconz           | 21:08:26:540 delay sending request 195 dt 3 ms to 0x90FD9FFFFE8294D2, cluster 0x0300
deconz           | 21:08:26:641 delay sending request 195 dt 3 ms to 0x90FD9FFFFE8294D2, cluster 0x0300
deconz           | 21:08:26:740 delay sending request 195 dt 3 ms to 0x90FD9FFFFE8294D2, cluster 0x0300
deconz           | 21:08:26:840 delay sending request 195 dt 3 ms to 0x90FD9FFFFE8294D2, cluster 0x0300
deconz           | 21:08:26:940 delay sending request 195 dt 3 ms to 0x90FD9FFFFE8294D2, cluster 0x0300
deconz           | 21:08:27:040 delay sending request 195 dt 3 ms to 0x90FD9FFFFE8294D2, cluster 0x0300
deconz           | 21:08:27:140 delay sending request 195 dt 3 ms to 0x90FD9FFFFE8294D2, cluster 0x0300
deconz           | 21:08:27:241 delay sending request 195 dt 3 ms to 0x90FD9FFFFE8294D2, cluster 0x0300
deconz           | 21:08:27:340 delay sending request 195 dt 4 ms to 0x90FD9FFFFE8294D2, cluster 0x0300
deconz           | 21:08:27:441 delay sending request 195 dt 4 ms to 0x90FD9FFFFE8294D2, cluster 0x0300
deconz           | 21:08:27:540 delay sending request 195 dt 4 ms to 0x90FD9FFFFE8294D2, cluster 0x0300
deconz           | 21:08:27:641 delay sending request 195 dt 4 ms to 0x90FD9FFFFE8294D2, cluster 0x0300
deconz           | 21:08:27:741 delay sending request 195 dt 4 ms to 0x90FD9FFFFE8294D2, cluster 0x0300
deconz           | 21:08:27:840 delay sending request 195 dt 4 ms to 0x90FD9FFFFE8294D2, cluster 0x0300
deconz           | 21:08:27:941 delay sending request 195 dt 4 ms to 0x90FD9FFFFE8294D2, cluster 0x0300
deconz           | 21:08:28:040 delay sending request 195 dt 4 ms to 0x90FD9FFFFE8294D2, cluster 0x0300
deconz           | 21:08:28:142 delay sending request 195 dt 4 ms to 0x90FD9FFFFE8294D2, cluster 0x0300
deconz           | 21:08:28:240 delay sending request 195 dt 4 ms to 0x90FD9FFFFE8294D2, cluster 0x0300
deconz           | 21:08:30:025 0x90FD9FFFFE8294D2 error APSDE-DATA.confirm: 0xA7 on task
deconz           | 21:08:33:522 0x90FD9FFFFE8294D2 error APSDE-DATA.confirm: 0xA7 on task
deconz           | 21:08:33:522 max transmit errors for node 0x90FD9FFFFE8294D2, last seen by neighbors 242 s
deconz           | 21:08:36:340 binding for cluster 0x0006 of 0x90FD9FFFFE820117 exists (verified by reporting)
deconz           | 21:08:37:765 discovery for zombie 0x000b57fffeefc2fd dropped, last try was 135 s ago
deconz           | 21:08:38:884 Current channel 25
deconz           | 21:08:38:905 Device TTL 968 s flags: 0x7
deconz           | 21:08:39:966 0x90FD9FFFFE901334 error APSDE-DATA.confirm: 0xA7 on task
deconz           | 21:08:43:046 max transmit errors for node 0x90FD9FFFFE01C6A2, last seen by neighbors 91 s
deconz           | 21:08:44:871 dev /dev/ttyAMA0
deconz           | 21:08:44:871 GW firmware version: 0x26330500
deconz           | 21:08:44:871 GW firmware version is up to date: 0x26330500
deconz           | 21:08:45:036 max transmit errors for node 0x90FD9FFFFE01C6A2, last seen by neighbors 93 s
deconz           | 21:08:46:106 max transmit errors for node 0x90FD9FFFFE01C6A2, last seen by neighbors 94 s
deconz           | 21:08:48:673 0x90FD9FFFFE792F0F error APSDE-DATA.confirm: 0xA7 on task
deconz           | 21:08:51:753 no button handler for: RWL021 ep: 0x02 cl: 0x0001 cmd: 0x0A pl[0]: 021
```

FezVrasta, 2019-09-16:
I'm on latest RaspBee firmware and latest DeCONZ software, running through Docker on a Raspberry Pi 4.

I already tried to change the channel to all of the available ones with no avail.

My wife hates me when lights stop working, help :-(

FezVrasta, 2019-09-17:
I tried to plug off the power from the RPi and plug it back, still no luck

FezVrasta, 2019-09-17:
It may be related to this:

https://github.com/marthoc/docker-deconz/issues/153#issuecomment-532163658

sub0ne, 2019-09-18:
Same for me. I got Conbee 2 stick running with Pi 3b+ & Raspbian. I got two groups with several light in them. After several hours the light can not be accessed directly anymore. Via the groups the still can be turned on and of. But the individual lights can not be controlled anymore. This makes the whole thing useless.

manup, 2019-09-19:
Can you please provide some details about which lights you are using, a screenshot from the *Phoscon App > Menu > Lights* would be useful.

FezVrasta, 2019-09-19:
[image]

now it marks "Cookers" as available, but actually it doesn't respond to any command.

FezVrasta, 2019-09-19:
I also got some switches and they are unavailable too.

[image]

FezVrasta, 2019-09-28:
@manup what's the plan to address this issue? There are ton of people reporting the very same problem.

I've been 2 weeks without lights for now. What am I supposed to do?

manup, 2019-09-30:
Big sorry for the delay. Can you please provide a screenshot from *Phoscon App > Menu > Settings > Gateway* or a log like in your first post.

I'd like to check if the RaspBee is still connected.

A screenshot from the Zigbee configuration page would also be helpful to see if the configuration has issues and was changed somehow.

The page is described in: https://github.com/dresden-elektronik/deconz-rest-plugin/releases/tag/V2_05_59

Do you have any other devices connected to your Raspberry Pi like USB disks?

FezVrasta, 2019-09-30:
[image]

I have an externally powered hard drive, but I tried to disconnect it and nothing changed. Also, it used to work just fine for over a week before the issue happened.

manup, 2019-09-30:
Looks finde, since the firmware version is shown the connection is there.

Please also check the Zigbee configuration page:

>A screenshot from the Zigbee configuration page would also be helpful to see if the configuration has issues and was changed somehow.

>The page is described in: https://github.com/dresden-elektronik/deconz-rest-plugin/releases/tag/V2_05_59

Here the network parameters should indicate that they haven't changed recently.

FezVrasta, 2019-09-30:
[image]

[image]

manup, 2019-09-30:
Looks fine too, I suppose the channel change from 15 to 25 was done a while ago?
I'm not sure what's causing the non responding devices.

So next step is to figure out if your lights are using the same parameters.

1. *Phoscon App > Menu > Help > Open Wireless Light Control (2016)*
2. *Old WebApp > Menu > Settings > Scan for Devices* (on the bottom of the page)

Hopefully that that should list a few nearby lights. In the list we can see if the same channel and PANID are used.

[image]

FezVrasta, 2019-10-01:
It finds only 1 light, but I have at least 7 lights powered on at the moment.

[image]

manup, 2019-10-01:
>It finds only 1 light, but I have at least 7 lights powered on at the moment.
The scanner can only see lights which are rather close to the gateway.

>I changed the channel a few times when it stopped working to see if it helped with the issue.

Hmm, this is a danger zone (that's why we've put a big red warning to the channel change dialog).
So in worst case your devices are now scattered to multiple channels (which can be solved).

But first things first, can you position the gateway near to some other lights (or bring the lights to the gateway) and issue the "Scan for Devices" again. I really wonder that the light in your screenshot is not only on a different channel but also has a different PANID of 0x4326 instead of your gateways 0x3f1d.

dxxxm, 2019-10-02:
Interesting.  No probs here. All fine. 62 Nodes. ~40 Lights
Just out of curiosity replayed the 'scan for Devices' instructions as given and get this:
[image]
None of the blink buttons make any of my lights blink.
@manup What is the danger zone about ?
ConBee1 on RPI3
`FW 26330500`
`{
  "apiversion": "1.16.0",
  "bridgeid": "00xxxxxxxxxxxxxxxxx",
  "datastoreversion": "60",
  "devicename": "ConBee",
  "factorynew": false,
  "mac": "b8:27:xxxxxxxxxx",
  "modelid": "deCONZ",
  "name": "deconz-xxxxxxxxx",
  "replacesbridgeid": null,
  "starterkitid": "",
  "swversion": "2.5.69"
}`

FezVrasta, 2019-10-02:
@manup here's the screenshot. I moved a light close to the gateway.

Also, discard my previous screenshot, the device listed was a spare smart plug I had plugged in and forgot to configure.

[image]

If I click "blink" nothing happens.

manup, 2019-10-08:
Ah okay so that indicates (at least some) lights are on channel 20.
Please switch the Zigbee channel to 20 in the Phoscon App.

If the lights are there it will take a while until the gateway picks them up, but after 10-20 minutes  they should be controllable again.

FezVrasta, 2019-10-08:
Thanks, I changed the channel to 20 a few hours ago and most of the lights are now back online.

I still have 1 light unresponsive, and all my Philips Dimmers.

What else can I try?

FezVrasta, 2019-10-10:
@manup may you give me guidance please? :-(

ebaauw, 2019-10-10:
To revive the Hue dimmer, press all four buttons until the LED starts cycling colour.

FezVrasta, 2019-10-10:
no luck, they seem death. maybe they kept looking for a network for the whole time my lights were unresponsive and the battery ran out 🤷‍♂️ I'll try to replace them

ebaauw, 2019-10-10:
They should blink red when pressed while dropped off from the network.  If they don’t it might indeed be a battery issue.  To re-pair them, press and hold the reset button (hole) on the back for 10 sec.

closed 2020-02-07 by FezVrasta

S474N, 2021-08-05:
After connecting Sonoff ZBMINI have all lights, IKEA repeater offline.

RPi 4B + Docker version

ConBee II
2.12.03 / 12. 7. 2021
266F0700

In log:
```
today at 18:45:30 18:41:11:656 0x00158D000XXXXX error APSDE-DATA.confirm: 0xF0 on task
today at 18:45:30 18:41:23:453 0x00158D000XXXXX error APSDE-DATA.confirm: 0xF0 on task
today at 18:45:30 18:41:30:453 0x00158D000XXXXX error APSDE-DATA.confirm: 0xF0 on task
today at 18:45:30 18:41:40:904 Device TTL 5290 s flags: 0x7
today at 18:45:30 18:41:53:458 0x00158D000XXXXX error APSDE-DATA.confirm: 0xF0 on task
today at 18:45:30 18:42:00:456 0x00158D000XXXXX error APSDE-DATA.confirm: 0xF0 on task
today at 18:45:30 18:42:23:458 0x00158D000XXXXX error APSDE-DATA.confirm: 0xF0 on task
today at 18:45:30 18:42:30:556 0x00158D000XXXXX error APSDE-DATA.confirm: 0xF0 on task
today at 18:45:30 18:42:40:890 Device TTL 5230 s flags: 0x7
today at 18:45:30 18:42:46:864 GW firmware version: 0x266f0700
today at 18:45:30 18:42:46:864 GW firmware version is up to date: 0x266f0700
```

stefpb, 2021-08-17:
Change the usb cable works for me!

S474N, 2021-08-18:
IoT are connected (with "old" USB gold plated cable), but my log is still flooded:

[image]

SwoopX, 2021-08-18:
As far as I can tell, this is harmless. The coordinator is trying to query the modelID from your Xiaomi deep sleepers.