dresden-elektronik/deconz-rest-plugin #1878: Unable to find and pair Philips Hue lights (LCT001)

madshell, 2019-09-17:
I'm very happy with my ConBee+Phoscon setup, I've meshed 35+ ZigBee devices together without much hassle so thank you for the great dev work.

My only problem is that I can't pair 3 Philips Hue E27 RGB bulbs. I don't have the dimmer switch so I have tested all the other methods that I've heard of:

- using the TouchLink feature of the old Phoscon interface at 10cm and even litteraly touching the Conbee II usb key -> I've never seen the lights showing
- using the Philips internal API of the bridge -> The lamp is doing its reset blinking but I can't find it in Phoscon app neither on the old TouchLink.
- deleting hub bulbs from the iOS v1 bridge app (official) while the lamp was turned on at 5cm or less -> the lamp doesn't seem to reset
- I even tried to switch Hue bridge on channel 15 to match the channel of my ConBee II stick -> no lights discovered

I'm litteraly stuck with no sign of life of Hue lights in the Deconz ecosystem whereas I can pair the bulbs very easily with the v1 bridge Hue app (iOS) even after a Hue bridge factory reset. What am I missing? Could this be a bug in latest firmware version with too old Hue bulbs?

My Pi setup:
- distrib: Raspbian GNU/Linux 9.9 (stretch)
- USB key: ConBee II usb stick
- version: 2.05.69
- firmware: 264A0700

The 3 hue bulbs:
- Philips Hue E27 RGB
- type: Extended color light
- modelid: LCT001
- swversion: 5.127.1.26581
- bridge: round (v1)

Connected devices:
- 24 various Ikea Tradfri bulbs
- 5 Ikea switches + 1 Philips Hue Tap
- 5 Xiaomi sensors + 1 Ikea motion detector

madshell, 2019-09-21:
Solved by buying a dimmer switcher. Power on all lights, hold O/I buttons near each light until it stops blinking. Launch a light scan in the Phoscon web app and the 3 Hue bulbs were instantly discovered. So it's seems it's the only reliable method currently!

closed 2019-09-21 by madshell

lucidyan, 2021-02-26:
Had the same problem and tried @madshell workaround, but not get success with it.
But I remembered that I already bind this lamps to Hue-BT App, so what I did next:

- Select the Lamp
- Go to (...) -> Settings
- Press Reset

After those procedure each lamp discovered quickly by the deconz. Maybe this app was the cause the problem, maybe not, but this workaround worked for me.

rage4design, 2022-10-10:
I have the same problem with LCT001, SW: 5.130.1.30000.
1.  I can connect the LCT001 to Hue Bridge v1.2 (7530) without any problems
2.  Then I remove it from Hue Bridge (this should reset the bulb)
3.  Search in Home Assistant Conbee II Home Integration.  However the LCT001 is not found

Conbee is working fine as I was able to pair an Hue LLC020 and can control it without problems.
The Hue-BT App cannot be used anymore (says that you need to use the Hue App) and besides, I think the LCT001 does not have bluetooth.