dresden-elektronik/deconz-rest-plugin #1879: LEDVANCE On/Off Smart Plug type "Smart plug" makes it uncontrollable via Hue Binding with Alexa

octobunny, 2019-09-17:
I saw a solution for with LEDVANCE lamp in another issue and I'm wondering if it's a similar issue where the type is not recognized as a valid Hue binding : https://github.com/dresden-elektronik/deconz-rest-plugin/issues/1835

This plug-in module in this case is sold as SYLVANIA SMART + ZigBee Indoor Smart Plug but is described in Phoscon and DeCONZ as a LEDVANCE device.  
https://www.amazon.com/Sylvania-Smart-Home-72922-LIGHTIFY/dp/B01M6UM8QD/ref=sr_1_4?keywords=zigbee+smart+plug&qid=1568750843&sr=8-4

Object information.  
{
"type": "Smart plug",
"name": "Family - Corner Display Cabinet",
"modelid": "PLUG",
"swversion": "00102101",
"uniqueid": "f0:d1:b8:00:00:11:45:6d-01",
"manufacturername": "LEDVANCE",
"state": {
"on": false,
"alert": "none",
"reachable": true
},
"etag": "9050550d8f6ca23eb7f56b443ffa15a4",
"hascolor": false
}

By comparison, in DeCONZ, it's given a question mark while Ikea Smart Plug is identified as a light:
[image]

Can anyone confirm if this might be the case?

Thanks,
-M

bdickhaus, 2019-09-18:
What version of deCONZ are you running? In earlier versions they were identified as LEDVANCE and modelid PLUG. I have three of these and this is what the REST API info looks like with deCONZ 2.05.69.

{
  "etag": "6d2558c553d81119c7e5a59e3ea209aa",
  "hascolor": false,
  "manufacturername": "Samjin",
  "modelid": "outlet",
  "name": "urcharger",
  "state": {
    "alert": "none",
    "on": false,
    "reachable": true
  },
  "swversion": "00000011",
  "type": "On/Off output",
  "uniqueid": "28:6d:97:00:01:03:e9:1a-01"
}

octobunny, 2019-09-18:
Currently on 2.05.69, but these were added in 2.05.57. Maybe the nodes need to be completely removed before running another discovery in .69?

bdickhaus, 2019-09-18:
Sorry, I confused myself. I had three of those but had some issues with them, so I replaced them with the Samsung plugs which is what I posted previously.

octobunny, 2019-10-23:
Unfortunately seems that running another discovery on .69 doesn't change anything.

closed 2020-02-27 by stale[bot]