dresden-elektronik/deconz-rest-plugin #1881: Conbee Stick II Update impossible (Docker on Ubuntu)

daendaen, 2019-09-18:
Hi,

i try to make an update. The message from my gw version is "does not exist" 
Source to make the update: https://github.com/marthoc/docker-deconz#updating-conbeeraspbee-firmware

The Stick works with deconz perfectly at the moment. But i will make the update from my version to the latest

Here are the logs:

`
[marthoc/deconz] Starting deCONZ...,
17:08:54:906 DB sqlite version 3.16.2,
17:08:54:907 DB PRAGMA page_count: 35,
17:08:54:906 GW sd-card image version file does not exist: /root/.local/share/dresden-elektronik/deCONZ/gw-version,
17:08:54:906 gw run mode: docker,
17:08:54:905 parent process /bin/sh,
17:08:54:831 ZCLDB init file /root/.local/share/dresden-elektronik/deCONZ/zcldb.txt,
17:08:54:830 COM: --dev: /dev/ttyACM0 (ConBee II),
17:08:54:791 HTTP Server listen on address 0.0.0.0, port: 80, root: /usr/share/deCONZ/webapp/,
17:08:54:799 CTRL. 3.16.217:08:54:830 COM: /dev/ttyACM0 / serialno: ,
libpng warning: iCCP: known incorrect sRGB profile,
[marthoc/deconz] Starting without VNC GUI,
[marthoc/deconz] Websockets port: 443,
[marthoc/deconz] Web UI port: 80,
[marthoc/deconz] Current deCONZ version: 2.05.66,
17:08:54:907 DB PRAGMA page_size: 4096,
17:08:54:907 DB PRAGMA freelist_count: 1,
17:08:54:907 DB file size 143360 bytes, free pages 1,
17:08:54:907 DB PRAGMA user_version: 6,
17:08:54:907 DB cleanup,
17:08:54:907 DB create temporary views,
17:08:54:917 don't close database yet, keep open for 900 seconds,
17:08:54:918 started websocket server at port 443,
17:08:54:919 GW firmware update not supported on x86 linux headless,
17:08:54:919 discovery updated announce interval to 10 minutes,
17:08:54:920 found node plugin: libde_rest_plugin.so - REST API Plugin,
17:08:54:925 found node plugin: libstd_otau_plugin.so - STD OTAU Plugin,
17:08:54:921 found node plugin: libde_signal_plugin.so - Signal Monitor Plugin,
17:08:54:941 COM: /dev/ttyACM0 / serialno: ,
17:08:54:941 COM: --dev: /dev/ttyACM0 (ConBee II),
This plugin does not support propagateSizeHints(),
17:08:54:967 Device firmware version 0x26490700,
17:08:54:984 unlocked max nodes: 200,
17:08:55:061 new node - ext: 0x00212effff0487b4, nwk: 0x0000,
17:08:55:055 Device protocol version: 0x010B`

Can anybody help/support me?

Greetings

manup, 2019-09-18:
Manually update steps can also be found at https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Update-deCONZ-manually#update-in-ubuntu

Note since you are already on version 0x26490700 that's fine and you can stay on that version. The update to latest 0x264a0700 is only needed for other projects like ZHA component and Mozilla IoT gateway.

MattL0, 2019-09-22:
is it better to goes back to 0x26490700 if i do not use those two?

manup, 2019-09-23:
No, 0x264a0700 is fine

closed 2020-01-28 by stale[bot]