dresden-elektronik/deconz-rest-plugin #1883: Light stop working, Groups still work

sub0ne, 2019-09-18:
I got Conbee 2 stick running with Pi 3b+ & Raspbian & Phoscon app. I got two groups with several light in them. After several hours of running the light can not be accessed directly anymore. Via the groups the still can be turned on and of. But the individual lights can not be controlled anymore. They are also greyed out in the Phoscon app. This makes the whole thing useless. Any ideas?

MB

madshell, 2019-09-19:
What do you mean by "the lights can not be accessed directly anymore"? What is "directly" here?

balu-, 2019-09-24:
I have seen similar behaviour:
when i switch the group on or off via the phoscon web app, all lights of the group turn on or off.
If i try to switch single lights in the group, or change the color/brightness some lights of the group seem to not "recive" the changes (don't react). 

So it seems as if they are ignoring their individual commands but react to the group commands.

JBS5, 2019-09-24:
Does this happen with Tradfri lights? If yes, take a look at https://github.com/dresden-elektronik/deconz-rest-plugin/issues/1261 where I describe the same issue with a lot of replies over the few monts.

In my case: sooner of later: also the group command doesn't work anymore.

balu-, 2019-09-25:
Thanks for the hint 
(TLDR for everyone else: seems to be like a firmeware problem)

For me personaly the hickups have been with Tradfri GU10 Spots,
while taking a closer look i've seen that one of the spots seems to have been running firmware version  1.2.217.
I've updated it to  1.2.221 and will take a close look if anything changes. 
(But my hopes are low)

closed 2020-01-31 by stale[bot]