dresden-elektronik/deconz-rest-plugin #1884: Issue reaching https://dresden-light.appspot.com/discover!

wvuyk, 2019-09-19:
[image]

Is the site off line, or to popular now?

manup, 2019-09-19:
The second one :) Albeit the quota was increased several times, it gets hit faster since way more gateways are up.

We are currently switching to a new solution as described in https://github.com/dresden-elektronik/deconz-rest-plugin/issues/1595#issuecomment-532987000

wvuyk, 2019-09-19:
Oh, ok. I am using this same link in my plugin. I better start using it too then. How long before the old link will be removed?

manup, 2019-09-19:
>I better start using it too then

Yes I'd recommend that. Please note that the new service might have very small down times (a few seconds here and there) while it is developed and until we have our load balancer finished.

Since the traffic already is heavy you might notice some timeouts or errors in the logs. But mostly it should work at least after a few seconds on the next query attempt.

>How long before the old link will be removed?

We're planning to keep it alive for at least one year, maybe more, since there are many gateways which use older versions.

wvuyk, 2019-09-19:
I will make sure access is allowing a bit of slack, with a few slow retries for now.

Thanks! It might indeed take some time here to for the users to switch over. Not all want to be on the latest....

closed 2019-09-19 by wvuyk