dresden-elektronik/deconz-rest-plugin #1885: Dimming problem with HUE dimming switch since 2_05_69

jo-me, 2019-09-19:
I upgraded from 2_05_67 to 2_05_69 yesterday and apparently that broke something regarding my HUE dimmer switch. 

It is part of a group with one HUE bulb. 
I have keybindings configured the default way for the 4 buttons: on / brighter / darker / off.

[image]

This worked until yesterday. 
Now, when I press the "darker" button, the bulb does not get one level darker but fades all the way to off.

On/off/brighter buttons seem to work normal.

I've deleted the key bindings and redid them but it did not change the behavior.

Is this a bug with the new deconz version?

YKO-de, 2019-09-19:
I've been recreating it. It worked for me. Did you press long or short while dimming?

jo-me, 2019-09-19:
Just a short press.

YKO-de, 2019-09-19:
Please make sure that the key is assigned with two functions. 
Press long for dimming and short for off.

jo-me, 2019-09-19:
Has this behavior changed in deconz?

I still can short press the other button to make the bulb one step brighter at a time. It just broke for the other way. In order to reduce to a certain brightness I now have to turn it off and then (short) press dim up repeatedly to set it to the desired level. It did work before..

YKO-de, 2019-09-19:
Do you have the bulb in the switch's group too? Or just the group you configured with the editor?

Now here's what happened to me. If I have the lamp in both groups, then it is exactly the behavior described by you. If I have the lamp only in the switch group, then the dimming works step by step. If the lamp is only in the group that was configured with the editor, then dimming only works if you press and hold the button.

jo-me, 2019-09-19:
Yeah you were right. Something caused the bulb to be be added to the switch's group. I'm pretty sure it was empty before. I can confirm the behavior you are describing: The bulb behaves differently when in the one group than in the other group. 

To me this is a bug. though. It should be consistent across groups.
I currently use a dedicated group because I use two remotes for the same bulb and it makes sense to have these 3 things in one group, doesn't it?

Als, the "press to turn off, hold to dim" action makes only sense if you have a remote consisting of only one or two buttons where the on/off functionality has to be overlayed with long press to offer dimming functionality. For the hue remotes there are 4 dedicated buttons (see screenshot above). You'd want the same behavior it comes with out of the box. On / off buttons just turn on and off. The other 2 buttons are dedicated to increasing and decreasing brightness. Holding the brightness buttons should continuously increase/decrease the brigthness until minimum or maximum are reached. (Minimum brightness is not completely off, though).

I understand that the action it is now might be useful for certain remotes in order to have short and long press combined, but then there should be a new action to increase / decrease brightness stepwise on short press. This would make it more intuitive.

YKO-de, 2019-09-24:
If you want to have the behavior that is provided by the switch, please use the switch group and not the editor. The editor is designed to do something different with the switch than the manufacturer intended. Stepwise dimming is currently only possible with switches that do not support a long press function.

jo-me, 2019-09-24:
Are you planning to implement this for custom groups as well? There are many switches with dedicated dim up/down buttons that would behave very strange if you'd always have to press&hold.

Also, why is this behavior different for dimming up and down? Like I said, short press on the dim-up button produces step-wise dimming while short press on dim-down fades the bulb to off.

YKO-de, 2019-10-01:
This is not currently planned. But we will take it up and deliver it later if necessary.

closed 2019-10-01 by YKO-de

Luigi-01, 2019-10-23:
I hope you will incorporate a standard dimmer function. Now I had to use hue essentials to have a proper dimmer function. I would say, first the basics and than the extras. The same for the ikea dimmer, Hue essentials made it work, not deconz. Most people want to run there software local, now I had to use extra outside software. For that reason I think that a standard dimmer function is essential. No/Off or what ever added.

hvorragend, 2019-11-17:
The current solution is really very strange. I would really appreciate another solution.

hoblins, 2021-11-24:
> Als, the "press to turn off, hold to dim" action makes only sense if you have a remote consisting of only one or two buttons where the on/off functionality has to be overlayed with long press to [...]

I totally agree with this description from @jo-me . A simple button behaviour would be very advantageous for e.g. the four button Hue dimmer switch. In addition it should be quite easy to implement, even easier than the current short / long press action.