dresden-elektronik/deconz-rest-plugin #1886: dump zigbee packets similar to tcpdump?

yamlCase, 2019-09-19:
I'd like to pipe my zigbee traffic from the raspbee into a capture file or remote instance to see exactly what the controller is seeing.  Similar to how tcpdump can work for tcp/ip packets.  So far all I've found is zshark needs a second device to sniff.  Is there a way I'm not seeing?  If not, I'd like to turn this into a feature request.  Thanks!

manup, 2019-09-19:
ZSHARK also works with RaspBee, but it can only be either a sniffer or coordinator, not both at the same time.

yamlCase, 2019-09-20:
there's gotta be a way to tee the pi serial device and somehow get that into usable format for zshark.

manup, 2019-09-20:
The pcap format is actually quite simple. It would be possible to write a forwarder but the problem is that the coordinator firmware only provides higher APS level frames and a full Zigbee frame has also MAC, NWK underneath – these would be needed to filled with reasonable bogus data.

closed 2020-01-26 by stale[bot]