dresden-elektronik/deconz-rest-plugin #1887: Busch-Jaeger Friends of Hue doesn't join network

UltraSub, 2019-09-19:
I recently purchased a Busch-Jaeger Friends of Hue (6716 UJ-84) switch after reading it was supported by Deconz. Whatever I try, I cannot get it to join the zigbee network, Phoscon doesn't see them, Deconz also doesn't. I tried:

- Devices>switches>add new>Busch-Jaeger>control element 2gang, and following the instructions
- Devices>switches>add new>other, let it scan while following commisioning procedure for the correct channel. (Used https://download.vimar.com/irj/go/km/docs/z_catalogo/DOCUMENT/03906IEN.82882.pdf since it is the same module).
- Devices>switches>add new>Busch-Jaeger>control element 2gang, not following the instructions but following the commisioning procedure above.

Switch was located a meter away from the Conbee stick. No dice. So I thought it must be a DOA, and I got out the Hue hub. Instant pairing, while providing a different button press sequence (for channel 20). So I moved the hub to channel 15, since that is what Deconz is running on currently, and paired it again. Instant, but different buttons as expected. It was indeed the left upper button for 10 seconds for commisioning to channel 15, and click lower left+upper right to end commisioning and save. After finding out that sequence, I removed it from the Hue hub and tried all Phoscon steps listed above, while using the Hue provided sequence for channel 15. Again, nothing.

I'm running Deconz in a container (marthoc/deconz:amd64-2.05.67) on a Conbee stick. Everything else working fine. Running a bunch of Hue lights and 2 Hue motion sensors, 3 Aqara temp sensors and 3 IKEA Fyrtur blinds. Very stable and fast network.

Does anyone have an idea why others seem to have these switches running (I tried to find and read all issues here) and what I'm doing wrong?

UltraSub, 2019-09-21:
Fixed. I noticed that while I was running a very recent Deconz, the Conbee stick wasn't at latest. Flashed the firmware of the stick to 26330500 and the switch immediately showed up in Phoscon.

closed 2019-09-21 by UltraSub

sokki, 2019-12-17:
@UltraSub I've a hard time adding the switch.
What did u do to get it working? `Devices>switches>add new>other` and hold the lower right button for 10 secs?

sokki, 2019-12-17:
nevermind - got it working by following https://phoscon.de/en/support#pairing-friends-of-hue-switch