dresden-elektronik/deconz-rest-plugin #1888: New device request without GUI

nigobo, 2019-09-19:
Hi!

I am wondering if it is possible to help with _New device requests_ when running the Deconz software without GUI. I recently bought three new IKEA lamps that aren't supported.

- IKEA 250 lm filament light E27 (2200K) LED1842G3 (https://www.ikea.com/fr/fr/p/tradfri-ampoule-led-e27-250-lumen-sans-fil-a-variateur-dintensite-lumiere-chaude-globe-verre-transparent-brun-70455676/)
- IKEA 806 lm E27 (2700K) LED1836G9 (https://www.ikea.com/fr/fr/p/tradfri-ampoule-led-e27-806-lumen-sans-fil-a-variateur-dintensite-blanc-chaud-opalin-90408797/)

Regards,
Jocke

aagator, 2019-09-24:
Either this or add IKEA 250 lm filament light E27 (2200K) LED1842G3 ;)

Smanar, 2019-09-26:
The pairing procedure don't work ?
Almost all bulbs are working out of the box.

aagator, 2019-09-26:
IT starts to pair but suddenly stops. The bulb will not enter pairing again unless reset. Tried several times...

nigobo, 2019-09-27:
Tried again today. Both works.  

The procedure was: 

1. Start the "new lights" search in Phoscon
2. Do the reset sequence on/off six times and make sure that the light starts to flicker/pulse
3. Wait, and now it works.

_The strange thing is that both worked on the first try today. I really don't understand what the difference is. But I'm happy it works!_

[image]
[image]

closed 2019-09-27 by nigobo