dresden-elektronik/deconz-rest-plugin #1890: GLEDOPTO led controller now working properly

cybergatto, 2019-09-20:
Hello,
    i'm facing issues with one gledopto ledstripe controller.
the controller is WW/CW GL-C-006.

i'm able to discover it correctly from phoscon app and using the sliders i can control the leds:

[image]

what is not working is the ON/OFF command, i can switch the light on and of from phoscon app but there's no response from the ledstripe that remain powered on anyway:

`deconz           | 10:50:08:111 0x00124B001D10FE06 onOff 0 --> 1
deconz           | 10:50:08:413 0x00124B001D10FE06 level 0 --> 1
deconz           | 10:50:09:333         0x00124B001D10FE06 force poll (2)
deconz           | 10:50:10:337         0x00124B001D10FE06 force poll (2)
deconz           | 10:50:37:692 Current channel 25
deconz           | 10:50:37:700 Device TTL 1791 s flags: 0x7
deconz           | 10:50:55:476 sensor 7 (lumi.sensor_motion.aq2): disable presence
deconz           | 10:50:55:776 Set sensor check interval to 1000 milliseconds
deconz           | 10:51:06:576         0x00124B001D10FE06 force poll (2)
deconz           | 10:51:07:585         0x00124B001D10FE06 force poll (2)
deconz           | 10:51:08:629         0x00124B001D10FE06 force poll (2)
deconz           | 10:51:09:113 0x00124B001D10FE06 level 182 --> 183
deconz           | 10:51:09:638         0x00124B001D10FE06 force poll (2)
deconz           | 10:51:10:831         0x00124B001D10FE06 force poll (2)
deconz           | 10:51:11:830         0x00124B001D10FE06 force poll (2)
deconz           | 10:51:12:925         0x00124B001D10FE06 force poll (2)
deconz           | 10:51:13:935         0x00124B001D10FE06 force poll (2)
deconz           | 10:51:24:217         0x00124B001D10FE06 force poll (2)
deconz           | 10:51:27:164         0x00124B001D10FE06 force poll (2)
deconz           | 10:51:28:601         0x00124B001D10FE06 force poll (2)
deconz           | 10:51:29:450         0x00124B001D10FE06 force poll (2)
deconz           | 10:51:30:154         0x00124B001D10FE06 force poll (2)
deconz           | 10:51:31:831         0x00124B001D10FE06 force poll (2)
deconz           | 10:51:32:114 0x00124B001D10FE06 onOff 1 --> 0
deconz           | 10:51:32:338         0x00124B001D10FE06 force poll (2)
deconz           | 10:51:33:428         0x00124B001D10FE06 force poll (2)
deconz           | 10:51:34:085         0x00124B001D10FE06 force poll (2)
deconz           | 10:51:35:213 0x00124B001D10FE06 onOff 1 --> 0
deconz           | 10:51:37:691 Current channel 25
`

is this a know issue? may i provide specific log that might hel you in developement?

thanks

robidog, 2019-09-25:
Can confirm that these controllers behave not as expected. Tested with two different GL-C-006 devices, running FW version 1.06 and 1.07. While setting scenes they show the correct color temp,  but when scene is recalled from Phoscon app, only CW is lit, WW remains off at all times.

cybergatto, 2019-09-30:
@robidog tks, title corrected.Yes i can reproduce your behaviour too.

closed 2020-02-04 by stale[bot]